nextcloud/nextcloudpi #144: "Expand Filesystem" not working for USB-device

frank-44, 2017-08-23:
I use my Raspberry 3 without SD-card. Only USB-device (program_usb_boot_mode=1).

$ lsblk
NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda      8:0    1 14,8G  0 disk 
├─sda1   8:1    1 41,8M  0 part /boot
└─sda2   8:2    1 14,8G  0 part /

The "7 Advanced Options -> A1 Expand Filesystem" from rasp-config seems not working for USB-devices. 
The message "sda2 is not an SD card. Don't know how to expand" is shown.

Pure rasbian expands the filesystem automatically after rebooting.
Maybe NextCloudPi could do so also with next update.

(No need for running:  sudo resize2fs /dev/sda2)

Used NextCloudPi version: v0.24.1

nachoparker, 2017-08-23:
Hi,

That is interesting, because there is not much different from 'pure Raspbian'. I have not tried that setup myself.

Could you try running this command, then rebooting and trying again?

```
sudo rm etc/udev/rules.d/90-qemu.rules
```

Thanks

nachoparker, 2017-08-23:
> Maybe NextCloudPi could do so also with next update.

What do you mean exactly? could you explain, plz ?

frank-44, 2017-08-23:
I try to explain:

After copying the img-file to SD-card or USB-device expanding of filesystems is necessary, because img-file size is 2.6GB and SD-card/USB size is typically bigger e.g. 8 or 16 GB.
 - 'Pure Raspbian' detects this issue during first boot and expands filesystem and reboots.
 - With 'NextCloudPi' it was necessary to run: '$ sudo resize2fs /dev/sda2' for using the whole size of the device (SD / USB).

That's all. No problem in my installation yet.

(For 'Pure Raspbian' installation I used monitor connected. 'NextCloudPi' installation was headless.)

nachoparker, 2017-08-23:
Mmm, ok, I thought pure Raspbian only auto-expands the boot partition. So it auto-expands the root partition as well?

nachoparker, 2017-08-30:
Looking at the code for `raspi-config`... this is a Raspbian issue. At the moment the method to run everything from USB is not supported or documented by NextCloudPi, even though it seems to work.

For the `raspi-config` issue, I suggest you report it to Raspbian developers

If somebody would be willing to fix this use case (run from USB) for NCP, support for that is welcome and I accept PRs. I do not use that setup myself, so I don't have the resources to support it myself.

Thanks for reporting!

closed 2017-08-30 by nachoparker