nextcloud/nextcloudpi #146: php-fpm7 Overload when browsing index.php/apps/activity/

ttoopp, 2017-08-23:
Hi,

since yesterday after a update, I have some problems with my nextcloud.
My system hangs if I navigate to index.php/apps/activity/ and nextcloud beginns to create thumbs.

```
top - 09:11:41 up 11:50,  1 user,  load average: 9,02, 11,19, 19,17
Tasks: 143 total,   4 running, 139 sleeping,   0 stopped,   0 zombie
%Cpu(s): 32,2 us, 23,7 sy,  0,0 ni,  1,1 id, 42,9 wa,  0,0 hi,  0,1 si,  0,0 st
KiB Mem:    945516 total,   831796 used,   113720 free,      284 buffers
KiB Swap:   102396 total,   102396 used,        0 free.    30792 cached Mem

 PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
6977 www-data  20   0  719796 297204   9368 R 100,0 31,4   0:13.97 php-fpm7.0
6971 www-data  20   0  494520  75168  11704 D 100,0  7,9   0:19.06 php-fpm7.0
40 root      20   0       0      0      0 S 100,0  0,0  39:12.95 kswapd0
6979 www-data  20   0  673012 164648   9672 D  12,5 17,4   0:00.77 php-fpm7.0
   1194 mysql     20   0  633304 120348      0 S  10,9 12,7   5:38.07 mysqld
6982 www-data  20   0  518444  50884   9660 R   9,9  5,4   0:00.38 php-fpm7.0
6980 www-data  20   0  567476  88776   9656 R   8,3  9,4   0:00.46 php-fpm7.0
92 root      20   0       0      0      0 S   4,2  0,0   0:31.68 mmcqd/0
5762 www-data  20   0  448916   4760     16 S   1,3  0,5   0:10.99 apache2
6166 root      20   0       0      0      0 S   1,3  0,0   0:06.71 kworker/0:0
```

Raspberry Pi 3 / NextcloudPi 12.0.2

nachoparker, 2017-08-23:
What kind of update? NCP update? NC update? distro update?

ttoopp, 2017-08-24:
Okay I think I found the issue. When nextcloud begin creating the preview thumbs for the website, the php-fpm7.0 services crash my Pi. There was 100 pics with each 10 MB, so the Pi need many power for creating the thumbs.

My solution:

` pico /etc/php/7.0/fpm/pool.d/www.conf`

```
pm = dynamic
pm.max_children = 3
pm.start_servers = 2
pm.min_spare_servers = 1
pm.max_spare_servers = 3
pm.max_requests = 500
```
`/etc/init.d/php7.0-fpm restart`

5 processes are to much for he Pi

nachoparker, 2017-08-24:
@ttoopp great, thanks

Yes, I have read before in the NC forums that the thumbnail generation brings the whole pi down. They recommend to disable the pics app IIRC.

Maybe we should disable it by default? 

Good info! I'll play with those parameters ;)

nachoparker, 2017-08-24:
@ttoopp 

Do you think this parameter is really important? I suspect that we can leave this off.

```
; The number of requests each child process should execute before respawning.
; This can be useful to work around memory leaks in 3rd party libraries. For
; endless request processing specify '0'. Equivalent to PHP_FCGI_MAX_REQUESTS.
; Default Value: 0
;pm.max_requests = 500
```

Also another question, once you reduced the number of children, did the thumbnail generation work while not hogging the whole system?

ttoopp, 2017-08-24:
I read in a other Forum about the pm.max_requests parameter. I think it is not necessary, to activate it.
I test my Pi with new files and the generation work well and don't hog the system. Furthermore I use this addon, to generate the previews of new uploaded files with a cronjob every 30 minutes https://apps.nextcloud.com/apps/previewgenerator

nachoparker, 2017-08-24:
nice, I didn't know that one ;)

Would you be willing to retest this with only and the other parameters unmodified, and report if it works well for you ?

```
pm.max_children = 3
```

If it does, we can introduce it by default in the configuration, or, if you are feeling brave,  you can submit a PR with this ;)

ttoopp, 2017-08-24:
Do you have the other original values? I have overwritten the original :)

nachoparker, 2017-08-24:
xD

```
pm.max_children = 5
;pm.max_requests = 500
```

The rest are the same

ttoopp, 2017-08-24:
Okay, I put some new pics (50 a 10 MB)  on my nextcloud. Change the conf to
```
pm.max_children = 5
;pm.max_requests = 500
```

restart php7.0-fpm.service, open the folder with the pics in my browser and check the load.
After 10 seconds (load over 37) I get no reaction (ssh and web) from the Pi and have to wait few minutes until I can work again.
For me is pm.max-_childern = 3 the better solution :) everyone has to decide for themselves ;)

nachoparker, 2017-08-24:
great, thank you!! 

3 it is then ;)

nachoparker, 2017-08-29:
Implemented `v0.24.6`

closed 2017-08-29 by nachoparker