nextcloud/nextcloudpi #147: install postfix and have it preconfigured

nachoparker, 2017-08-23:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

### Expected behavior

### Actual behaviour

### Steps to reproduce, from a freshly installed image

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -n 100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -n 100 /var/log/mysql/*.log`
```
</details>

Rufio87, 2017-08-24:
isn't this a built-in feature of nextcloud?: https://docs.nextcloud.com/server/12/admin_manual/configuration_server/email_configuration.html

nachoparker, 2017-08-24:
Hi, 

I would like that we investigated this a little bit to provide support. Maybe we'll end up using the PHP method, maybe we'll end up using postfix, which is recommended.

In any case, at the very least we will need to have some `sendmail` preinstalled, and probably some preconfiguration.

It would be nice that out of the box, everything is installed and configured for NCP to be able to send emails.

Thoughts?

JimmyKater, 2017-08-24:
afaik if you want to use php/sendmail mail it is necessary to have an installed and working email system on your server.

so as NCP is designed for newbies and noops like me who of us HAVE such a fully installed and working emailsystem?
so i'd prefer to sent up smtp as a first step... but then again: who of us noobs does know the credentials to put in if you ask for imap or post3/smtp?

ObfuscateTheWeb, 2017-08-25:
A fully working email-capable nextcloud would be great !

However, according to the [documentation](https://docs.nextcloud.com/server/9/admin_manual/configuration_server/email_configuration.html), 

> Nextcloud does not contain a full email server, but rather connects to your existing mail server. You must have a functioning mail server for Nextcloud to be able to send emails. You may have a mail server on the same machine as Nextcloud, or it may be a remote server.

So yep, @nachoparker  you'll need to also preinstall and configure a email server for this to be pulled off
:)

nachoparker, 2017-09-06:
Implemented `v0.26.16`

closed 2017-09-06 by nachoparker

ObfuscateTheWeb, 2017-09-07:
How do we use it?

nachoparker, 2017-09-07:
Well, you can send a test email to yourself from the settings/admin/additional section. You need to have an email in your admin personal profile.

Also if a user loses a password you can receive an email to reset it, provided the user has set up your email in his/her personal profile.

Also notifications can be sent to emails like when you share directories, probably other things as well and more that might come in the future.

JimBobHarris, 2017-11-14:
Please put in the instructions an example for using gmail to send notifications etc., filling in the actual form at

https://ourdomain.noip.com/index.php/settings/admin/additional

All fields, as for an _ordinary_ gmail user, not some Linux expert who runs their own SMTP server on their Linux box and knows all these answers already. So, for example, in the "From address" fields, do we _make_up_ a desired name, like OurNextCloud @ ourdomain.noip.com, or must we put in our _real_ gmail address, e.g. john.doe@gmail.com ? And what encryption, server address, port number, etc., and do we check "Authentication required" or not, etc.?

Please repeat the above for any _other_ email servers that have _many_ posts on various forums etc. with obvious frustrations with the lack of clear instructions for filling out this web page. Of course, you can't be expected to do _every_ such server, but the most-common 2 or 3 examples would help _many_ lost users. 

Thanks, and I hope you find in this post some useful info about how lost we noobs are.

Daniel-Krebs, 2017-11-17:
@JimBobHarris I don't have a gmail account to test with but please have a look here:
https://support.google.com/mail/answer/7126229?hl=en
You should try your real mail address like jimbobharris@gmail.com, use SMTP with authentification enabled (try cleartext from the drop down menu), STARTTLS on Port 587, and use smtp.gmail.com as a server.
Also (quoted from the link):
```
Step 1: Check that IMAP is turned on
On your computer, open Gmail.
In the top right, click Settings Settings.
Click Settings.
Click the Forwarding and POP/IMAP tab.
In the "IMAP Access" section, select Enable IMAP.
Click Save Changes.

```

nachoparker, 2017-11-17:
if anyone configures this successfully, it would be benefitial to share the steps with us so we can write them on the wiki.

I use PHP method and don't have a gmail account, so cant help

JimBobHarris, 2017-11-22:
It seems I got it to work, thanks to links and info from Daniel Krebs and others. Here are my entries at ...settings/admin/additional:

Send mode: SMTP  Encryption: STARTTLS

From address: james.bob.harris   @   gmail.com

Authentication method: Login   [checked] Authentication required

Server address: smtp.gmail.com   :   587

Credentials: james.bob.harris@gmail.com   [my normal gmail login password] [Store credentials] (clicked)

Thanks again; I will test further with my actual nextcloudpi system and try to cause it to generate email from the system; I will report success or failure; others may get to it before I do.

Pant, 2017-11-22:
I verify the right entries are in [JimBobHarris comment](https://github.com/nextcloud/nextcloudpi/issues/147#issuecomment-346227234).

nachoparker, 2017-11-22:
thanks @Daniel-Krebs @Pant and @JimBobHarris 

you can generate a test email from the admin/other settings section. Once we verify this, it would be good to add it to the wiki.

also it is interesting to cover 2 factor authentication

Pant, 2017-11-22:
I have successfully sent an email to myself, from the test mail section. So I verify this.

nachoparker, 2017-11-22:
awesome, can someone add this to the wiki please? @Daniel-Krebs  or @Pant ?

JimBobHarris, 2017-11-22:
I agree the "Send email" button in .../settings/admin/additional sent me an email, but I have yet to figure out some artifice to cause the System to actually initiate and send an email to a User. I tried changing a test User's Limit to 2 MB and they have way more than that in use, but no emails were triggered. Any suggestions?

nachoparker, 2017-11-23:
I don't know myself.. you might get more ideas in the [nextcloud forums](https://help.nextcloud.com/)

I do recieve emails on activity, like shared links

nachoparker, 2017-11-23:
@Pant  wrote a guide on the wiki

https://github.com/nextcloud/nextcloudpi/wiki/Email-settings

Thanks!