nextcloud/nextcloudpi #148: moving database

Segdon, 2017-08-23:
(First of all, some praise: you are doing a fantastic job with nextcloudpi! you're a lifesaver)

Somewhere i read (on your blog), if you got issues with a blank page after moving the database, just put de database back on the sd-card. 
This truly works (although not a great solution :-)  )

After some installations and tests (i usually re-install new stuff 20 times, just to get the hang of it (selfthaught :-/ ), I noticed that the blank page only appears if you move it by way of nextcloudpi-config. I don't have the problem when I move the database with the web-interface.

Taught it might interest you.

Greetings
Angelos

nachoparker, 2017-08-23:
Hi,

Thanks for letting me know!

I suspect that has something to do with the HDD going idle and the database crashing. It does not happen in my old drives.

I don't think that it has to do with either doing it from `nextcloudpi-config` or ncp-web, because at the end of the day, they run the exact same code for moving the database, so that must be a coincidence.

Let us know if you find anything else interesting or confusing during the setup ;)

Duplicate of #15

closed 2017-08-23 by nachoparker

nachoparker, 2017-09-05:
Duplicate of #15