nextcloud/nextcloudpi #149: No ocdata since 0.24.1

deljones, 2017-08-23:
Hi..

Since updating to 0.24.1 I can no longer get access to ocdata...:(

Should I just restore to previous back up?

Del

[image]

nachoparker, 2017-08-23:
Hi del,

How did you update?

ObfuscateTheWeb, 2017-08-23:
I also had this issue a few days back, I just did a `sudo df -h` , saw that it wasn't mounted in the correct directory, kept doing `sudo umount /media/USBdrive(whatever number)` and finally manually mount it to the correct directory (in my case, though I believe yours is the same) with `sudo mount /dev/sda1 /media/USBdrive`

Reboot, and all is fine again. Don't know what caused it to begin with, but when this happens you need to fix manually your data directory

nachoparker, 2017-08-23:
Thanks, @ObfuscateTheWeb 

I fixed some stuff with automount, where it was being incorrectly being mounted to `/media/USBdrive1` instead of `/media/USBdrive`. In any case, the most failproof thing to do is to use `/media/disklabel` because that will always point to the right place even if you plug in more than one HDD.

In any case, if you verify that it should be `/media/USBdrive`, check (and correct) the value in `/var/www/nextcloud/config/config.php`

deljones, 2017-08-23:
I just did a check for updates, did the upgrade date, restated and voila... No access 😕

Dj

deljones, 2017-08-23:
I'll have a quick ssh and see which USBdrive number ocdata is on.. 

Dj

deljones, 2017-08-23:
Hi... Yes fixed. Had to edit config.php and edit the location from USBdisk to USBdisk1

Don't a few updates with no issues. First time that happens.. 

Thanks all for your help. Back up again 

Dj

nachoparker, 2017-08-23:
Cool, I have to review automount with Stretch because it seems like it works differently in the new Raspbian version now :S

closed 2017-08-24 by nachoparker