nextcloud/nextcloudpi #150: Question: do trusted domains need to be configured at all?

niconoob, 2017-08-23:
Steps: 
- flashed sd card with image 2017-08-20 (v0.24)
- configure pi (new password, new hostname, locale), reboot
- configure wifi, reboot
- upgrade to v0.24.1
- access the web interface via https://192.168.xxx.xx:4443 (works after the usual warning from firefox)
- acces nextcloud under https://192.168.xxx.xx -> I get the "accessing from untrusted domain" error from nc as described in e.g. issue #21 and [this link](https://help.nextcloud.com/t/how-to-disable-you-are-accessing-the-server-from-an-untrusted-domain/4723)

I haven't used letsencrypt yet (I am planning on connecting the pi to the outside world at a later stage).

The threads I mentioned above (and the message from nc itself) suggest among others configuring trusted domains in config.php, but if I follow [the description here](https://ownyourbits.com/2017/02/13/nextcloud-ready-raspberry-pi-image/) it seems not to be necessary, hence the question:
- is it normal that I get this error if I'm not using dyndns (e.g. accessing from lan intead of a static domain)?
  - if no, how can I fix it / what did I do wrong in the sequence? I'll try the config.php fix if anything, unless advised otherwise
  - if yes, could it be added to the "first steps" section [here](https://ownyourbits.com/2017/02/13/nextcloud-ready-raspberry-pi-image/) and what to do about it? I really haven't done much (heck, I haven't even started nc)

JimmyKater, 2017-08-24:
> do trusted domains need to be configured at all?

from what i know i think they don't need to be configured at all. but then you can't access NC by no other names than declared in `config.php` - most probably it will look like this
`'trusted_domains' =>
  array (
    0 => 'localhost',
    1 => '192.168.xxx.yyy',
    2 => 'nextcloud.local',
  ),
`
from inside a lan that would be acceptable, i guess. if you want to access if from outside your own (w)lan - i see no other way than entering a new entry with an outside URL to this array. - and you should make sure that said outside URL is pointing to your device... and that it's reachable via https.

nachoparker, 2017-08-24:
> is it normal that I get this error if I'm not using dyndns ?

no, you shouldn't be getting that. It should be done automatically for you

> if no, how can I fix it / what did I do wrong in the sequence? I'll try the config.php fix if anything, unless advised otherwise

You can fix in manually editing `config.php`, but let's see... why did it fail for you?

 - you did `nc-wifi`, right? or did you set up wifi manually?

nachoparker, 2017-08-24:
@JimmyKater lol, we were writing at the same time! xD

Just so everybody learns how NCP works... 

 - your local IP will be added to the array in `config.php` on boot and on `nc-wifi`
 - your noip2 DDNS domain in another field ( [code](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/no-ip.sh#L75) )
 - your dnsmasq domain in another field ( [code](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/dnsmasq.sh#L63) )
 - your letsencrypt domain in another field ( [code](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/letsencrypt.sh#L61) )

So I think that all cases are covered... as long as you use `nextcloudpi-config` for everything :D

thx for the help

niconoob, 2017-08-24:
I did use nc-wifi (or whatever the menu item under nextcloud-config is called). So the question, indeed, is what happend?

If I look at the ip address as recommended by nacho [here](https://github.com/nextcloud/nextcloudpi/issues/135), I get
```
    inet 192.168.178.25/24 brd 192.168.178.255 scope global wlan0
    inet6 2a02:810d:8540:4f8:caef:xxxx:xxxx:xxxx/64 scope global mngtmpaddr noprefixroute dynamic 
```
Upon looking at the config.php file under /var/www/nextcloud/config, what I can see in "trusted domains" is the beginning of the ipv6 address: 2a02:810
 'trusted_domains' => 
 ```
 array (
    0 => 'localhost',
    1 => 
    array (
      '2a02:810' => '6',
    ),
```
Can it be that somehow my router is using ipv6 and that is what ncp picked up? I've tried entering the ipv6 address in the address bar in firefox but that doesn't work. I also have no clue what I could change in my router.

I've tried manually adding my ipv4 lan address like this:
`  'trusted_domains' => 
  array (
    0 => 'localhost',
    1 => 
    array (
      '2a02:810' => '6',
    ),
    2 ==>  '192.168.178.25',
  ),
` but after rebooting all I see is a blank page, not even the nc "untrusted" page. What is wrong with that syntax?

nachoparker, 2017-08-24:
> Can it be that somehow my router is using ipv6 and that is what ncp picked up? I've tried entering the ipv6 address in the address bar in firefox but that doesn't work. I also have no clue what I could change in my router.

Mmmm, that's what it looks like. I haven't tested with ipv6. Maybe we should just disable it in NCP

> but after rebooting all I see is a blank page, not even the nc "untrusted" page. What is wrong with that syntax?

Do this

```
array (
    0 => 'localhost',
    1 => '192.168.178.25',
  ),
```

nachoparker, 2017-08-24:
Ok, we detected a bug with ipv6. Thanks for the info

closed 2017-08-24 by nachoparker

nachoparker, 2017-08-30:
fixed `v0.24.7`