nextcloud/nextcloudpi #151: Question: NextCloudPi on external HDD booting from USB

elessarelfstar, 2017-08-23:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
0.24.1

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
NextCloudPi_08-16-17 (but currently up to date)

### Question
I've used the official method to enable boot from USB. I've flashed the NextCloudPi image directly to an external HD and am running straight from that with no sdcard present. All of the functions I use have been running fine (including updating from 0.24.0 to 0.24.1 and the updating to the new Rasbian version). I haven't seen anyone else say they were doing this, so just want to see if there are any known reasons that this would be a bad idea for NextCloudPi. Thanks!

frank-44, 2017-08-23:
I'm also running NextCloudPi without sd-card.
Everything fine.

elessarelfstar, 2017-08-24:
@frank-44 Good to know. Thanks!

_Sent from my Google Nexus 6 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

nachoparker, 2017-08-24:
yeah guys, good to know. I haven't even tried that myself ;)

Let us know if there are issues.

closed 2017-08-24 by nachoparker

deljones, 2017-08-24:
Interested to loose the SD card for nextcloudpi..Where can I find the HOWTO? 

Dj

frank-44, 2017-08-24:
I used this description: 

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

deljones, 2017-08-24:
@frank-44 many thanks.. I have a good working system right now but would love to get it of the SD card for the usual reason… Bit nervous about changing it now.. I’ll have a look at this and probably try with a new setup to see how it goes…

 

Thanks

 

Dj

 

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md