nextcloud/nextcloudpi #153: backup/restore/import/export nextcloudpi-config config

nachoparker, 2017-08-24:
maybe add it to `nc-backup`? maybe do it separatedly ?

JimmyKater, 2017-08-24:
suggestion: backup/restore together in one point. and import/export into another?

Ricardosgeral, 2017-11-10:
Hi
My auto backup is not working. 
I've putted  

7 days
2 files limit
data:yes
place in the tar file in /media/myCloudDrive/nextcloud-bkp

My suspicius are:

1- from time to time i reboot the pi. So if I reboot it before ending 7 days it losts the ability to do the backup! Could be this?

2 - permission issues to record in the folder. Which i think its strange because if i do directly the backup using nc-backup it works.

3 - the nc-backup-auto is not working. Which i think it should be working because at this point someone would have notice!

Some feedback is appreciated
thanks

nachoparker, 2017-11-10:
@Ricardosgeral It does work for me, so I think it is number 1.

nachoparker, 2017-11-28:
implemented in `v0.39.0`

closed 2017-11-28 by nachoparker

Ricardosgeral, 2017-11-28:
what is the difference between backup and  export and between restore and
import?

nachoparker, 2017-11-28:
backup and restore is for your NC instance: database, files, accounts, calendars...

import and export is for NCP: save the config of ncp-web, and go back to the original state on import, activating the services that are on in the config

Ricardosgeral, 2017-11-28:
ok. If I understand, Let's encript, duckdns, bruteforce, ...  will be
working with import after doing a nc-restore?

nachoparker, 2017-11-28:
that's the idea ;D

well, only those that have an `ACTIVE` field will be 'activated' again at this point, so let's encrypt for instance would have the config restored but wouldn't be activated.