nextcloud/nextcloudpi #154: ncp-scan error

fab1ton, 2017-08-24:
Out of nowhere I am getting the following error. Anybody knows how to fix that?

[image]

nachoparker, 2017-08-24:
aaah that happened to me once but can't remember right now how I fixed it.

Have you tried rebooting? xD

fab1ton, 2017-08-24:
Yes, couple of times but unfortunately that didn't help. I also rebooted my router. No success either.

nachoparker, 2017-08-24:
does that file `<your datadir>/<your username>/files/e848...` exist?

Is your database ok? ( output of `systemctl status mysqld` )

Segdon, 2017-08-24:
I have the same thing
i want to view the locked files in mysql, but how can i get in?
i have no password for the root user

nachoparker, 2017-08-24:
@Segdon really? what where you guys doing when it happened?

Segdon, 2017-08-24:
i think i found the problem
when copying from the disk, it took $RECYCLE.BIN along
i think scan flips on it? -does this sound possible?
deleted it and scan is running

nachoparker, 2017-08-24:
lol, it is quite possible because in bash and php the dollar sign `$` is a special character

closed 2017-08-24 by nachoparker