nextcloud/nextcloudpi #155: Problem with some special characters for password (username?) in nc-init

PartTimeDataScientist, 2017-08-24:
There is a problem with special characters in nc-init in 0.24.1

I am using  [Master Password](http://masterpasswordapp.com) to generate long and unique passwords for the different sites I am using. For my nextcloud appliance the generated password contains a closing bracket ')'. 

Entering this password in the nc-init script results in a  

**/usr/local/etc/library.sh: line 246: configure: command not found**

error.

Although I guess that this issue doesn't affect too many people and is also not a security issue like in the SQL injection cases I wanted to report it so that it might be fixed in future versions and other users running into that error might find the solution faster.

Obvious workaround: Use a dummy password in nc-init and change later using the web-interface

nachoparker, 2017-08-24:
hi,

I will see how we can make that work better with `nextcloudpi-config`.

In ncp-web it won't even work because for security reasons, the set of allowed characters is pretty restrictive.

Sounds right?

Thanks for reporting

PartTimeDataScientist, 2017-08-24:
Hi,

the very same password worked fine when I did create a new nextcloud user from the nextcloud web-interface. I didn't find any rules for allowed or disallowed characters - do you have any information about that?

nachoparker, 2017-08-24:
Well, that is just me trying to prevent XSS / CSRF attacks on the backend for ncp-web.

It's a basic security meassure to _sanitize_ the input, this is, to prevent usage of characters that could be used in a code injection attack.

The Nextcloud web interface is much more complicated, and it has its own handling for the input field for the password, but ncp-web just uses the same handling for all the text fields, so it is good to be extra cautious.

Looks like you already know that :D

PartTimeDataScientist, 2017-08-24:
I didn't try to change the password using ncp-web, but sanitizing input is a good idea and I appreciate the effort. 

For the nc-init a simple error message with disallowed characters would be fine or even a hint in the interface to trigger thinking about ) } ; # or whatever might be problematic. I had a "real" bug report almost ready when I looked at my password which I typically don't do (typically I just have it starred out in the interface) and thought about that this might be the root cause. ;-)

nachoparker, 2017-08-24:
> For the nc-init a simple error message with disallowed characters would be fine or even ...

agreed

nachoparker, 2017-10-06:
For security reasons, we will sanitize input but will not allow every possible character in the password

closed 2017-10-06 by nachoparker