nextcloud/nextcloudpi #156: trusted domain detection fails with ipv6

nachoparker, 2017-08-24:
From #150

nachoparker, 2017-08-30:
fixed `v0.24.7`

closed 2017-08-30 by nachoparker

mrkskwsnck, 2019-02-02:
I ran into an issue considering IPv6 with a fresh install from the Raspberry Pi image [NextCloudPi_RPi_01-13-19](https://ownyourbits.com/downloads/NextCloudPi_RPi_01-13-19/). The NCP version was 1.4.x something...

The IPv6 of the Raspberry Pi was already added to the trusted domains list. But I needed to surround it with [square brackets] to have an effect, like it is mentioned in the [documentation](https://docs.nextcloud.com/server/15/admin_manual/installation/installation_wizard.html#trusted-domains).

nachoparker, 2019-02-05:
@mrkskwsnck thanks for the heads up, would you be able to send a PR with this?