nextcloud/nextcloudpi #157: Let's Encrypt - seems like a python error

RyanTG, 2017-08-25:
### What version of NextCloudPi are you using? 0.24.3

### What is the base image that you installed on the SD card? I just upgraded to latest raspbian and nextcloud.

### Expected behavior
Running letsencrypt for first time.

### Actual behaviour
Seems to be a python error. **[GIST](https://gist.github.com/RyanTG/464ecb05e7bb22971c32fdb0489d01e8)**

</details>
<details>
<summary>Apache logs</summary>

```
~ $ systemctl status apache2
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: active (running) (Result: exit-code) since Mon 2017-08-07 18:01:48 UTC; 2 weeks 3 days
 Main PID: 1092 (apache2)
   CGroup: /system.slice/apache2.service
           ├─ 1092 /usr/sbin/apache2 -k start
           ├─23062 /usr/sbin/apache2 -k start
           └─23080 /usr/sbin/apache2 -k start

Aug 23 06:41:24 raspberrypi systemd[1]: Reloaded The Apache HTTP Server.
Aug 24 06:52:04 raspberrypi systemd[1]: Reloading The Apache HTTP Server.
Aug 24 06:52:05 raspberrypi apachectl[23017]: AH00558: apache2: Could not reliably determine the 
Aug 24 06:52:05 raspberrypi systemd[1]: Reloaded The Apache HTTP Server.
Aug 25 06:25:02 raspberrypi systemd[1]: Reloading The Apache HTTP Server.
Aug 25 06:25:03 raspberrypi apachectl[19319]: apache2: Syntax error on line 147 of /etc/apache2/a
Aug 25 06:25:03 raspberrypi apachectl[19319]: Action 'graceful' failed.
Aug 25 06:25:03 raspberrypi apachectl[19319]: The Apache error log may have more information.
Aug 25 06:25:03 raspberrypi systemd[1]: apache2.service: Control process exited, code=exited stat
Aug 25 06:25:03 raspberrypi systemd[1]: Reload failed for The Apache HTTP Server.

```
</details>
<details>
<summary>mariaDB logs</summary>

```
$ systemctl status mysqld
● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2017-08-25 03:12:11 UTC; 3h 36min ago
 Main PID: 29046 (mysqld)
   Status: "Taking your SQL requests now..."
   CGroup: /system.slice/mariadb.service
           └─29046 /usr/sbin/mysqld

Aug 25 03:12:09 raspberrypi systemd[1]: Starting MariaDB database server...
Aug 25 03:12:10 raspberrypi mysqld[29046]: 2017-08-25  3:12:10 1988472832 [Note] /usr/sbin/mysqld
Aug 25 03:12:11 raspberrypi systemd[1]: Started MariaDB database server.

tail -n 100 /var/log/mysql/*.log
2017-08-25 06:36:01 53ab9340 InnoDB: Error: Column last_update in table "mysql"."innodb_table_stats" is INT UNSIGNED NOT NULL but should be BINARY(4) NOT NULL (type mismatch).
2017-08-25 06:45:04 53ab9340 InnoDB: Error: Column last_update in table "mysql"."innodb_table_stats" is INT UNSIGNED NOT NULL but should be BINARY(4) NOT NULL (type mismatch).
2017-08-25 06:45:04 75f1e340 InnoDB: Error: Column last_update in table "mysql"."innodb_table_stats" is INT UNSIGNED NOT NULL but should be BINARY(4) NOT NULL (type mismatch).
2017-08-25 06:45:04 75f1e340 InnoDB: Error: Fetch of persistent statistics requested for table "nextcloud"."oc_notifications" but the required system tables mysql.innodb_table_stats and mysql.innodb_index_stats are not present or have unexpected structure. Using transient stats instead.
2017-08-25 06:45:23 75f4f340 InnoDB: Error: Column last_update in table "mysql"."innodb_table_stats" is INT UNSIGNED NOT NULL but should be BINARY(4) NOT NULL (type mismatch).
2017-08-25 06:45:23 75f4f340 InnoDB: Error: Fetch of persistent statistics requested for table "nextcloud"."oc_admin_sections" but the required system tables mysql.innodb_table_stats and mysql.innodb_index_stats are not present or have unexpected structure. Using transient stats instead.
2017-08-25 06:45:23 75f4f340 InnoDB: Error: Column last_update in table "mysql"."innodb_table_stats" is INT UNSIGNED NOT NULL but should be BINARY(4) NOT NULL (type mismatch).
2017-08-25 06:45:23 75f4f340 InnoDB: Error: Fetch of persistent statistics requested for table "nextcloud"."oc_admin_settings" but the required system tables mysql.innodb_table_stats and mysql.innodb_index_stats are not present or have unexpected structure. Using transient stats instead.
2017-08-25 06:45:56 53ab9340 InnoDB: Error: Column last_update in table "mysql"."innodb_table_stats" is INT UNSIGNED NOT NULL but should be BINARY(4) NOT NULL (type mismatch).
2017-08-25 06:48:56 53ab9340 InnoDB: Error: Column last_update in table "mysql"."innodb_table_stats" is INT UNSIGNED NOT NULL but should be BINARY(4) NOT NULL (type mismatch).

```
</details>

nachoparker, 2017-08-25:
what image are you doing this on? did you download the latest image?

RyanTG, 2017-08-25:
I upgraded via dist-upgrade. 

````

nachoparker, 2017-08-25:
Ok, then first, you should try out the web interface. See [this post](https://ownyourbits.com/2017/07/24/nextcloupi-gets-a-web-interface/) for the copypaste code to install it.

That's unrelated to your problem however... let'sencrypt is failing to install. That's weird.

You can try to install it from zero again

```
sudo rm -rf /etc/letsencrypt /usr/local/etc/nextcloudpi-config.d/letsencrypt.sh
sudo ncp-update
```

Also, being letsencrypt related, maybe you can look in the letsencrypt forums

RyanTG, 2017-08-25:
Thanks. I was able to install letsencrypt via your suggestion. 

(I get an error when I run it, but I believe that's a port issue. I use a vpn, and they assign my port, so that's what I'm using. I'll try opening 80 and 443 also. I know this is a separate issue..)

The web interface install failed at the last step.
````Aug 25 07:18:22 raspberrypi systemd[1]: Reloading The Apache HTTP Server.
Aug 25 07:18:23 raspberrypi apachectl[20867]: apache2: Syntax error on line 147 of /etc/apache2/a
Aug 25 07:18:23 raspberrypi apachectl[20867]: Action 'graceful' failed.
Aug 25 07:18:23 raspberrypi apachectl[20867]: The Apache error log may have more information.
Aug 25 07:18:23 raspberrypi systemd[1]: apache2.service: Control process exited, code=exited stat
Aug 25 07:18:23 raspberrypi systemd[1]: Reload failed for The Apache HTTP Server.
````

RyanTG, 2017-08-25:
Ah, line 9 is `IncludeOptional /etc/modsecurity/*.conf` 
And I saw that there's an issue with modsecurity right now. Guess it's related.

I'll close this. Bedtime.

closed 2017-08-25 by RyanTG