nextcloud/nextcloudpi #158: Is there something similar to nextcloudpi on the internet?

ObfuscateTheWeb, 2017-08-25:
duckduckgoing around, the first result that comes up is nextcloudpi, doesn't show anything else 
I'm sure there's something out on the internet that is similar to nextcloudpi but is 64-bit?

If there exist such thing, it means I could use that and get unlimited upload sizes to nextcloud without the 32-bit php and 32-bit raspbian limiting me 

@nachoparker  I'm pretty sure you know of such thing?

nachoparker, 2017-08-25:
Well, don't panic!

Even for 32bit systems, there's a workaround to the PHP limitation.

It is scheduled to land in NC13, but if you can patch it yourself today

https://github.com/nextcloud/server/issues/1707#issuecomment-291835214

Also, you can use NFS, SSH or SAMBA to upload the big files. See the related posts in own your bits. There's one called 'different ways to access your nextcloud files'

In any case, NCP will probably end up being ported to 64bit, but it's a bit lower priority

ObfuscateTheWeb, 2017-08-26:
Do I just edit according to that thread? 
I followed the edit and rebooted but it still doesn't work?

Am I suppose to recompile it or something? Is it as simple as edit,save, and reboot? Cause it isn't working for me :(

nachoparker, 2017-08-28:
No, no need to recompile, it should be instantaneous

That's the file. I don't remember if you have to make other server config changes.

ObfuscateTheWeb, 2017-08-29:
Is the edit suppose to fix the upload/download sizes ?

nachoparker, 2017-08-29:
Hi, 

Thanks for the help on this.

The problem is only with uploads I think.

that looks about right

Maybe there's some server configuration that you want to review, but I think everything is correct so it surprises me that still doesn't work for you. I assume you have rebooted, just in case?

I was waiting for NC13, because applying the patch as part of the build process is possible but messy, and there's more guarantee that it's going to work after upstream has released it, but if there's people interested in making it work it would be nice, to be ready for when it lands, or maybe implement it as a `nextcloudpi-config` extra.

I'll paste here some resources. Maybe somebody else can help review this. I need to address some bugs that came with Stretch first

https://devsidestory.com/build-a-64-bit-kernel-for-your-raspberry-pi-3/
https://github.com/fruux/sabre-http/pull/74
https://github.com/nextcloud/server/issues/1707#issuecomment-291835214
https://forum.owncloud.org/viewtopic.php?f=17&t=32517
https://github.com/owncloud/core/issues/9832#issuecomment-148380791
https://docs.nextcloud.com/server/11/admin_manual/configuration_files/big_file_upload_configuration.html

ObfuscateTheWeb, 2017-08-29:
Yes I rebooted, and I guess I'll just have to wait for NC 13 to patch this

nachoparker, 2017-08-30:
Ok, I'll leave the help wanted tag if anybody want to try this. In that case we can reopen

closed 2017-08-30 by nachoparker

gboudreau, 2017-12-25:
In case anyone else finds this looking for a workaround, installing the Nextcloud 13 beta (beta 3 was published a few days ago) resolves the upload issues. I was able to upload 8-9GB files using the web UI with that beta installed.

nachoparker, 2017-12-25:
@gboudreau awesome, thanks for letting us know

applecargo, 2018-01-03:
Following @gboudreau, I did also upgrade to NC13 beta 3 (from NC12 latest stable). Apparently it works somewhat 'better', but no success. :(

Error message:
Error while copying file to target location (copied bytes: 2147483647, expected filesize: -1 )

I am using Odroid XU4 / Archlinux which is 32bit machine.
and 2147483647 == 2^32-1

What should I look into now? How to start to fix this?
Is it achievable with 32bit machine, BTW?

maanloper, 2018-04-02:
Sorry to reopen this:
I have RPI3 running NC13.0.1 with a variation of the problem:

When I upload a 5.0GB MOV file through the web interface while normally logged in everything seems to go smooth(ish).* I used microsoft FC /B (binary file compare) command to check the original and uploaded-and-synced-back-to-client versions were identical bit-to-bit.

However: when I create a share link for a folder for someone to upload their files to, say for a friend with GoPro footage of my snowboard crashes, then the file gets uploaded "successfully", but **truncated at 2.0GB**.
Example: Movies/Crashes/Myfriend
Create share link for Myfriend folder (no password, expire date >2 weeks away, both upload&edit capabilities)
If I upload from logged in account to that folder (so not using the share link), no problem; me, the browser, the server: all happy. If either me or my friend uploads a >2GB file through the shared link: 2.0 GB truncated file. No error, nothing. Just a 2.0 GB files synced back to my desktop through the client.

My guess: the share-link page uses different code than the normal web UI and does not have the updated code in it that handles >2GB files correctly?

*Sometimes it keeps saying "a few seconds" for a looooong time when uploading is done (no data upload according to Windows Task Manager). Then after a while desktop client starts syncing the file while the web environment still says "a few seconds". When pressing F5, file appears in the list without errors, but otherwise "a few seconds" won't disappear. This problem does not occur when the files are truncated, then the upload bar disappears when both are uploaded and truncated.

nachoparker, 2018-04-02:
> My guess: the share-link page uses different code than the normal web UI and does not have the updated code in it that handles >2GB files correctly?

Your guess sounds correct to me. Better report this in the main NC repo, or in the forums

maanloper, 2018-04-03:
No problem, I have created the thread also in the main Nextcloud repo, [#9053](https://github.com/nextcloud/server/issues/9053).

At first I thought my friend was completely retarded; I could upload big files and he couldn't. Guess I owe him a beer now ;)

nachoparker, 2018-04-03:
lol I guess you are right xD

keep us posted

nachoparker, 2018-04-03:
mmm, if anything I would create a new issue, and close it as a external bug, so let's just leave the issue you created linked here for people investigating this in the wrong place

nachoparker, 2018-08-17:
This was fixed with the update to NC13

See https://ownyourbits.com/2018/02/06/nextcloudpi-updated-to-nc13-brings-ufw-security-improvements-auto-snapshots-and-more/

stfast, 2019-06-17:
The problem is tmpfs size for PHP-FPM. Since user is anonymus over shared link, temprary upload file isn't written  to Nextcloud user directory (over PHP session), but is written to system tmpfs (/tmp) which is 2Gb by default, thus upload file over shared link being truncated: 

[root@nextcloud ~]# df -h|grep tmp
tmpfs           2.0G     0  2.0G   0% /dev/shm

Our soution is to change PHP-FPM tmpdir settings:

[root@nextcloud ~]# cat /etc/php-fpm.d/custom.conf
[www]
env[TMP] = /Nextcloud_Data/upload_tmp_dir
env[TMPDIR] = /Nextcloud_Data/upload_tmp_dir
env[TEMP] = /Nextcloud_Data/upload_tmp_dir

[root@nextcloud ~]# systemctl restart php-fpm

Another possible solution would be tmpfs resize, but actually this is not a good idea, since tmpfs is really RAM allocation/comsumption.

https://wiki.archlinux.org/index.php/Tmpfs

delfer, 2019-10-24:
Fixed by docker-library/php#901
Tested in Docker image `nextcloud:17-apache` (`nextcloud@sha256:58836d449406b396c1c119c2661577fd7d8284ff41a7d181a0e2be232d2757fc`, `2019-10-24T05:04:49.946856094Z`)

CajonCP4, 2020-11-28:
I tried to upload a video file >5GB via Desktop client to my cloud, but it seems not to work. It says it has uploaded the file, but only 1,3GB got uploaded. I have the Pi4 model on NextCloudPi 19.0.2 and I have set the maximum file size to 1000GBs.
Do I need additional apps like [Flow Upload ](https://apps.nextcloud.com/apps/flowupload) to upload files bigger then 2GBs? Or am I doing something wrong? Thanks for your help!

nachoparker, 2020-11-29:
You will have to migrate to the 64bit version. This week we'll have raspberry pi NCP image that supports 64bits

Vergossen, 2020-11-30:
> 
> 
> You will have to migrate to the 64bit version. This week we'll have raspberry pi NCP image that supports 64bits

Thats wonderful, because yesterday I found that I have the same problem. I have a Raspberry Pi 4 with NCP and I could't upload a mp4-File with 2,3GB.  

If I understand you correctly, I can download the image and then I have a 64 bit NCP system on a Raspberry Pi 4, which actually only has 32 bit, without the 2GB limitation? That would be great! :-) 

Where do we find out when the new version is ready and where can we download it? Is there a manual for people who are not so familiar with Linux? 

Many greetings
David

nachoparker, 2020-11-30:
Raspberry Pi 4 is 64 bits, but Raspbian is 32bits. We are moving to a different OS that is 64 bits.

you can join the Telegram group to get a notification, also check out docs.nextcloudpi.com

Larsene, 2020-11-30:
You can use the RaspiOS 64. It works perfectly

[RaspiOS lite 64bits](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/)

nachoparker, 2020-11-30:
the latest image uses RaspiOS 64 bits. It's already downloadable

Vergossen, 2020-11-30:
Is also the Nextcloud Installation included in this Image?  Or must I manually install nextcloud pi? 
    
Regards David

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/nextcloud/nextcloudpi/issues/158#issuecomment-735865775",
"url": "https://github.com/nextcloud/nextcloudpi/issues/158#issuecomment-735865775",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Vergossen, 2020-11-30:
I hope that I find a tutorial how to transfer the actual Database to the new OS and the New NCP-installation without loss of Data.

ovpc, 2020-11-30:
@Vergossen 

https://docs.nextcloudpi.com/en/how-to-backup-and-restore-a-nextcloudpi-instance-using-ncp-config/

> RaspiOS lite 64bits

does not boot on my rpi3B, is that to be expected?

nachoparker, 2020-11-30:
https://ownyourbits.com/downloads/

CajonCP4, 2020-12-01:
@nachoparker Thank you very much! Installed the new update and everything worked fine!

ovpc, 2020-12-01:
https://ownyourbits.com/downloads/NextCloudPi_RPi_11-27-20/
does not boot on my rpi2B (armv7), is that to be expected?

Larsene, 2020-12-01:
> does not boot on my rpi2B (armv7), is that to be expected?

yes. rpi2B cards are not ARM64.