nextcloud/nextcloudpi #159: some kind of dashboard would be fine

hurrikan77, 2017-08-25:
Hi 

I use the nextcloudpi and the webinterface.
The usage of the webinterface is a little bit diffcult.
Especialy for these options where you could chnage the path of something (e.g. backup, data dir, ..)

It would be good to have some kind of dashboard where you could see the actual settings.

Best regards

hurrikan

nachoparker, 2017-08-25:
Hi,

Could you explain a little bit what you mean. what are 'actual settings'?

If you write something in a field the web interface will remember it, so those are the current settings. Is that what you mean?

Thanks for the feedback

hurrikan77, 2017-08-28:
Hi

thats good to  knwo that the webinterface remember the entrys. But if you set up the nextcloudpi the first time there are entrys as well and I could not see have I set them or not.
It is good that there are inital entrys because for me its not easy to find the correct path to an external USB drive or something.

And that is some thing I want to see. Which external drives are mounted? What path did I set for the backup funktionality? When was the last automated backup? Was it suchessfull?
There are other things you could show on a dashboard.

Best regards

Hurrikan

nachoparker, 2017-08-28:
> for me its not easy to find the correct path to an external USB drive or something.

A filepicker is in the oven for that ;)

> And that is some thing I want to see. Which external drives are mounted? What path did I set for the backup funktionality? When was the last automated backup? Was it suchessfull?
There are other things you could show on a dashboard.

Got it, thanks for the ideas!

> What path did I set for the backup funktionality?

That one, you can see because it remember what you wrote last time

hurrikan77, 2017-08-29:
>What path did I set for the backup funktionality?

>That one, you can see because it remember what you wrote last time

Important here and for other functionalities as well is "Have I activated them initialy?". But may be if the filepicker comes out of the oven ;-) then the inital value could be empty and changed during inital activation into an user selected value.

One more thought about the dashboard. I could be a possibility for a litte bit of sorting of the functionality. I know your development of the webinterface has just started and this is not the first priority. 

Seeing forward for the next versions. ;-)

nachoparker, 2017-08-29:
> a litte bit of sorting of the functionality. 

do you mean ordering the entries on the left?

> Have I activated them initialy?". But may be if the filepicker comes out of the oven ;-) then the inital value could be empty and changed during inital activation into an user selected value.

Do you think that is less confusing than having a default value? most of the time you can just use the default.

Or maybe we should have two types: those that will normally be the default, such as `/media/USBdrive`, and others that won't, like `mydomain.ownyourbits.com`, and have them behave differently

hurrikan77, 2017-08-29:
> Do you think that is less confusing than having a default value? most of the time you can just use the default.
Yes most of the time I use the default. But I do not see have I used (activated) this function for the first time.

nachoparker, 2017-08-29:
Ok, gotcha

Yes, I have something in the backlog for that but it has been on low priority

hurrikan77, 2017-08-29:
Good to her.
I know that this is more a nice to have but its good to know it is registered.

Looking forward to every improvment of this great nexcloud image.

nachoparker, 2017-11-28:
we can extract the info from `nc-info` and show it in the landing page for ncp-web. We now have the material to be presented

nachoparker, 2018-02-13:
@theCalcaholic now that I am getting things out of the way, I would like to implement this soon... just do some AJAX call and get the contents from `nc-info` in the background to show in the landing page, maybe also add an icon to the top (next to the wizard icon) if we want to go back to this info.

We can plot the contents from `nc-info` somewhat nicer than on plain text. What do you think? ideas? suggestions?

nachoparker, 2018-04-03:
implemented a simple dashboard in `v0.53.10`

closed 2018-04-03 by nachoparker