nextcloud/nextcloudpi #160: Support ODROID-HC1

renne, 2017-08-25:
The [ODROID-HC1 (Home Cloud One)](http://www.hardkernel.com/main/products/prdt_info.php?g_code=G150229074080) is optimized for NAS-functionality and is much more powerful than a RaspberryPi.

Please support an Nextcloud image for the HC1.

miodzicho, 2017-09-05:
I'm also interested in Nextcloud image on HC-1

nachoparker, 2017-09-05:
There has been some discussion on wether it would be more interesting to do a full port of NextCloudPi or just use the [NextCloudPi ARM container](https://ownyourbits.com/2017/06/08/nextcloudpi-docker-for-raspberry-pi/)

If anybody is interesting in helping with this, we could start by trying the docker image and see how it goes.

Since NextCloudPi moved recently to Raspbian 9, the docker image still has not been updated, so it is a bit older and there is no web access to the NCP extras. Still NC12 though.

miodzicho, 2017-09-05:
Sounds good to me. I'm going to try docker image next week, will report how it works.

nachoparker, 2017-09-05:
great

testbird, 2017-09-17:
Maybe your installer script also works, or can be made to work, in other non-raspi centered debian based embedded distributions like armbian, or the debian based distros from the vendors.

testbird, 2017-09-17:
Make it a debian meta package?

nachoparker, 2017-09-17:
Yes, we will probably take that path.

Conversations started [here](https://help.nextcloud.com/t/best-cheap-hardware-to-run-nextcloud-on/4715/92) about using armbian and porting to other boards.

My odroid HC1 just arrived, and I am updating the docker container to test it on the odroid. Depending on the resources I have I will probably end up porting to Armbian, but supporting and maintaining NextCloudPi already takes a lot of my time.

Also there is the issue that

 - Using arm64 will get rid of the ridiculous 2GB upload size limit. 
 - That limitation has been worked around in NC13, so when it lands the problem will 'solve itself'
 - Raspi 2 is 32bit, so that would mean either stop supporting it or supporting more hardware and we need more people involved in the build process for that. I cannot support several boards, so I am focusing in a docker container that can run in any ARM board.

testbird, 2017-09-17:
Adjusting the ncp scripts to install on openmediavault might also become a strong option now with its new hardware support, at the same time joining with a larger community. (Even making it installable in plain debian and its derivates [1]) 

See
https://forum.openmediavault.org/index.php/Thread/19618-OMV-3-for-ODROID-XU4-HC1-HC2-MC1/
https://forum.armbian.com/index.php?/topic/3953-preview-generate-omv-images-for-sbc-with-armbian/

[1] https://wiki.openmediavault.org/index.php?title=Install_OMV_on_Debian_via_apt

nachoparker, 2017-09-18:
More brainstorming here! awesome

Mmmm I have no idea of how OMV works. Is it based on debian? I would need to have a close look. 

It would maybe be interesting to cooperate with them but, how would that go? would they be interested?

I would also need to have a look at the .deb for OMV. Does it install the whole LAMP stack also? maybe as .deb dependencies, or contained inside?

I was thinking to take the docker container approach to magically have it working on all ARM boards. Thoughts?

Pant, 2017-09-18:
+1 for the docker container approach. It even opens up the way for docker swarm experiments!

testbird, 2017-09-19:
I think many people are looking for a build-system that is able to also produce iso images (amd,intel,...)  as well as for multiple embedded devices and docker.

Openmediavault is a debian based distro, you'd first need to try it out (install, configure) to see if you like what and how they made it. There was once an owncloud plugin, but it was discontinued and an installation manual was posted. When you try it out and have to ask a question, tell them you created nextcloudpi and may be interested in a base system that is more universal than raspbian.

You'd need to know if they would welcome your contributions, say in apache https configurations and you get along well.

OMV has a gui plugin to manage docker images, maybe try to setup your nextcloudpi docker image ontop of the OMV image to get your HC1 running.

Docker is a good option for admins these days. There are also OMV docker images out ("mostly for plugin development". Your strength is knowing the configurations for a ready-to-use, well preconfigured nextcloud image. If you'd like to share the workload of maintaining such a system, your scripts may more easily evolve into an omv script or plugin (collaborating on the underlying server configuration) then a debian package. (Nevertheless, the OMV community seems to work towards making their server configuration installable in plain debian.).

For updates I prefer distribution package managers updating individual files (possibly within a container, used only to run the system on arbitrary, otherwise unsupported platforms). This avoids redundancy and the burden on container maintainers to provide security updates with large container updates (possibly missing from some not so well maintaining container provider).

nachoparker, 2017-09-20:
Thanks for the info, I'll definitely have a look

For updates, I think you can provide a light update if you optimize docker layers. I am still thinking the whole thing, but I would like to reach the point where you can update even across NC versions, like the snap supposedly does. That would be the final goal, but I am still undecided on how to do it exactly.

testbird, 2017-09-20:
Because I had overlooked it for a long time: The important things on https://www.openmediavault.org/ can only be found in the footer...! (very unfortunate)

miodzicho, 2017-09-28:
Hi All. I have received HC1, I tried to  install it without container. But unofrtunatelly facing weird HDD errors, trying to narrow if this is nextcloud related, or hardware. If interested, here is story : https://forum.odroid.com/viewtopic.php?f=97&t=28372&p=203017#p203017

I think will try container now, to see if this issue will happen again.

nachoparker, 2017-09-28:
Ok, thanks for the info. Hopefully I will start playing with the HC1 soon. The container is a bit old ( Nextcloud 12.0.3 ), but it should work fine. Follow the instructions in ownyourbits to move your docker storage to the HDD and keep in mind that files over 2GB will be truncated because of PHP stupid 32bit limitation.

Let us know how it works!

testbird, 2017-10-19:
Update on the base system comparison, that seem muvh more geared towards public self-hosting than openmediavault:

I found https://freedombox.org has become usable, and had not seen https://yunohost.org before.

Freedombox looks as it has become the debian way for self-hosting services now.
* Images for the well suited A20 OLinuXino, but also rpi2 (said to also work for rpi3)
* Installable on plain debian (and derived distros, thus armbian?)
* Nextcloud package does not exist yet, but dev docs provided https://wiki.debian.org/FreedomBox/Manual/Developer
* features automatic updates, fail2ban
* supports self-hosting and libre/free software efforts

Yunohost
* the manual install procedure for arm says to install armbian as the first step
* A nextcloud package already exists
* looks like someone tries to creating an app-store eco-system

nachoparker, 2017-10-19:
Hi,

didn't know about those projects, thanks.

freedombox looks **really** interesting. I will probably try to have a look at it. The approach seems to be to install deb packages.

We have also been looking into armbian to provide 64 bit images for RPi3 and other boards.

While the deb approach is great for regular software, I do not think is ideal for web services. I think that the docker/container approach is more appropriate because you can also manage and load balace the resources, and pack interacting services together in a nicer way ( dnsmasq + apache + mysql combo for example ).

You can as well stop and start the containers easily. You have installation + load balancing + management covered, it's pretty neat.

Maybe there is a nice way to start from a freedombox image and put the docker daemon on top.

nachoparker, 2017-10-19:
In that regard, we could also add 

[linuxserver.io](linuxserver.io)

[picluster](https://github.com/picluster/picluster)

nachoparker, 2017-10-19:
also this, but seems a bit incomplete

https://resin.io/blog/announcing-balena-a-moby-based-container-engine-for-iot/

testbird, 2017-10-19:
> While the deb approach is great for regular software, I do not think is ideal for web services. I think that the docker/container approach is more appropriate because you can also manage and load [...]

I am not sure if the is really necessary, and more importantly, if putting the docker image updating really better. Also don't underestimate the amount of (time critical) workload to keep a docker image current and secure (build, test, signing infrastructure). You would need to compete with the debian security team or always lag behind.

The update strategy you do now with docker image actually looks like another of those intelligent things I've seen you do in ncp. The apt system for libs and things within the docker container where possible, plus ncp scripts where needet to check for and trigger the ncp and official nextcloud updates (php website code).

If you where to build a new image for every update this automated system installs you may be unnecessarily creating a centralised system (like those ubun* things that are becoming increasingly unpopular). Thus, there may be no need to introduce the complexity any of those elaborated docker management systems you mentioned.

Nevertheless, running the public nextcloud in a container has security benefits, and splitting the ncp docker image into a system and a data container may allow straight forward snapshoting and backup. So a simplistic docker based deployment does make sense.

testbird, 2017-10-19:
A downside of docker image updating: They tend to overwrite user customizations, making local fixes or tests unreliable.

testbird, 2017-10-19:
Q: Unfortunately I don't have a HC1 yet, anybody experimented / forked ncp to install on armbian yet?

I think I would try starting with the armbian image and dist-upgrade to stretch. Then I would
* follow https://wiki.debian.org/FreedomBox/Hardware/Debian
* adapt the nextcloudpi scripts by introducing "if freedombox" (or better yet, specific  "if ... is already installed/ configured/available") clauses to the ncp scripts to prevent installing unnecessary system config tools like ip configuration etc, and to adapt paths etc. for the remaining things.
* keep the separate ncp-web for now, and get aquainted with the freedombox webinterface.

testbird, 2017-10-19:
A yunohost downside: The web service package for nextcloud does not trigger official nextcloud updates as they become available, but needs to be updated for every tiny version step. https://github.com/YunoHost-Apps/nextcloud_ynh/commit/8f98d512b17c12b3f6225414c8dd5546ea482307

testbird, 2017-10-20:
Stepping back for a second. We want nextcloud (or web apps in general) to reasily run on own hardware boxes.

Ncp delivers that, but we need more hardware platforms than raspbian. => Conclusion: Make it work in Debian and thus in armbian images.

Ncp is based on plain shell scripts that set up and manage what is needed to run the web app. This allows that users familiar with manual administration can contribute. Using some hype container orchestration would [ruin or deviate that](https://www.dedoimedo.com/computers/software-development-cancer.html).

Ncp consist of cli-tools and web-frontends to configure distro/system (IP, DNS, updates,...), packages (db, webserver,...) and the web app itself (backups, updates,...). It is similar to cpanel and others plus consistent shell tools, presenting things to do for running  the web app. The OS, packages, and web app tools might be able to work on all debian systems. A docker build allows for other systems and/or containment, but not more in terms of running nextcloud, or any other web app.

There is no web application manager in debian, ncp currently is a nextcloud application manager for the pi (raspbian at this time), but apparently we are now reaching out for hardware beyond the pi.

Iolaum, 2017-10-21:
As a user of Nextcloudpi on a rpi2b nextcloudbox I 'll jump in for some feedback on this - hope you don't mind.

I appreciate staying as close to the base system as possible. This means that most of the problems a user will be encountering could be solved through `google $error > copy > paste` which will hopefully also mean less support burden for developers here.

I 'm not qualified on comparing the debian meta-package approach versus the docker approach. I 'd like to highlight some security factors into that. Those are the two most important I can think of:

- A frictionless upgrade path.<br>Nextcloud may be the latest version but if the apache web server or the kernel are outdated on an Nextcloudpi connected to the internet it won't be pretty.
- Hardening measures.<br> They are important on anything connected to the internet. Nextcloudpi already has some measures like fail2ban. Hopefully it will be easy to port them and use them on a meta package or container implementation.

P.S. I can see why the web administration feature gains so much traction (familiar and streamlined user experience for non technical people) but hopefully you will not neglect the the `nextcloudpi-config` option over an ssh connection (feature parity wise, looks don't matter in this case imo).

nachoparker, 2017-10-23:
@testbird 

Hi, 

Sorry, I have been away for a few weeks.

There are good points there. The docker image is an interesting experiment, with some very interesting use cases.

> keep a docker image current and secure (build, test, signing infrastructure). You would need to compete with the debian security team or always lag behind.

This would be automated, it is not hard.

> If you where to build a new image for every update this automated system installs you may be unnecessarily creating a centralised system

Docker does not need registration. You can save and sideload images

> elaborated docker management systems

That's something for a different project. The docker image for NCP doesn't need that, but I was thinking about creating a SD card image with one of those and docker already running with the NCP image.

> A downside of docker image updating: They tend to overwrite user customizations, making local fixes or tests unreliable (e.g. my pwauth customization would be reverted as long it was not yet mergered into the image updates).

For sure, if you want to do customizations this solution would not work for you. There are people though that want to install NCP along other things, like pihole, and with containers this can be achieved more easily.

Also it is easy to 'try NC out' because you only have to docker pull which is really easy ;)

> anybody experimented / forked ncp to install on armbian yet?

We are planning to move everything to armbian in order to support ARM64, and other boards. Regarding freedombox management interface... looks really cool. I would have to try it out.

Still we have the problem of different web based apps reconfiguring the database and web server. Containers solve that.

> Ncp is based on plain shell scripts that set up and manage what is needed to run the web app. This allows that users familiar with manual administration can contribute. Using some hype container orchestration would ruin or deviate that.

The SD card version will always be there. It is more powerful, as it can, for instance, configure wifi, DHCP or other things that don't belong to a containerized service.

> Ncp consist of cli-tools and web-frontends to configure distro/system (IP, DNS, updates,...), packages (db, webserver,...) and the web app itself (backups, updates,...). It is similar to cpanel and [...]

I am trying to achieve the best of both worlds... easy sideload of NCP with docker, but it is still based on debian, so the same code builds also the SD card version, people can contribute to the SDcard, and the container would benefit.

Shouldn't be hard to make it work on non Raspbian Debian systems... basically the big difference is the existence of a sudoers allpowerful pi user... I think packages should be almost identical.

The problem we have is that this is really slow because of lack of manpower. I have the boards at home, and won't have time to try them out for weeks because I am working on the wizard right now.

I don't think I can afford to support more boards by myself. We would need an armbian maintainer for this to happen. Probably not too much work, but my hands are full. Help is welcome.

nachoparker, 2017-10-23:
@lolaum

Thanks for your opinions. We need many of those :)

NextCloudPi already automatically gets the latest packages for Apache and the rest. NC docker would have to do something similar, of course. Security and easy upgrading is the most important thing here. If everything was based on docker, the migration to Stretch would have been automatic and painless for the user (that had to do dist-upgrade, and there are still some that remain in Jessie)

> Hardening measures.
They are important on anything connected to the internet. Nextcloudpi already has some measures like fail2ban. Hopefully it will be easy to port them and use them on a meta package or container implementation.

I aggree. I don't see the meta-package approach feasible for that same reason. The strong point of NCP in my opinion is that it comes with a collection of preconfigured utilities that work together. That can be achieved with docker, but not (I think) with .deb's. 

(Nobody mentions modsecurity... is the single most valuable security layer that NCP has).

> I can see why the web administration feature gains so much traction (familiar and streamlined user experience for non technical people) but hopefully you will not neglect the the nextcloudpi-config option over an ssh connection (feature parity wise, looks don't matter in this case imo).

No worries for that. The backend scripts that run all the extras are the same

nachoparker, 2017-10-29:
@testbird 

Looks like somebody already implemented a similar idea, using docker for autoupdates. When I have time I will investigate their solution

https://home-assistant.io/hassio/

testbird, 2017-11-11:
Hey, quick update: it was a piece of cake to get the OS on the HC1 up and running:

* prepare an armbian/stretch (dev version) sdcard image
https://dl.armbian.com/odroidxu4/Debian_stretch_dev_nightly.7z
https://docs.armbian.com/User-Guide_Getting-Started/#how-to-check-download-authenticity
* boot it and login

* The freedombox setup in stretch worked, but is missing newer things, so adapt the /etc/apt/sources.list to buster (testing) release and `apt dist-upgrade` (stay with armbian stretch repository for now)
* package dependencies that I needed to install manually (I installed freedombox on stretch first):
	ipset
	ebtables
* Follow the setup steps for the freedombox debian blend
https://wiki.debian.org/FreedomBox/Hardware/Debian
* go to the webinterface
https://wiki.debian.org/FreedomBox/Manual#FreedomBox.2FManual.2FQuickStart.Quick_Start

Further on, it looks like the nextcloudpi script to setup nextcloud may be made conditional (to skip the install and setup of readily configured debian packages) with `which plinth` (i.e. looking for the freedombox web interface).

testbird, 2017-11-11:
> measures like fail2ban. Hopefully it will be easy to port them and use them on a meta package or container 

I see freedombox already installed fail2ban. Also syncthing, gnudip, and other nice things like matrix/xmpp server are already available.

nachoparker, 2017-11-13:
@testbird thanks for the info! I definitely want to try freedombox, even though I still feel like the deb approach is less powerful to manage complex web systems comprised of several services in coordination.

I am working on building images for the odroid HC1 and the pine64 boards that I own with NCP in them.

In any case, the catch all solution to support other systems right now is the script

```
# curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash
```

but it needs to be on Debian Stretch. For the rest of systems I still think that the docker approach is the most universal. The new docker containers will also come out this week, but it will take some time until they do everything that the SD version does.

Opinions welcome

nachoparker, 2017-11-22:
Well, aside from the curl installer and the docker option, I just created the first version of the SD card for the odroid HC1 / XU4. I consider it testing at this point, but it can already be used.

https://ownyourbits.com/downloads/testing/

Feedback is welcome.

edit:

Let's encrypt didn't build ok this first attempt. We can fix it with

```
sudo rm /usr/local/etc/nextcloudpi-config.d/letsencrypt.sh
sudo  ncp-update
```

closed 2017-11-22 by nachoparker

testbird, 2017-11-23:
Oh, for those that can test:
The armbian-config tool is said to allow installing to sata (HDD/SSD)

testbird, 2017-11-24:
Ok, ncp-web came up fine. 
* NC does not have the local domain configured: nextcloupdi.lan (.lan is a router managed dns zone).
* Setting a static ip failed.
* Could not access SSH on Armbian, and found no SSH config in ncp-web. Did you disable SSH?

Iolaum, 2017-11-25:
In the raspbian nextcloudpi image the ssh is not enabled through an option [but through adding an ssh file at the boot partition](https://github.com/nextcloud/nextcloudpi/wiki/How-to-install-NextCloudPi-on-a-Raspberry-Pi#first-steps). Maybe armbian requires somthing similar?

testbird, 2017-11-25:
I was not successful with that method either. Default Armbian has ssh enabled and requires changing the 1234 password on first login.

nachoparker, 2017-11-25:
hi,

thanks for your feedback!

- a couple things, mainly nextcloudpi.local and the logo on the terminal have not been included in the 'curl installer'. The reason being that maybe you want to install NCP, but not 'create an NCP image', and maybe you have other things intalled and don't want to mess with your current .local domain. 

Thoughts?

 - SSH is default armbian... root/1234, did not disable it. In contrast to Raspbian, it comes enabled by default and it has you change it upon first login.

Isn't it fast? <3<3<3 and still haven't optimized specifically

- static IP-> did you try through `nc-static-IP`?

testbird, 2017-11-25:
Strange that I have no ssh then. Maybe it's iptables related: Read from socked failed: Connection reset by peer.
Accordingly, I could only try the ncp-web ip setting form, it said "could not find dhcpd.service" or simmilar.

I like your careful attitude towards changing existing systems. Yes, the .local (mDNS) should be configured by Armbian/FBX. May be worth a note in the install guide.

But I can't see if the ".lan" search domain is in /etc/resolv.conf, yet (no ssh) I thought the resolv.conf would be evaluated for NC's secure domains' configuration in newer NCP versions.

Can't see the speed from just the NC error page, yet. :-)

nachoparker, 2017-11-27:
It's strange, I burn the SD card with the image, turn it on, and boom, I have SSH. I wonder why you don't

> Can't see the speed from just the NC error page, yet. :-)

What error page are you seeing? you can't use Nextcloud either?

testbird, 2017-11-27:
Ok, re-wrote the image, and noticed the ethernet hub was hanging :-(, after restarting both I could access ssh. (Ok, that issue seemed not to be related to NCP.)

What's still strage with NCP is that I was not asked for a new root password to replace the 1234 default, as with stock armbian. The login went right into creating a new user account.

With nextcloud I got the "no trusted domain" error page. But retrying with the IP, hey I am very happy with the speed (ok, no data there yet) but finally it gives the user the impression that that thing is working right!

nachoparker, 2017-11-27:
great! 

Regarding the password issue: you are right, I want to restore it to armbian defaults, but I had to do that for the thing to build. See [here](https://forum.armbian.com/topic/5729-nextcloudpi-on-armbian-tested-odroid-hc1/). That is on the list.

> With nextcloud I got the "no trusted domain" error page

What did you try?

There's a lot of polish to do for sure, it will be slow unless I find some help

thanks for the info

testbird, 2017-11-27:
Hi, thanks for the link, and really also your other articles!
Just a thought, aren't you using the word "we" your writings maybe a small litle bit questionable or better improvable?

What I tried was "nextcloudpi.lan" (router/dhcp managed DNS), which works with port 4443 and ssh (now ;-)

Was anybody successful in installing docker on the HC1 on the stretch or buster release? I myself could not find a docker.io package in the repos.

testbird, 2017-11-27:
Now I checked, /etc/resolv.conf contains my search domain "lan" in a line ' search lan'. Is this already supposed to be parsed with the commit you made and adding `<hostname>.<search domain>' to the list of NC's trusted domains?

nachoparker, 2017-11-28:
welcome! ;)

> Just a thought, aren't you using the word "we" in your writings maybe a small litle bit questionable or better improvable?

What do you mean? don't follow :P 

> Was anybody successful in installing docker on the HC1 on the stretch or buster release? I myself could not find a docker.io package in the repos.

I am using docker on the HC1 right now. You have to install docker first. See [here](https://ownyourbits.com/2017/06/08/nextcloudpi-docker-for-raspberry-pi/) and [here](https://ownyourbits.com/2017/11/15/nextcloudpi-dockers-for-x86-and-arm/)

> Now I checked, /etc/resolv.conf contains my search domain "lan" in a line saying search lan. Is this already supposed to be parsed with the commit you made and adding <hostname>.<search domain> to the list of NC's trusted domains?

Mmm no, I don't think I add anything other than nextcloudpi.local (and the IPs and domains from ncp-web)

testbird, 2017-11-28:
Ok, i think the `sudo apt-get install docker.io docker-composet ` as in your second link can only work on oldstable (jessie) and unstable. There is currently no docker.io package in stable and testing.

For me on the HC1 the Armbian/stretch even the get.docker script had first failed, the docker-ce package stayed in an unconfigured state and service restarting failed. Strangely, after a reboot, now a service restart worked and I could successfully reinstall the package (configuraiton now completed).

testbird, 2017-11-28:
With regards to the trusted domains, I had thought it would be good to have at least these included by default:

$IP
$HOSTNAME (local and shorthand of the following)
$HOSTNAME.local (mDNS)
$HOSTNAME.$RESOLV_SEARCH_DOMAIN (local, manual or dhcp configured regular DNS)
$NCP_CONFIGURED_DOMAINS

testbird, 2017-11-28:
That could be `hostname -I`,  `hostname -A` + .local and ncp configured domains.

nachoparker, 2017-11-28:
domains -> aggree

docker -> I have installed it on odroid HC1 armbian stretch with

```
# curl -sSL get.docker.com | sh
```

testbird, 2017-11-28:
Well, with that script I needed to do a reboot on the armbian dev (stretch) before it worked. I did not try the ncp docker on the ncp image. ;-)

nachoparker, 2017-12-03:
hey @testbird, are you still using NCP on the HC1?

if you are interested we might use your opinion on the dev group, because I don't think anyone else there is using anything other than the Pi ATM

Send me an email if interested

testbird, 2018-01-18:
Has been a while. Of course, you can use the posts I made here.

Found it really helpful you enabled NCP to run on debian/armbian, as it also allows for other higher performance boards, like with 3xGbit, mPCIe / multi sata, and USB3 for backup (espressobin).

Related projects of interest seen meanwhile: https://ubos.net/ and
https://www.indidea.org/gael/blog/leaving-apple-google-eelo-odyssey-introduction/

niladam, 2018-01-18:
I'm chipping in here too. I'm currently researching the odroid. Will order it soon and probably I'll be able to help too. This comment is mostly so that I can get the updates until I'll get a chance to help out.

testbird, 2018-01-23:
The Fredombox project switched their issue tracking to the new debian site. So if anybody wants to continue to get updated on the former github issue that was linked here, you'd need to register at the debian site:
https://salsa.debian.org/freedombox-team/plinth/issues/1129

---
Sent with my mobile E-Mail Messenger: https://github.com/deltachat

nachoparker, 2018-02-05:
@niladam  great!!

well, the nextcloudpi telegram group is now open, so just join there if anyone is interested