nextcloud/nextcloudpi #161: Correct config fstab (automount)

matzel82, 2017-08-27:
I Need please the correct config for fstab.

I have only plug in 1 Drive (sda1)

After a reboot the mountpoint (USBdrive1...) is changed.

How do it to config correct ?

matzel82, 2017-08-27:
Automount is switch on at config-tool in nc.

matzel82, 2017-08-27:
pi@raspberrypi:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 931.5G  0 disk 
└─sda1        8:1    0 931.5G  0 part /media/USBdrive1
mmcblk0     179:0    0  14.9G  0 disk 
├─mmcblk0p2 179:2    0  14.8G  0 part /
└─mmcblk0p1 179:1    0  41.7M  0 part /boot

matzel82, 2017-08-27:
proc /proc proc defaults 0 0
PARTUUID=32ebf657-01 /boot vfat defaults 0 2
PARTUUID=32ebf657-02 / ext4 defaults,noatime 0 1
/dev/sda1 /media/USBdrive1 auto defaults,noatime,auto,nofail 0 2
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

# Rules for automounting both at boot and upon USB plugin. Rely on udev rules
/dev/USBdrive  /media/USBdrive         auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive1 /media/USBdrive1        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive2 /media/USBdrive2        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive3 /media/USBdrive3        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive4 /media/USBdrive4        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive5 /media/USBdrive5        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive6 /media/USBdrive6        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive7 /media/USBdrive7        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive8 /media/USBdrive8        auto    defaults,noatime,auto,nofail    0       2

matzel82, 2017-08-27:
each time after reboot or loose the energy it must be USBdrive1

JimmyKater, 2017-08-27:
could that be something similar as on #109? symbolic link to usb drives not working reliably
i know that automount wasn't working at its best for some time but i thought @nachoparker had solved it. at least made it better.
i ahd some problems with it, myself. and then i used the UID-label in fstab... since then it works without any more problems for me.

nachoparker, 2017-08-28:
Hi,

I was away for some days

I fixed things in #109, but then when updating to Stretch it broke again #142 

AFAIK the only problem is duplicated links, but it was working. Still have to try it myself

@matzel82 I don't understand very well your problem. Can you explain in detail please?

thanks @JimmyKater  ;)

nachoparker, 2017-08-28:
In any case, if you want to do it manually, you can add to fstab

```
/dev/sda1      /media/USBdrive ext4 defaults,noatime,norelatime,async     00
```

matzel82, 2017-08-28:
I have a hard drive attached (sda1 USBdrive1). A new mount point is given to the hard drive every time ", but I want the hard drive to be assigned the same mount point for each restart.
Automount is activated in the Config.

matzel82, 2017-08-28:
Now i have it configured in fstab with the UUID... at Time it work, but automount sounds better, the cause if i attached a another Drive.

nachoparker, 2017-08-28:
Sure, this is a priority. It broke in Stretch and I didn't see it

nachoparker, 2017-08-28:
you can follow #134 to check when it is fixed

closed 2017-08-28 by nachoparker