nextcloud/nextcloudpi #166: restore backup error

notAstereo, 2017-08-30:
Hello everyone,

I tried to install Nextcloud on my pi but did not manage it to work. With nextcloudPi it was quit easy. So thank you very much for this.
But now i encountered a problem restoring a backup. i get 
<details>
Launching nc-restore
INFO: overwriting old instance
restore files...
restore database...
no datadir found in backup
Nextcloud is in maintenance mode - no app have been loaded

Maintenance mode disabled

Scanning files for 4 users
Starting scan for user 1 out of 4 (admin)
Starting scan for user 2 out of 4 (person1)
Home storage for user person1 not writable
Make sure you're running the scan command only as the user the web server runs as
Starting scan for user 3 out of 4 (person2)
Home storage for user person2 not writable
Make sure you're running the scan command only as the user the web server runs as
Starting scan for user 4 out of 4 (person3)
Home storage for user person3 not writable
Make sure you're running the scan command only as the user the web server runs as

+---------+-------+--------------+
| Folders | Files | Elapsed time  #|
+---------+-------+--------------+
| 2       | 0     | 00:00:00     |
+---------+-------+--------------+
Done. Press any key...
</details>

There is a good chance i messed up the system while i tried to install GitLab. Before that everything worked fine for me. Is it still possible to use this backup or should i reinstall the image? 
The actual data should be save at a USB device as well as the database.
Please tell me if i can provide additional information to solve this problem.

### What version of NextCloudPi are you using? ( eg: v0.17.2 )
NextCloudPi 0.24.9
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
NextCloudPi_06-28-17
### Include logs
<details>
<summary>Nextcloud logs</summary>
The web interface does not work
and /var/www/nextcloud/data/nextcloud.log does not exist.
</details>
<details>
<summary>Apache logs</summary>

Paste the output of `systemctl status apache2`
 apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2017-08-27 22:19:50 CEST; 2 days ago
  Process: 614 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

Aug 27 22:19:48 raspberrypi3 systemd[1]: Starting The Apache HTTP Server...
Aug 27 22:19:50 raspberrypi3 apachectl[614]: AH00558: apache2: Could not reliably determine the server's fully qua
Aug 27 22:19:50 raspberrypi3 apachectl[614]: (2)No such file or directory: AH02291: Cannot access directory '/var/
Aug 27 22:19:50 raspberrypi3 apachectl[614]: AH00014: Configuration check failed
Aug 27 22:19:50 raspberrypi3 apachectl[614]: Action 'start' failed.
Aug 27 22:19:50 raspberrypi3 apachectl[614]: The Apache error log may have more information.
Aug 27 22:19:50 raspberrypi3 systemd[1]: apache2.service: Control process exited, code=exited status=1
Aug 27 22:19:50 raspberrypi3 systemd[1]: Failed to start The Apache HTTP Server.
Aug 27 22:19:50 raspberrypi3 systemd[1]: apache2.service: Unit entered failed state.
Aug 27 22:19:50 raspberrypi3 systemd[1]: apache2.service: Failed with result 'exit-code'.
Paste the output of `tail -n 100 /var/log/apache2/*.log`
tail: cannot open '/var/log/apache2/*.log' for reading: No such file or directory
</details>
<details>
<summary>mariaDB logs</summary>

Paste the output of `systemctl status mysqld`
mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2017-08-27 22:19:54 CEST; 2 days ago
  Process: 1113 ExecStartPost=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/
  Process: 1098 ExecStartPost=/etc/mysql/debian-start (code=exited, status=0/SUCCESS)
  Process: 644 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recover
  Process: 593 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SU
  Process: 562 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUC
 Main PID: 835 (mysqld)
   Status: "Taking your SQL requests now..."
   CGroup: /system.slice/mariadb.service
           └─835 /usr/sbin/mysqld

Aug 27 22:19:52 raspberrypi3 mysqld[835]: 2017-08-27 22:19:52 1996369920 [Note] InnoDB: Highest supported file for
Aug 27 22:19:53 raspberrypi3 mysqld[835]: 2017-08-27 22:19:53 1996369920 [Note] InnoDB: 128 rollback segment(s) ar
Aug 27 22:19:53 raspberrypi3 mysqld[835]: 2017-08-27 22:19:53 1996369920 [Note] InnoDB: Waiting for purge to start
Aug 27 22:19:53 raspberrypi3 mysqld[835]: 2017-08-27 22:19:53 1996369920 [Note] InnoDB:  Percona XtraDB (http://ww
Aug 27 22:19:53 raspberrypi3 mysqld[835]: 2017-08-27 22:19:53 1421148992 [Note] InnoDB: Dumping buffer pool(s) not
Aug 27 22:19:53 raspberrypi3 mysqld[835]: 2017-08-27 22:19:53 1996369920 [Note] Plugin 'FEEDBACK' is disabled.
Aug 27 22:19:53 raspberrypi3 mysqld[835]: 2017-08-27 22:19:53 1996369920 [Note] Server socket created on IP: '127.
Aug 27 22:19:54 raspberrypi3 mysqld[835]: 2017-08-27 22:19:54 1996369920 [Note] /usr/sbin/mysqld: ready for connec
Aug 27 22:19:54 raspberrypi3 mysqld[835]: Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.soc
Aug 27 22:19:54 raspberrypi3 systemd[1]: Started MariaDB database server.

Paste the output of `tail -n 100 /var/log/mysql/*.log`
tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory
</details>

nachoparker, 2017-08-31:
Oh, I guess we have found something. You have several users and you did not select `INCLUDEDATADIR` in backup.

Is that right?

Does Nextcloud work, though?

Thanks for reporting

notAstereo, 2017-08-31:
Thanks for the reply.
I do not remember a INCLUDEDATADIR option when was doing the backup. In the current version there is only DESTDIR INCLUDEDATA BACKUPLIMIT.

No, Nextcloud does not work. I'm just able to use SSH connection and run the nextcloudpi-config tool. 
Yes, i have the datadir on a usb stick at the raspberrypi.
Do you think it is still possible to use this backup?

nachoparker, 2017-08-31:
INCLUDEDATA is what I meant, I think you didn't select that

In order to restore your backup (until this gets fixed) you can point your datadir to your old one, in `/var/www/nextcloud/config/config.php`

Then, rescan your files with

```
sudo ncp-scan
```

notAstereo, 2017-09-03:
I set the path in config.php and scanned. I seems to work. See here the output:

`pi@raspberrypi3:~ $ sudo nano /var/www/nextcloud/config/config.php
pi@raspberrypi3:~ $ sudo ncp-scan

Scanning files for 4 users
Starting scan for user 1 out of 4 (admin)
Starting scan for user 2 out of 4 (person1)
Starting scan for user 3 out of 4 (person2)
Starting scan for user 4 out of 4 (person3)

+---------+-------+--------------+
| Folders | Files | Elapsed time |
+---------+-------+--------------+
| 94      | 8827  | 00:09:23     |
+---------+-------+--------------+`

Unfortunately i am still not able to connect to my nextcloud via browser.

nachoparker, 2017-09-03:
did you change the apache config trying to install gitlab?

output of ?
```
systemctl status apache
```

notAstereo, 2017-09-03:
i did not change the apache config by hand. Is it possible that GitLab applied changes?
systemctl status apache gives me:
` Unit apache.service could not be found.`

Surprisingly, since i got another result when i posted this question...

nachoparker, 2017-09-03:
sorry, is `systemctl status apache2`

notAstereo, 2017-09-03:
thanks, here it is:

` apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2017-09-03 22:01:32 CEST; 46min ago
  Process: 616 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

Sep 03 22:01:30 raspberrypi3 systemd[1]: Starting The Apache HTTP Server...
Sep 03 22:01:32 raspberrypi3 apachectl[616]: (2)No such file or directory: AH02291: Cannot access directory '/var/log/apache2/' for main error log
Sep 03 22:01:32 raspberrypi3 apachectl[616]: AH00014: Configuration check failed
Sep 03 22:01:32 raspberrypi3 apachectl[616]: Action 'start' failed.
Sep 03 22:01:32 raspberrypi3 apachectl[616]: The Apache error log may have more information.
Sep 03 22:01:32 raspberrypi3 systemd[1]: apache2.service: Control process exited, code=exited status=1
Sep 03 22:01:32 raspberrypi3 systemd[1]: Failed to start The Apache HTTP Server.
Sep 03 22:01:32 raspberrypi3 systemd[1]: apache2.service: Unit entered failed state.
Sep 03 22:01:32 raspberrypi3 systemd[1]: apache2.service: Failed with result 'exit-code'.

`

nachoparker, 2017-09-03:
somehow you deleted the log folder
```
mkdir /var/log/apache2
```
note: use three backticks to encapsulate code. Like this (without spaces)

` ` `
code
` ` `

notAstereo, 2017-09-03:
okay i tried it again
`systemctl status apache2`
gives me now
 ```
apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2017-09-03 23:02:03 CEST; 6min ago
  Process: 602 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

Sep 03 23:02:00 raspberrypi3 systemd[1]: Starting The Apache HTTP Server...
Sep 03 23:02:03 raspberrypi3 apachectl[602]: AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive globally to suppress this
Sep 03 23:02:03 raspberrypi3 apachectl[602]: (2)No such file or directory: AH02291: Cannot access directory '/var/log/apache2/' for main error log
Sep 03 23:02:03 raspberrypi3 apachectl[602]: AH00014: Configuration check failed
Sep 03 23:02:03 raspberrypi3 apachectl[602]: Action 'start' failed.
Sep 03 23:02:03 raspberrypi3 apachectl[602]: The Apache error log may have more information.
Sep 03 23:02:03 raspberrypi3 systemd[1]: apache2.service: Control process exited, code=exited status=1
Sep 03 23:02:03 raspberrypi3 systemd[1]: Failed to start The Apache HTTP Server.
Sep 03 23:02:03 raspberrypi3 systemd[1]: apache2.service: Unit entered failed state.
Sep 03 23:02:03 raspberrypi3 systemd[1]: apache2.service: Failed with result 'exit-code'.
```
The log file says that it cannot access `/var/log/apache2/` but it exists now and is empty
```
pi@raspberrypi3:~ $ cd /var/log/apache2
pi@raspberrypi3:/var/log/apache2 $ ls

```

nachoparker, 2017-09-03:
permissions?

```
ls -ld /var/log/apache2
ls -l   /var/log/apache2
```

notAstereo, 2017-09-03:
`ls -ld /var/log/apache2`
`drwxr-xr-x 2 root root 40 Sep  3 23:09 /var/log/apache2`
and
`ls -l   /var/log/apache2`
`total 0`

nachoparker, 2017-09-03:
weird, it's empty, I really don't know what you have done lol, maybe just install --reinstall apache will be easier

notAstereo, 2017-09-03:
Yes ok. Thank you a lot for your time.

nachoparker, 2017-09-04:
Ok, I tested backups with several users without INCLUDEDATA, and it works fine here

```
restore files...
restore database...
no datadir found in backup
Nextcloud is in maintenance mode - no app have been loaded

Maintenance mode disabled

Scanning files for 3 users
Starting scan for user 1 out of 3 (admin)
Starting scan for user 2 out of 3 (test1)
Starting scan for user 3 out of 3 (test2)

+---------+-------+--------------+
| Folders | Files | Elapsed time |
+---------+-------+--------------+
| 10      | 14    | 00:00:13     |
+---------+-------+--------------+
```

But it was worth checking. Thanks for feedback

closed 2017-09-04 by nachoparker