nextcloud/nextcloudpi #167: Install XMPP chat server for chat app

nachoparker, 2017-08-31:
so the chat app can have federated chat access

testbird, 2017-10-19:
Ha, ha, missed this issue. Thats how I came to try out making a sdcard from freedombox.org

nachoparker, 2017-10-19:
we're undecided wether this would be useful/popular to allocate resources to. Opinions are welcome.

deljones, 2017-10-19:
For me the nextcloudpi has been excellent, I use it every day. I use all of the core service from nextcloud so adding chat, whilst useful for some users, can only eat away at the limited and finite resources of the pi and I would not like to see speed and efficiency compromised...

There are lots of chat clients available that are private and 'off the radar' so to speak so for me leave it out

Just my thoughts

Dj

nachoparker, 2017-10-19:
thanks very much @deljones

Keep in mind that if it is not activated from ncp-web it would not consume resources because it would not be running, but it would be preinstalled and preconfigured if somebody wants to activate it.

deljones, 2017-10-19:
Yep get that but even so tight and tiny works for me for the pi.. :)

Dj

nachoparker, 2017-10-19:
well, I guess it is a little bit out of scope of NCP. We'll use our limited resources for something else.

closed 2017-10-19 by nachoparker