nextcloud/nextcloudpi #169: notify updates/autoupdate (ncp) doesn't work

JimmyKater, 2017-08-31:
somehow i never get informed about new ncp-updates via the instance itself (admin-notify).
either i somehow got it wrong (most prolly) or it doesn't work correctly.

am running nc 12.0.2, ncp 24.3 (just manually updated to 24.9) on a pi3.

and if you didn't release 24.4, .4, .5, .6, .7, .8 ncp-autoupdate didn't work neither.

awww and btw... updating through ncp-web got me an 503 error (after telling me: something went wrong, pls try reloading page). that could be solved via cli

nachoparker, 2017-08-31:
Well, I released from 0.24.3 to 0.24.9 in one go, as you can see [here](https://github.com/nextcloud/nextcloudpi/releases)

The notifications show up in the 'bell' icon on top right. Like in this picture

[image]

We can check what happened, let's see the output of

```
systemctl status nc-notify-updates
```

My output
```
systemctl status nc-notify-updates
● nc-notify-updates.service - Notify in NC when a NextCloudPi update is available
   Loaded: loaded (/etc/systemd/system/nc-notify-updates.service; disabled)
   Active: inactive (dead) since Thu 2017-08-31 10:31:09 UTC; 21min ago
  Process: 7700 ExecStart=/usr/local/bin/ncp-notify-update (code=exited, status=0/SUCCESS)
 Main PID: 7700 (code=exited, status=0/SUCCESS)

Aug 31 10:31:09 raspberrypi systemd[1]: Started Notify in NC when a NextCloudPi update is available.
Aug 31 10:31:09 raspberrypi ncp-notify-update[7700]: Found update from v0.24.3 to v0.24.9. Already notified
```

JimmyKater, 2017-08-31:
` $ systemctl status nc-notify-updates
● nc-notify-updates.service - Notify in NC when a NextCloudPi update is available
   Loaded: loaded (/etc/systemd/system/nc-notify-updates.service; disabled)
   Active: inactive (dead) since Do 2017-08-31 12:52:09 CEST; 34min ago
  Process: 10296 ExecStart=/usr/local/bin/ncp-notify-update (code=exited, status=0/SUCCESS)
 Main PID: 10296 (code=exited, status=0/SUCCESS)
`

nachoparker, 2017-08-31:
Mmmm, it is like you are missing `/var/run/.ncp-latest-version`.

Output of 

```
cat /var/run/.ncp-latest-version
```
?

Do you shut down the raspberry pi every night or something like that?

JimmyKater, 2017-08-31:
```
$ cat /var/run/.ncp-latest-version
v0.24.9
```

> Do you shut down the raspberry pi every night

in the contrary. i have it running 24/7

nachoparker, 2017-08-31:
ok, sorry if it's a silly question, but did you activate in ncp-web or nextcloudpi-config?

nachoparker, 2017-08-31:
You seem to be missing the line "Started Notify in NC when a NextCloudPi update is available."

Then let's see if that line appears if we do

```
sudo systemctl start nc-notify-updates
systemctl status nc-notify-updates
```

JimmyKater, 2017-08-31:
` 
$ sudo systemctl start nc-notify-updates
pi@nextcloud:~ $ systemctl status nc-notify-updates
● nc-notify-updates.service - Notify in NC when a NextCloudPi update is available
   Loaded: loaded (/etc/systemd/system/nc-notify-updates.service; disabled)
   Active: inactive (dead) since Do 2017-08-31 16:30:28 CEST; 7s ago
  Process: 11660 ExecStart=/usr/local/bin/ncp-notify-update (code=exited, status=0/SUCCESS)
 Main PID: 11660 (code=exited, status=0/SUCCESS)
`

> ok, sorry if it's a silly question, but did you activate in ncp-web or nextcloudpi-config?

it's ok to ask this. and yes, of course i enabled it.

nachoparker, 2017-08-31:
I'm lost. Let's reinstall the thing and try again

```
sudo rm /usr/local/etc/nextcloudpi-config.d/nc-notify-updates.sh
sudo ncp-update
```

Then let's see if that line appears

```
sudo systemctl start nc-notify-updates
systemctl status nc-notify-updates
```

JimmyKater, 2017-08-31:
`
$ sudo rm /usr/local/etc/nextcloudpi-config.d/nc-notify-updates.sh
$ sudo ncp-update
Downloading updates
Performing updates
Installing nc-notify-updates
--2017-08-31 16:35:27--  https://github.com/nextcloud/admin_notifications/releases/download/v1.0.0/admin_notifications-1.0.0.tar.gz
Auflösen des Hostnamen »github.com (github.com)«... 192.30.253.112, 192.30.253.113
Verbindungsaufbau zu github.com (github.com)|192.30.253.112|:443... verbunden.
HTTP-Anforderung gesendet, warte auf Antwort... 302 Found
Platz: https://github-production-release-asset-2e65be.s3.amazonaws.com/81433081/532af79e-eebf-11e6-8b62-7ca29ec02c5a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20170831%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20170831T143527Z&X-Amz-Expires=300&X-Amz-Signature=eb39f37e33c494f644fd7e6d1dbc182225311495f20e16d417d02f25537f0d1d&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dadmin_notifications-1.0.0.tar.gz&response-content-type=application%2Foctet-stream[folge]
--2017-08-31 16:35:27--  https://github-production-release-asset-2e65be.s3.amazonaws.com/81433081/532af79e-eebf-11e6-8b62-7ca29ec02c5a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20170831%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20170831T143527Z&X-Amz-Expires=300&X-Amz-Signature=eb39f37e33c494f644fd7e6d1dbc182225311495f20e16d417d02f25537f0d1d&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dadmin_notifications-1.0.0.tar.gz&response-content-type=application%2Foctet-stream
Auflösen des Hostnamen »github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)«... 52.216.86.91
Verbindungsaufbau zu github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.86.91|:443... verbunden.
HTTP-Anforderung gesendet, warte auf Antwort... 200 OK
Länge: 18415 (18K) [application/octet-stream]
In »»admin_notifications.tar.gz«« speichern.

admin_notifications.tar.gz        100%[============================================================>]  17,98K  --.-KB/s   in 0,1s

2017-08-31 16:35:28 (183 KB/s) - »»admin_notifications.tar.gz«« gespeichert [18415/18415]

admin_notifications enabled
Activating nc-notify-updates
update web notifications enabled
Activating unattended-upgrades
Unattended upgrades active: yes (autoreboot yes)
NextCloudPi updated to version v0.24.9

$ sudo systemctl start nc-notify-updates

$ systemctl status nc-notify-updates
● nc-notify-updates.service - Notify in NC when a NextCloudPi update is available
   Loaded: loaded (/etc/systemd/system/nc-notify-updates.service; disabled)
   Active: inactive (dead) since Do 2017-08-31 16:36:18 CEST; 10s ago
  Process: 12281 ExecStart=/usr/local/bin/ncp-notify-update (code=exited, status=0/SUCCESS)
 Main PID: 12281 (code=exited, status=0/SUCCESS)
`

nachoparker, 2017-08-31:
maybe that line doesn't show up because you are in Stretch? Can you run this?

```
/usr/local/bin/ncp-notify-update
```

JimmyKater, 2017-08-31:
```
 $ sudo /usr/local/bin/ncp-notify-update
NextCloudPi up to date

```

you mean it doesn't show up b/c i'm **NOT** on stretch, yet?

nachoparker, 2017-08-31:
yes, sorry that's what I meant

but that detects that you are up to date, so it seems like it works. Can you double check that you have no 'bell' notifications pending icon in your nextcloud, like in the picture above?

We can try to fool NCP into thinking that we are in another version.

```
sudo su -c 'echo "v0.24.15" > /var/run/.ncp-latest-version'
sudo /usr/local/bin/ncp-notify-update
```

JimmyKater, 2017-08-31:
```
$ sudo su -c 'echo "v0.24.15" > /var/run/.ncp-latest-version'

$ sudo /usr/local/bin/ncp-notify-update
Found update from v0.24.9 to v0.24.15. Sending notification...
```

strange. it doesn't appear in NC-instance. neither was there a notification about another app-update that just appears)... so the problem could be caused by admin-notification app then?

nachoparker, 2017-08-31:
still doesn't appear on the nextcloud web?

Just to rule things out, output of this should be the same

```
sudo systemctl start nc-notify-updates
systemctl status nc-notify-updates
```

JimmyKater, 2017-08-31:
> still doesn't appear on the nextcloud web?

sorry - i misunderstood, my bad. of course it appears on top on ncp-web. i can see that there's a new version waiting.

```
$ sudo systemctl start nc-notify-updates

$ systemctl status nc-notify-updates
● nc-notify-updates.service - Notify in NC when a NextCloudPi update is available
   Loaded: loaded (/etc/systemd/system/nc-notify-updates.service; disabled)
   Active: inactive (dead) since Do 2017-08-31 17:01:27 CEST; 6s ago
  Process: 12544 ExecStart=/usr/local/bin/ncp-notify-update (code=exited, status=0/SUCCESS)
 Main PID: 12544 (code=exited, status=0/SUCCESS)
```

nachoparker, 2017-08-31:
no, I mean the nextcloud web, not ncp-web like in the screenshot above

JimmyKater, 2017-08-31:
ok. then i didn't misunderstood. and no, i doesnt appear in nextcloud web

nachoparker, 2017-08-31:
ok, then there's some issue with the admin_notifications app I would say

finally, let's try calling the app directly

```
sudo -u www-data php /var/www/nextcloud/occ notification:generate admin "Hello I am Jimmy Carter" 
```
One thing that I am thinking... are you logged in as the admin user in nextcloud? or as a regular user?

JimmyKater, 2017-08-31:
> One thing that I am thinking... are you logged in as the admin user in nextcloud? or as a regular user?

signed in with both (have set up a "support"-group and notifications go to admin and support. so my normal user is member of support (i made this to not needing to work under an admin-user with admin-privs all the time).
yesterday i was informed about updates via the notifications. 

just now with `sudo -u www-data php /var/www/nextcloud/occ notification:generate admin "Hello I am Jimmy Kater"` (note, i changed it a bit, since i'm not mr peanut Jimmy Carter but Jimmy Tomcat (in german: Kater - which is a soundalike for "Carter", of course - thus my buddy icon) i could hear a "bing" on my desktop  but i don't see any message. the "bing" could have marked an error as well... its exactly this kind of "bing"

nachoparker, 2017-08-31:
> yesterday i was informed about updates via the notifications.

what updates?

In firefox it asks you once wether you allow desktop notifications, but IDK if that's related. Maybe you hit 'no' a long time ago?

in any case, we know that there's something about the app that doesn't work for you :S 

Notifications should appear briefly on your desktop and be kept permanently on the nextcloud web (bell like notification icon)

nachoparker, 2017-08-31:
also, try the command again, and write `yourusername` instead of `admin`

JimmyKater, 2017-08-31:
> what updates?

see above: app-updates. no ncp-update :(

> In firefox it asks you once wether you allow desktop notifications, but IDK if that's related. Maybe you hit 'no' a long time ago?

i just checked it and ff is allowed to send me notifications from my nc-instance-URL

and i guess you just are coming closer to the point. when i used the direct adressing with my username (not admin, since i've disabled this account for security reasons - there's just a group called like this) i could generate a notification (no bell, though) and it got shown under nextcloud web AND by ff-notification. exactly the way it should be.

JimmyKater, 2017-08-31:
is there a way to address the group called "admin"?

nachoparker, 2017-08-31:
I am pretty sure that that's what's happening, lol

I should have thought about that, but your systemctl messages were different (bc jessie) so I was confused :S

It seems that there's no way to notify all users or groups, just individual users

https://github.com/nextcloud/admin_notifications/issues/5
https://github.com/nextcloud/admin_notifications/issues/4

This does give me the idea that we could at least configure what user gets notified. 

Have you tried using just your username (not group) in the command?

Thanks for the patience debugging ;)

JimmyKater, 2017-08-31:
> This does give me the idea that we could at least configure what user gets notified.

ya, please. 

> Have you tried using just your username (not group) in the command?

works brilliant. with both. 

> Thanks for the patience debugging ;)

and back to you for your patience of not giving up on it.

nachoparker, 2017-08-31:
> works brilliant. with both.

yesss, that's it then! I just assumed you were using the admin user. dumb me xD

To the enhacement list!

JimmyKater, 2017-09-01:
ahhhh... btw: ncp doesn't update automatically is understood as being a follow-up error, then?

nachoparker, 2017-09-01:
Implemented in `v0.24.17`. Will release it soon

nachoparker, 2017-09-01:
> ahhhh... btw: ncp doesn't update automatically is understood as being a follow-up error, then?

If you are talking about #170, I recommend that you either

 - change back to 'jessie' in `/etc/apt/apt.conf.d/20nextcloudpi-upgrades`
 - do dist-upgrade and hop to latest Raspbian 9 (instructions in ownyourbits)

closed 2017-09-01 by nachoparker

JimmyKater, 2017-09-01:
> and if you didn't release 24.4, .4, .5, .6, .7, .8 ncp-autoupdate didn't work neither.

hehehe. no. i was referring to one sentence from my initial #169-posting... --> 
> and if you didn't release 24.4, .4, .5, .6, .7, .8 **ncp-autoupdate didn't work neither.**

but thanks for your recommendations about upgrading. i'm really planning to upgrade to stretch rather sooner than later.