nextcloud/nextcloudpi #170: unattended upgrades seem to be without function, at least sometimes

JimmyKater, 2017-08-31:
hey nacho,

i always thought that unattended upgrades would update raspbian automatically, fully. so as i'm still running raspbian 8 and as i just ssh'd in to rpi3 and making an apt-get update and apt-get upgrade i was surprised that it indeed DID update and upgrade... but i thought that this would be done by ncp-option "unattended upgrades", automatically.
this happened, running ncp-24.3. i'm on 24.9 now

nachoparker, 2017-08-31:
hi,

Well it updates everything but it will not hop to the next release Raspbian 9. That has to be done manually ( see ownyourbits )

I check it every now and then, and it works for me (unattended upgrades).

Please, can we have a look at the logs for unattended upgrades?

```
cat /var/log/unattended-upgrades/unattended-upgrades.log
```

For example, my output is
```
2017-08-31 06:00:36,397 INFO Initial blacklisted packages:                                                             
2017-08-31 06:00:36,400 INFO Starting unattended upgrades script                                                       
2017-08-31 06:00:36,401 INFO Allowed origins are: ['o=Raspbian,n=stretch,l=Raspbian']                                  
2017-08-31 06:01:33,129 INFO No packages found that can be upgraded unattended
```

And this is an older one where you can see the update that took place

```
2017-08-25 06:02:51,286 INFO Initial blacklisted packages: 
2017-08-25 06:02:51,289 INFO Starting unattended upgrades script
2017-08-25 06:02:51,291 INFO Allowed origins are: ['o=Raspbian,n=stretch,l=Raspbian']
2017-08-25 06:04:12,136 INFO Packages that will be upgraded: augeas-lenses libaugeas0 libxml2
2017-08-25 06:04:12,150 INFO Writing dpkg log to '/var/log/unattended-upgrades/unattended-upgrades-dpkg_2017-08-25_06:04:12.142091.log'
2017-08-25 06:05:30,161 INFO All upgrades installed
2017-08-26 06:00:34,992 INFO Initial blacklisted packages: 
2017-08-26 06:00:34,994 INFO Starting unattended upgrades script
2017-08-26 06:00:34,996 INFO Allowed origins are: ['o=Raspbian,n=stretch,l=Raspbian']
2017-08-26 06:01:31,265 INFO No packages found that can be upgraded unattended
2017-08-27 06:00:41,808 INFO Initial blacklisted packages: 
2017-08-27 06:00:41,812 INFO Starting unattended upgrades script
2017-08-27 06:00:41,813 INFO Allowed origins are: ['o=Raspbian,n=stretch,l=Raspbian']
2017-08-27 06:01:57,876 INFO No packages found that can be upgraded unattended
2017-08-28 06:00:34,902 INFO Initial blacklisted packages: 
2017-08-28 06:00:34,904 INFO Starting unattended upgrades script
2017-08-28 06:00:34,906 INFO Allowed origins are: ['o=Raspbian,n=stretch,l=Raspbian']
2017-08-28 06:01:32,979 INFO No packages found that can be upgraded unattended
2017-08-29 06:02:49,174 INFO Initial blacklisted packages: 
2017-08-29 06:02:49,177 INFO Starting unattended upgrades script
2017-08-29 06:02:49,179 INFO Allowed origins are: ['o=Raspbian,n=stretch,l=Raspbian']
2017-08-29 06:03:43,590 INFO No packages found that can be upgraded unattended
2017-08-30 06:02:03,946 INFO Initial blacklisted packages: 
2017-08-30 06:02:03,949 INFO Starting unattended upgrades script
2017-08-30 06:02:03,951 INFO Allowed origins are: ['o=Raspbian,n=stretch,l=Raspbian']
2017-08-30 06:02:57,982 INFO No packages found that can be upgraded unattended
```

JimmyKater, 2017-08-31:
` $ cat /var/log/unattended-upgrades/unattended-upgrades.log
2017-08-02 06:00:18,764 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-02 06:00:18,765 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
2017-08-02 06:00:18,766 INFO erlaubte Ursprünge sind: ['o=Raspbian,n=jessie,l=Raspbian', 'o=Raspbian,n=stretch,l=Raspbian']
2017-08-02 06:01:26,971 INFO Es wurden keine Pakete gefunden, von denen ein unbeaufsichtigtes Upgrade durchgeführt werden kann.
2017-08-03 06:00:23,403 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-03 06:00:24,212 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
2017-08-03 06:00:24,213 INFO erlaubte Ursprünge sind: ['o=Raspbian,n=jessie,l=Raspbian', 'o=Raspbian,n=stretch,l=Raspbian']
2017-08-03 06:01:18,302 INFO Es wurden keine Pakete gefunden, von denen ein unbeaufsichtigtes Upgrade durchgeführt werden kann.
2017-08-04 06:03:38,198 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-04 06:03:38,200 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
2017-08-04 06:03:38,200 INFO erlaubte Ursprünge sind: ['o=Raspbian,n=jessie,l=Raspbian', 'o=Raspbian,n=stretch,l=Raspbian']
2017-08-04 06:04:33,997 INFO Es wurden keine Pakete gefunden, von denen ein unbeaufsichtigtes Upgrade durchgeführt werden kann.
2017-08-05 06:01:32,316 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-05 06:01:32,318 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
2017-08-05 06:01:32,318 INFO erlaubte Ursprünge sind: ['o=Raspbian,n=jessie,l=Raspbian', 'o=Raspbian,n=stretch,l=Raspbian']
2017-08-05 06:02:33,696 INFO Es wurden keine Pakete gefunden, von denen ein unbeaufsichtigtes Upgrade durchgeführt werden kann.
2017-08-06 06:01:25,717 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-06 06:01:25,719 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
2017-08-06 06:01:25,720 INFO erlaubte Ursprünge sind: ['o=Raspbian,n=jessie,l=Raspbian', 'o=Raspbian,n=stretch,l=Raspbian']
2017-08-06 06:02:29,679 INFO Es wurden keine Pakete gefunden, von denen ein unbeaufsichtigtes Upgrade durchgeführt werden kann.
2017-08-07 06:00:21,863 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-07 06:00:21,866 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
2017-08-07 06:00:21,867 INFO erlaubte Ursprünge sind: ['o=Raspbian,n=jessie,l=Raspbian', 'o=Raspbian,n=stretch,l=Raspbian']
2017-08-07 06:01:16,961 INFO Es wurden keine Pakete gefunden, von denen ein unbeaufsichtigtes Upgrade durchgeführt werden kann.
2017-08-08 06:00:33,479 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-08 06:00:34,267 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
2017-08-08 06:00:34,269 INFO erlaubte Ursprünge sind: ['o=Raspbian,n=jessie,l=Raspbian', 'o=Raspbian,n=stretch,l=Raspbian']
2017-08-08 06:01:33,276 INFO Es wurden keine Pakete gefunden, von denen ein unbeaufsichtigtes Upgrade durchgeführt werden kann.
2017-08-09 06:00:21,274 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-09 06:00:22,524 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
2017-08-09 06:00:22,526 INFO erlaubte Ursprünge sind: ['o=Raspbian,n=jessie,l=Raspbian', 'o=Raspbian,n=stretch,l=Raspbian']
2017-08-09 06:01:24,867 INFO Es wurden keine Pakete gefunden, von denen ein unbeaufsichtigtes Upgrade durchgeführt werden kann.
2017-08-10 06:00:24,056 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-10 06:00:24,058 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
2017-08-10 06:00:24,059 INFO erlaubte Ursprünge sind: ['o=Raspbian,n=jessie,l=Raspbian', 'o=Raspbian,n=stretch,l=Raspbian']
2017-08-10 06:01:13,745 INFO Es wurden keine Pakete gefunden, von denen ein unbeaufsichtigtes Upgrade durchgeführt werden kann.
2017-08-11 06:00:22,702 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-11 06:00:22,704 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
2017-08-11 06:00:22,705 INFO erlaubte Ursprünge sind: ['o=Raspbian,n=jessie,l=Raspbian', 'o=Raspbian,n=stretch,l=Raspbian']
2017-08-11 06:01:18,411 INFO Es wurden keine Pakete gefunden, von denen ein unbeaufsichtigtes Upgrade durchgeführt werden kann.
2017-08-12 06:00:22,000 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-12 06:00:22,001 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
2017-08-12 06:00:22,002 INFO erlaubte Ursprünge sind: ['o=Raspbian,n=jessie,l=Raspbian', 'o=Raspbian,n=stretch,l=Raspbian']
2017-08-12 06:01:21,211 INFO Es wurden keine Pakete gefunden, von denen ein unbeaufsichtigtes Upgrade durchgeführt werden kann.
2017-08-13 06:01:54,987 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-13 06:01:54,989 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
2017-08-13 06:01:54,989 INFO erlaubte Ursprünge sind: ['o=Raspbian,n=jessie,l=Raspbian', 'o=Raspbian,n=stretch,l=Raspbian']
2017-08-13 06:03:14,523 INFO Packages that will be upgraded: git git-man libgd3
2017-08-13 06:03:15,971 INFO Dpkg-Protokoll wird nach »/var/log/unattended-upgrades/unattended-upgrades-dpkg_2017-08-13_06:03:15.963487.log« geschrieben
2017-08-13 06:05:23,544 INFO alle Upgrades installiert
2017-08-14 06:00:27,896 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-14 06:00:28,465 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
[... cut ...]
2017-08-30 06:03:02,187 INFO Es wurden keine Pakete gefunden, von denen ein unbeaufsichtigtes Upgrade durchgeführt werden kann.
2017-08-31 06:00:15,692 INFO Pakete, die anfangs auf die schwarzen Liste standen:
2017-08-31 06:00:15,694 INFO Skript für unbeaufsichtigte Upgrades wird gestartet.
2017-08-31 06:00:15,695 INFO erlaubte Ursprünge sind: ['o=Raspbian,n=stretch,l=Raspbian']
2017-08-31 06:01:26,652 INFO Es wurden keine Pakete gefunden, von denen ein unbeaufsichtigtes Upgrade durchgeführt werden kann.
`

JimmyKater, 2017-08-31:
ossry, it's all german... 
and no, i wanted to wait some more time with updating to stretch... plus i know that i need to do it manually (and how)

JimmyKater, 2017-08-31:
ok, i see an auto-update in my log as well... on aug, 25th. BUUUUUT... when i ran apt-get update and apt-get upgrade today it found 22mb worth of data to be installed..
could that be b/c of `Allowed origins are: ['o=Raspbian,n=stretch,l=Raspbian']`

nachoparker, 2017-08-31:
Oh @JimmyKater ,

You hit the nail in the head. I changed it to update only stretch when we moved to stretch. You can replace that for `jessie` if you don't want to upgrade yet.

I recommend to update to the latest thing because now all the updates and developments will go over stretch. The release cycle for Raspbian is about two years.

Thanks for the feedback again, as usual

closed 2017-08-31 by nachoparker