nextcloud/nextcloudpi #171: Slowwww since upgrade & changelog

deljones, 2017-08-31:
Hi all

Running Nextcloudpi v0.24.7 with NC12.0.2
Everything works just fine however I'm aware of speed feeling rather sluggish sing 0.24.7 upgrade and the proceeding version. Before these two versions my instance was quite 'nippy' and 'sharp'. Has anyone else noticed a speed or response change?

Maybe I'm not looking in the right place but the ncPI updates seem to happen on a regular basis. Where can I find the changelog for the version update?

Love my ncpi though, relying on it more each day!!!

Dj

nachoparker, 2017-08-31:
Hi del,

[here](https://github.com/nextcloud/nextcloudpi/releases) are the releases ( click in the three dots ... )

Can you try something for me? Try reverting the change in `v0.24.5`
```
sudo sed -i 's|pm.max_children =.*|pm.max_children = 5|' /etc/php/7.0/fpm/pool.d/www.conf
sudo reboot
```
And see if you notice a difference.

Thanks for reporting!

deljones, 2017-08-31:
Hi np

 

Entered the code as given but still shows as 0.24.9 ?

 

Dj

nachoparker, 2017-08-31:
Hi, yes I didn't change the version, just did a manual change in the configuration ( that will be gone next time you nc-update )

Do you notice any difference?

deljones, 2017-08-31:
I see..

 

OK. I’m just giving it a good thrashing now to see how it responds…

 

Will report back during the day

Dj

nachoparker, 2017-08-31:
Cool, thanks.

Please pay attention to what is slow... uploads? downloads? the web interface?

Anybody else noticed this?

thibmaek, 2017-08-31:
Yeah it's taking over 1 minute to log in to the admin interface through my local network.
And this is on a clean install btw, first time I tried NCP.

nachoparker, 2017-08-31:
hi @thibmaek 

The first time always takes a loooong time because it is initializing and caching things.

After the first time everything loads it should be much faster.

deljones, 2017-09-01:
I think the problem for me was my local network.... I'm out and about now and it's flying.... I'll keep an eye on it when I get back to Base... 

Dj

w: derekjones.net

e: derek AT derekjones.net

t: +44 7957 140 512

nachoparker, 2017-09-01:
good to know, please reopen if it happens again

closed 2017-09-01 by nachoparker

nachoparker, 2017-09-07:
Hi Del,

You were asking about a changelog, so now it generates automatically with each build

https://github.com/nextcloud/nextcloudpi/blob/master/changelog.md

This is to implement #139 

Cheers,

deljones, 2017-09-07:
Brilliant.... Thanks for your work and time... 

Dj

w: derekjones.net

e: derek AT derekjones.net

t: +44 7957 140 512

https://github.com/nextcloud/nextcloudpi/blob/master/changelog.md

nachoparker, 2017-09-07:
glad to help. I am sorry about your issues with automount

deljones, 2017-09-07:
Hey no worries... It's great learning for us all.... 

Dj

w: derekjones.net

e: derek AT derekjones.net

t: +44 7957 140 512