nextcloud/nextcloudpi #175: 0.26.3 server error :(

deljones, 2017-09-04:
Hi all

Have just done the uodate to 0.26.3. After a reboot I now get an internal server error :(

Tried a power recycle and another reboot. Should I try a restore from backup or is this a known error?

Dj
[image]

nachoparker, 2017-09-04:
Hi del,

Sorry to hear that.

After all the problems that automount was causing with the change to Stretch plus old issues, I completely rewrote it so probably something changed in the way you were accessing the mounted folders.

It is probably the database, and maybe the datadir. Let's see

```
ls -l /media
```

and

```
systemctl status mysqld
```

and 

```
sudo -u www-data /var/www/nextcloud/occ config:system:get datadirectory
```

Automount should be rock solid from now on, and we won't need to do this kind of change again.

deljones, 2017-09-04:
Last item...

Command not found

Dj

nachoparker, 2017-09-04:
really?

Have you moved the nextcloud directory? in any case, I want to know the `datadir` value in `config.php`

deljones, 2017-09-04:
Np

Config.php data directory is as follows 

media/USBdrive1/ncdata

Dj

w: derekjones.net

e: derek AT derekjones.net

t: +44 7957 140 512

nachoparker, 2017-09-04:
ok, what about
```
ls -l /media
```
```
systemctl status mysqld
```

Ricardosgeral, 2017-09-04:
Hi.
I've made update to 26.3 and apparently lost my database. It's saying that don't have .ocdata file or don't have permissions.

But the stranger is that when I do
cd /media
and then
ls -l
I have this !!!

pi@raspberrypi:/media $ ls -lh
total 36K
drwxr-xr-x 2 root root 4.0K Aug 30 10:35 USBdrive
drwxr-xr-x 2 root root 4.0K Aug 30 10:42 USBdrive1
drwxr-xr-x 2 root root 4.0K Aug 30 14:33 USBdrive2
drwxr-xr-x 2 root root 4.0K Sep  1 08:19 USBdrive3
drwxr-xr-x 2 root root 4.0K Sep  2 17:21 USBdrive4
drwxr-xr-x 2 root root 4.0K Sep  2 21:18 USBdrive5
drwxr-xr-x 2 root root 4.0K Sep  2 21:29 USBdrive6
drwxr-xr-x 2 root root 4.0K Sep  3 12:23 USBdrive7
drwxr-xr-x 2 root root 4.0K Sep  3 15:16 USBdrive8
pi@raspberrypi:/media $

Why do I have 8 USB drive? I only have 1 drive attached.
Why do I have 4k in each one¿ Where is the data?
Did the update formatted my drive? 
Help here

deljones, 2017-09-04:
Hello there Np.... Sorry I've been out and about doing this using SSH on a mobile on a bus in bad weather....

So...
**ls -l /media** gives me the following:
------
lrwxrwxrwx 1 root root   16 Sep  3 17:47 myCloudDrive -> /media/USBdrive1
drwxr-xr-x 6 root root 4096 Aug 18 16:29 myCloudDrive1
lrwxrwxrwx 1 root root   15 Sep  3 17:47 ncbackup -> /media/USBdrive
drwxr-xr-x 4 root root 4096 Aug 26 01:49 ncbackup1
drwxr-xr-x 2 root root 4096 Sep  3 17:47 USBdrive
drwxr-xr-x 2 root root 4096 Sep  3 17:47 USBdrive1
drwxr-xr-x 2 root root 4096 Sep  3 17:47 USBdrive2
drwxr-xr-x 2 root root 4096 Sep  3 17:47 USBdrive3
------

**systemctl status mysqld** Gives me....

------
● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled)
   Active: active (running) since Mon 2017-09-04 10:42:15 UTC; 3h 54min ago
  Process: 1194 ExecStartPost=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 1191 ExecStartPost=/etc/mysql/debian-start (code=exited, status=0/SUCCESS)
  Process: 1039 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
  Process: 935 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP_START_POSITION=$VAR || exit 1 (code=exited, status=0/SUCCESS)
  Process: 918 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 887 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 1164 (mysqld)
   Status: "Taking your SQL requests now..."
   CGroup: /system.slice/mariadb.service
           └─1164 /usr/sbin/mysqld

Warning: Unit file changed on disk, 'systemctl daemon-reload' recommended.
--------
**sudo -u www-data /var/www/nextcloud/occ config:system:get datadirector**y, gives me...

--------
sudo: /var/www/nextcloud/occ: command not found
--------

I think I understand why the drives keep on 'breeding' based on a past update, this is also seen on my ncbackup, which has now sprouted an ncbackup1?

In my NC config.php my data directory is:
  **'datadirectory' => '/media/USBdrive/ncdata',**

I'm sure that it will be something really simple. I'm going to have a poke around the drives and hunt down where my database it and my ncdata is as its going to be where things are pointing to since the last update. As far as I was aware, when I started everything was on the SD except the data, now I'm not so sure  :)

All the best 

Dj

nachoparker, 2017-09-04:
Hi

About having eight folders... this appeared in Stretch ( #142 ) and is one of the reasons I finally decided to completely start automount from zero.

You should clean them up...

```
sudo umount /media/*
sudo rmdir /media/*
sudo reboot
```
(you will see some warnings)

Then, your stuff should be in `/media/myclouddrive`. You should use that path in config.php and in
`/etc/mysql/mariadb.conf.d/50-server.cnf` (datadir value)

That should leave your system clean

In any case, it seems like mysql is running, but if you can, cleanup like I said, reboot and let's see the status of mysqld and php

nachoparker, 2017-09-04:
I only recommend to use `/media/USBdrive` when you only have one USB drive

If you have more than one the order cannot be guaranteed, so it is better to use `/media/labelname`

deljones, 2017-09-04:
Hi Np,
is your reply to both ricardosgeral and myself or just ricardosgeral?
Dj

nachoparker, 2017-09-04:
aaah I didn't see that he was in the conversation :/

I thought that was you, ok now it gets more confusing. edited

deljones, 2017-09-04:
sorry

dj

deljones, 2017-09-04:
Hi Np

OK... Working again!! yee ha!

When I first installed ncp all those nights ago I gave my data drive a name, USBdrive1, so that I could keep an eye on where my data was going..

Since this recent update the name has been changed to myCloudDrive1! 
myCloudDrive is empty. Guess this is the mounted drive and adding 1 is the actual drirectory?
my original drive USBdrive1 is also empty. DOn't know why

A second drive I had attached for backups I called ncbackup which as become ncbackup1/ncbackup, again this is the volumeName/directory?

I've not moved any directories, I have kept everything in the SD card except the datadir. thangs have certainly been renamed etc and I think I know where everything is so I guess I shoukd start on the mission of rm'ing dead mount/drirectories?

It took a bit of poking about but thats the fun and learning side of it, best part is I have everthing ...:)

Good work

Dj

deljones, 2017-09-04:
I'm guessing that there is only two of us that have bumped into this issue as nobody elese has reported it, if indeed it is an issue?

On another note. Since this update the gallery takes an age now to load up in any browser and doesnt work at all now on the v2 RC5 android app. It used to before this recent update.. 

Dj

nachoparker, 2017-09-04:
good to know

Yeah, I am really sorry about that. That is the reason I did not want to rewrite automount in the first place, I knew that some people would have used the `/media/USBdriveX` names.

This new implementation looks way better, hopefully won't happen again ;)

nachoparker, 2017-09-04:
> if indeed it is an issue?

Well, not a major one but still I changed things which is not desirable. Anyhow the instructions say clearly to use the label name so that's why most people are fine.

> On another note. Since this update the gallery takes an age now to load up in any browser and doesnt work at all now on the v2 RC5 android app. It used to before this recent update..

Really? I suspect it can be related to #146 

Would you mind trying going back to `pm.max_children = 5` and rebooting to see if it gets any better?

Thanks, we can continue on that thread

closed 2017-09-04 by nachoparker

deljones, 2017-09-04:
pm.max_children = 5

command not found

nachoparker, 2017-09-04:
sorry, you have to change that line in `/etc/php/7.0/fpm/pool.d/www.conf`

it is not a command

let's continue that converation in #146

deljones, 2017-09-04:
doh!

dj

Ricardosgeral, 2017-09-04:
Hi Nacho
Sorry for the crossed conversation.

The server was installed with nextcloudpi then I've made the mount usb and changed data to myCloudDrive (I guess because I used all default suggestions in nextcloudpi-config).
The bad news is that after the update to 26.3 I don't have nothing on the /media folder.

NextCloudPi v0.26.3 is up to date
pi@raspberrypi:~ $ cd /media
pi@raspberrypi:/media $ ls
pi@raspberrypi:/media $ ls -l
total 0

It seems that the update erased the folder with the ncdata in myCloudDrive.

I don't understand

nachoparker, 2017-09-04:
nono, it was my fault xD

try activating `nc-automount` again (even if it says it's active)

Ricardosgeral, 2017-09-04:
Ok now it appears myCloudDrive ufff

Ricardosgeral, 2017-09-04:
But I still don't have the server running.
What should I do?
It still says the thing about .ocdata file

nachoparker, 2017-09-04:
Nice, now we have to update your Nextcloud to point to the new place

First, let's make sure that the data folder is still there

```
ls /media/myCloudDrive
```

Then edit the file `/var/www/nextcloud/config/config.php`

```
nano /var/www/nextcloud/config/config.php
```

And change the value of `datadir` to the new path. Something like

```
 'datadirectory' => '/media/myCloudDrive/ncdata',
```

Press ctrl-x to exit and hit `y` to save

If you didn't move your database, that should be it

Ricardosgeral, 2017-09-04:
This is the content of myCloudDrive root@raspberrypi:/home/pi# ls /media/myCloudDrive
aria_log.00000001  ib_logfile0  ncdata
aria_log_control   ib_logfile1  nextcloud-bkp_20170904.tar
ibdata1            lost+found   tc.log
root@raspberrypi:/home/pi# ls -l /media/myCloudDrive
total 15048552
-rw-rw---- 1 mysql    mysql           8192 Sep  4 20:03 aria_log.00000001
-rw-rw---- 1 mysql    mysql             52 Sep  4 20:03 aria_log_control
-rw-rw---- 1 mysql    mysql       12582912 Sep  4 20:03 ibdata1
-rw-rw---- 1 mysql    mysql       50331648 Sep  4 20:03 ib_logfile0
-rw-rw---- 1 mysql    mysql       50331648 Sep  4 20:03 ib_logfile1
drwxrwxrwx 2 root     root           16384 Aug 30 10:34 lost+found
drwxrwx--- 9 www-data www-data        4096 Aug 30 10:47 ncdata
-rw-r--r-- 1 root     root     15296409600 Sep  4 08:30 nextcloud-bkp_20170904.tar
-rw-rw---- 1 mysql    mysql          24576 Sep  4 20:03 tc.log

I've made your suggestions and now in web browser says

”This Nextcloud instance is currently in maintenance mode, which may take a while.
This page will refresh itself when the Nextcloud instance is available again.
Contact your system administrator if this message persists or appeared unexpectedly.
Thank you for your patience.”

Should I wait?

nachoparker, 2017-09-04:
sweet! we are almost there, just use this command

```
sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --off
```

Ricardosgeral, 2017-09-04:
Yes you did ;). It's working. Thanks a lot.
One last question. I've made a backup and included the files. I was expecting the bak file to be very big (400Gb is the total size of my data). However when I look at the size of my bak file it only has 15Gb (at least is what ls -lh shows). Is that normal¿

nachoparker, 2017-09-04:
> and included the files

If that means that you checked INCLUDEDATA, then no, there's something wrong : (

Sorry for the confusion with automount

Ricardosgeral, 2017-09-04:
No worries Nacho Parker
You are doing a great job (pro bono) for the community. Even so I Can't ask for better ;)

I've to see better the backup. all data is on Nextcloud web and sync with clients. What does backuplimit=4  means?

nachoparker, 2017-09-04:
haha thanks

the backup limit means that once you have 4 backups, the next one will delete the oldest one. That way you will not run out of space.

yeah, thanks for being patience. All the issues you are having will help polish the experience for other people.