nextcloud/nextcloudpi #176: Update NextCloudPi via webGUI

user-jan, 2017-09-04:
Hi Nacho,

When I try to update NextCloudPi (v0.25.2) via the webGUI it does everything fine until the end when it gives me the message "Something went wrong. Try refreshing the page" and when I try to refresh the page it says "Service Unavailable". When I do the update via ssh & terminal it works fine.

Cheers,
Jan

Here are 2 screenshoots of it:

[image]
[image]

JimmyKater, 2017-09-04:
+1 here.

nachoparker, 2017-09-04:
Hi,

Yes, I noticed and I created a bug #174 

Everything works, but when it finishes it throws that error instead of the green circle.

I'll close as duplicate. Just ignore it for now, is a false alarm

Thanks for reporting ;)

user-jan, 2017-09-04:
You're welcome ;)

nachoparker, 2017-09-04:
Actually I am going to reopen, I want to make sure it's a different thing.

BTW jan, I was talking about you the other day at the Nextcloud conference, it will be on video soon ;)

JimmyKater, 2017-09-04:
could be b/c there might be missing an apache reload or such... at the end. just guessing. not knowing. since running ncp-update from cli works as a problem solver. it occured after 24.3 with me.

nachoparker, 2017-09-04:
yes, that's exactly when SSE events were introduced.

There's a restart of PHP in the update script, to make the change to the new system, so at this point it can be removed. Unfortunately, it causes that message in the web UI

I think it's just a matter of removing that restart, in any case it has already taken place so it's no longer needed.

nachoparker, 2017-09-04:
Hi Jan, you were right

Fixed in `v0.26.3`

Thanks for telling us

closed 2017-09-04 by nachoparker

nachoparker, 2017-09-12:
@user-jan [my thanks to you in public](https://ownyourbits.com/2017/09/10/nextcloudpi-at-the-nextcloud-conference/)

arikalfus, 2018-10-03:
Note: this exact error occurred for me for the update to 0.62.2 from 0.60.x (Forget what my previous patch version was).

My guess is because of the change from PHP 7.0 to 7.2.

nachoparker, 2018-10-03:
probably. That's a harmless thing though. The process went through fine

arikalfus, 2018-10-03:
Yeah, it failed over the web UI but proceeded fine with `ncp-update` on the box. Didn't see a more relevant issue but wanted to drop that comment for anyone else that would search the issues list for this.