nextcloud/nextcloudpi #179: Errors during docker image creation

ingorichter, 2017-09-04:
Creating the docker image on my Odroid-C2 results in a bunch of errors.

I'm running a Linux version based on `https://github.com/hypriot/image-builder-odroid-c2`

`Linux black-pearl 3.14.79-109 #1 SMP PREEMPT Thu Mar 16 20:05:25 BRT 2017 aarch64 GNU/Linux`

```
$ make -C nextcloudpi/
make: Entering directory '/home/pirate/nextcloudpi'
docker build . -f docker/raspbian.dockerfile  -t ownyourbits/miniraspbian:latest
Sending build context to Docker daemon 1.035 MB
Step 1/3 : FROM ownyourbits/miniraspbian:raw
raw: Pulling from ownyourbits/miniraspbian
4cbc6a1c3846: Already exists
Digest: sha256:bbd0b9e065d1001752ed8827525af240a00cfd7ae68b6059127ecbcbaa1a7174
Status: Downloaded newer image for ownyourbits/miniraspbian:raw
 ---> 2d8cfcd8d89f
Step 2/3 : MAINTAINER Ignacio Núñez Hernanz <nacho@ownyourbits.com>
 ---> Running in d39a2ffdfe23
 ---> 4f966da71080
Removing intermediate container d39a2ffdfe23
Step 3/3 : CMD /bin/bash
 ---> Running in 654277ddcd21
 ---> e34adeaff104
Removing intermediate container 654277ddcd21
Successfully built e34adeaff104
docker build . -f docker/lamp.dockerfile      -t ownyourbits/lamp-arm:latest
Sending build context to Docker daemon 1.035 MB
Step 1/8 : FROM ownyourbits/miniraspbian
 ---> e34adeaff104
Step 2/8 : MAINTAINER Ignacio Núñez Hernanz <nacho@ownyourbits.com>
 ---> Running in 728913b43e1d
 ---> 8acda42418d9
Removing intermediate container 728913b43e1d
Step 3/8 : SHELL /bin/bash -c
 ---> Running in b456210987ff
 ---> 3292ca52f242
Removing intermediate container b456210987ff
Step 4/8 : COPY etc/library.sh lamp.sh /usr/local/etc/
 ---> 1276fa3cd975
Removing intermediate container 177faa4bc862
Step 5/8 : RUN source /usr/local/etc/library.sh; set +x; install_script /usr/local/etc/lamp.sh;      apt-get autoremove -y; apt-get clean; rm /var/lib/apt/lists/* -f; rm -rf /usr/share/man/*; rm -rf /usr/share/doc/*;      mkdir -p /data/;      mv /var/lib/mysql /data/database;      sed -i "s|^datadir.*|datadir = /data/database|" /etc/mysql/mariadb.conf.d/50-server.cnf;      rm /data/database/ib_logfile*;      rm /var/cache/debconf/*-old;      rm /var/log/alternatives.log /var/log/apt/* ;      rm /usr/local/etc/{lamp.sh,library.sh}
 ---> Running in 292825e2c3c3
Installing lamp
Get:1 http://mirrordirector.raspbian.org jessie InRelease [14.9 kB]
Get:2 http://archive.raspberrypi.org jessie InRelease [22.9 kB]
Get:3 http://mirrordirector.raspbian.org jessie/main armhf Packages [9535 kB]
Get:4 http://mirrordirector.raspbian.org jessie/contrib armhf Packages [43.3 kB]
Get:5 http://archive.raspberrypi.org jessie/main armhf Packages [170 kB]
Get:6 http://mirrordirector.raspbian.org jessie/non-free armhf Packages [84.2 kB]
Get:7 http://mirrordirector.raspbian.org jessie/rpi armhf Packages [1356 B]
Get:8 http://archive.raspberrypi.org jessie/ui armhf Packages [58.9 kB]
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Fetched 9931 kB in 38s (256 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
The following extra packages will be installed:
  libapt-inst1.5
The following NEW packages will be installed:
  apt-utils libapt-inst1.5
0 upgraded, 2 newly installed, 0 to remove and 20 not upgraded.
Need to get 519 kB of archives.
After this operation, 1280 kB of additional disk space will be used.
[... 472 lines cut ...]
Removing apache2-data (2.4.10-10+deb8u10) ...
Removing apache2-utils (2.4.10-10+deb8u10) ...
Removing libaprutil1-dbd-sqlite3:armhf (1.5.4-1) ...
Removing libaprutil1:armhf (1.5.4-1) ...
Removing libapr1:armhf (1.5.1-3) ...
Removing libexpat1:armhf (2.1.0-6+deb8u4) ...
Removing libgmp10:armhf (2:6.0.0+dfsg-6+rpi1) ...
Removing liblua5.1-0:armhf (5.1.5-7.1) ...
Removing procps (2:3.3.9-9) ...
Removing libncursesw5:armhf (5.9+20140913-1) ...
Removing libsasl2-2:armhf (2.1.26.dfsg1-13+deb8u1) ...
Removing libsasl2-modules-db:armhf (2.1.26.dfsg1-13+deb8u1) ...
Removing libsqlite3-0:armhf (3.8.7.1-1+deb8u2) ...
Removing libxml2:armhf (2.9.1+dfsg1-5+deb8u5) ...
Removing mime-support (3.58) ...
Processing triggers for libc-bin (2.19-18+deb8u7) ...
rm: cannot remove '/var/lib/apt/lists/partial': Is a directory
 ---> 172c81c29cb1
Removing intermediate container a2c05c10c254
Step 6/6 : COPY docker/run-nc.sh /usr/local/bin/run.sh
 ---> 21506900e7ed
Removing intermediate container 7e8bfcdafd9e
Successfully built 21506900e7ed
make: Leaving directory '/home/pirate/nextcloudpi'
```

nachoparker, 2017-09-04:
Hi,

The docker image still has not been updated to use Raspbian 9, and the ncp-web interface. Have been pretty busy lately with the migration to Raspbian 9, so that has not yet been addressed.

I suggest you docker pull the image

Of course, any help is appreciated. Also, I am really interested in trying this on an odroid. I have bought an odroid HC1 that still has not arrived.

On a side note... you seem to have a pretty old 3.14 kernel. I don't know if all the kernel features needed for containers will be in place. Is there support for a more recent kernel for that board?

ingorichter, 2017-09-04:
No worries, I just wanted to make sure that this doesn't fall off the table. ;-)

The kernel is pretty old, but the Hypriot guys added everything to run docker. I ran some other container with the same kernel and it was alright. I don't know if they are going to update the kernel any time soon. In the meantime, I'll stick with it and try to get the image working.

I've never tried the HC1. I'm curious how that machine will perform.

I pulled the docker image, but it seemed to be a bit older and I wasn't able to setup owncloud. The web interface showed me the login dialog. Is there anything I was overlooking.

Thanks for your great work. I'm happy to contribute and make it even better.

nachoparker, 2017-09-04:
It is a bit older, but it is NC12.

> The web interface showed me the login dialog. Is there anything I was overlooking.

By default, the login credentials are user "admin" and password "ownyourbits". You can get information about NextCloudPi [here](https://ownyourbits.com/2017/02/13/nextcloud-ready-raspberry-pi-image/)

> Thanks for your great work. I'm happy to contribute and make it even better.

Great!  Testing/helping to port to other boards is already great help! 

Looking forward to the feedback

nachoparker, 2017-11-15:
The containers have been refactored and updated, with ncp-web and more stuff (also for x86 now).

https://ownyourbits.com/2017/11/15/nextcloudpi-dockers-for-x86-and-arm/

closed 2017-11-15 by nachoparker