nextcloud/nextcloudpi #180: webUi Stopped working

deljones, 2017-09-04:
Hi All.

After cleaning up multiple mounted drives I now have a clean and fully working instance. Two drives, 1. myCloudDrive which has ncdata, ncdatabase and swap, the second drive, ncbackup is where I back up my instance upto the default 4.. All good and everythig restored. However I've lost the webUI at https://myLocalAddress:4443. I just get a 'can't connect' also if I try to check for updates I get an error as attached.

[image]

nachoparker, 2017-09-05:
did you remove the `/home/www` directory by any chance?

nachoparker, 2017-09-05:
If you did somehow delete `/home/www`, you can recreate with

```
mkdir /home/www -p
  chown www-data:www-data /home/www
  chmod 700 /home/www

  cat > /home/www/ncp-launcher.sh <<'EOF'
#!/bin/bash
DIR=/usr/local/etc/nextcloudpi-config.d
test -f $DIR/$1 || { echo "File not found"; exit 1; }
source /usr/local/etc/library.sh
cd $DIR
touch /run/ncp.log
chmod 640 /run/ncp.log
chown root:www-data /run/ncp.log
launch_script $1 &> /run/ncp.log
EOF
chmod 700 /home/www/ncp-launcher.sh
```
If you do not have access to ncp-web, try activating it from `nextcloudpi-config`, option `nc-webui`

deljones, 2017-09-05:
Np

 

Many thanks. Again… great job

 

Yes I’m a dope I did manage to delete the home folder. Must have happened when I was rmdir’s and moving stuff around, restoring and so on and in the excitement I deleted it.

 

Regarding the webUI, I’ve deactivated and reactivated it but with no joy.

 

Dj

nachoparker, 2017-09-05:
Weird. So you can access Nextcloud but not ncp-web

Let's see first that it is indeed configured in port 443

```
$ sudo apache2ctl -S
AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive globally to suppress this message
VirtualHost configuration:
*:80                   127.0.1.1 (/etc/apache2/sites-enabled/000-default.conf:1)
*:4443                 127.0.1.1 (/etc/apache2/sites-enabled/ncp.conf:2)
*:443                  pkbrister.no-ip.org (/etc/apache2/sites-enabled/nextcloud.conf:2)
ServerRoot: "/etc/apache2"
Main DocumentRoot: "/var/www/html"
Main ErrorLog: "/var/log/apache2/error.log"
Mutex ssl-stapling: using_defaults
Mutex proxy: using_defaults
Mutex ssl-cache: using_defaults
Mutex default: dir="/var/run/apache2/" mechanism=default 
Mutex watchdog-callback: using_defaults
Mutex rewrite-map: using_defaults
Mutex ssl-stapling-refresh: using_defaults
PidFile: "/var/run/apache2/apache2.pid"
Define: DUMP_VHOSTS
Define: DUMP_RUN_CFG
User: name="www-data" id=33
Group: name="www-data" id=33
```

Also, the files need to be there of course

```
sudo ls -l /var/www/ncp-web
```

deljones, 2017-09-05:
Hi NP

 

sudo apache2ctl -S

 

Gives the same as yours

 

Below is the contents of the directory

 

pi@raspberrypi:~ $ sudo ls -l /var/www/ncp-web

total 124

-rw-r--r-- 1 www-data www-data   566 Sep  5 07:25 close.svg

-rw-r--r-- 1 www-data www-data  1494 Sep  5 07:25 csrf.php

-rw-r--r-- 1 www-data www-data  1825 Sep  5 07:25 favicon.png

-rw-r--r-- 1 www-data www-data   890 Sep  5 07:25 green-circle.svg

-rw-r--r-- 1 www-data www-data  5781 Sep  5 07:25 index.php

-rw-r--r-- 1 www-data www-data  1772 Sep  5 07:25 loading-small.gif

-rw-r--r-- 1 www-data www-data   118 Sep  5 07:25 menu.svg

-rw-r--r-- 1 www-data www-data 20962 Sep  5 07:25 minified.js

-rw-r--r-- 1 www-data www-data 30530 Sep  5 07:25 ncp.css

-rw-r--r-- 1 www-data www-data  4907 Sep  5 07:25 ncp.js

-rw-r--r-- 1 www-data www-data  4456 Sep  5 07:25 ncp-launcher.php

-rw-r--r-- 1 www-data www-data  4842 Sep  5 07:25 ncp-logo.png

-rw-r--r-- 1 www-data www-data  2504 Sep  5 07:25 ncp-output.php

-rw-r--r-- 1 www-data www-data  3686 Sep  5 07:25 poweroff.svg

-rw-r--r-- 1 www-data www-data   860 Sep  5 07:25 red-circle.svg

 

 

DJ

nachoparker, 2017-09-05:
whops, my reply got lost?

Everything seems in place, so let's have a look at the logs

```
sudo tail /var/log/apache2/*.log
```

deljones, 2017-09-05:
Hi

 

Really shabby I know but the cut and paste on my phone just provides white bands instead of the copied text

 

Dj

nachoparker, 2017-09-06:
try to access ncp-web right before reading the logs. Everything seems fine with your instalation so I do not understand why it would show NC and not ncp-web

nachoparker, 2017-09-11:
have you looked at the log right after trying to access ncp-web? your web server seems to be fine, so let's look also at php

```
systemctl status php7.0-fpm
```

cottoa00, 2017-09-11:
Hello... I've had NCPi running for a few months but recently realized the UI was inaccessible. Domain name access hangs so tried IP access... looks almost like it gets the page but it's blank.

Followed some of the diags mentioned in this thread... systemctl status apache2.service and php7.0-fpm... most feedback conformed to this thread (one exception is my admin UI is not setup at :4443

I'm currently at V0.26.29 on NCPi and V 4.9.34-v7+ linux on Pi.

A look at the browser console shows msg about inability to retrieve bookmarks from my external ddns address?
"https://xxxx.ddns.net/index.php/apps/bookmarks/public/ rest/v... HTTP/2.0 500..." etc.
"Failed to retrieve bookmarks from OwnCloud"

Any diagnostic suggestions welcome.

Not sure this image post worked... on my Pi now using Firefox...
[image]

cottoa00, 2017-09-11:
Digging deeper into error.log

PHP Fatal error:  Uncaught Doctrine\\DBAL\\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [2002] No such file or directory in /var/www/nextcloud/lib/private/DB/Connection.php:61\nStack trace:\n#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\\DB\\Connection->connect()\n#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\\DBAL\\Connection->getDatabasePlatformVersion()\n#2 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\\DBAL\\Connection->detectDatabasePlatform()\n#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\\DBAL\\Connection->getDatabasePlatform()\n#4 /var/www/nextcloud/lib/private/DB/Connection.php(148): Doctrine\\DBAL\\Connection->setTransactionIsolation(2)\n#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\\DB\\Connection->__construct(Array, Object(Doctrin...\n', referer: https://192.168.1.151/index.html

cottoa00, 2017-09-11:
... mysql... can't connect to local MySQL server... upon diggin further...
● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled)
  Drop-In: /run/systemd/generator/mysql.service.d
           └─50-mariadb-$database.conf
   Active: failed (Result: exit-code) since Mon 2017-09-11 15:33:09 EDT; 39s ago
  Process: 2550 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=exited, status=1/FAILURE)
  Process: 2548 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
  Process: 2448 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP_START_POSITION=$VAR || exit 1 (code=exited, status=0/SUCCESS)
  Process: 2445 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 2442 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 2550 (code=exited, status=1/FAILURE)
   Status: "MariaDB server is down"

Sep 11 15:33:09 ecnextcloudpi mysqld[2550]: 2017-09-11 15:33:09 1989111808 [Note] /usr/sbin/mysqld (mysqld 10.1.23-Mar...50 ...
Sep 11 15:33:09 ecnextcloudpi mysqld[2550]: 2017-09-11 15:33:09 1989111808 [Warning] Can't create test file /media/USB...r-test
Sep 11 15:33:09 ecnextcloudpi mysqld[2550]: [109B blob data]
Sep 11 15:33:09 ecnextcloudpi mysqld[2550]: 2017-09-11 15:33:09 1989111808 [ERROR] Aborting
Sep 11 15:33:09 ecnextcloudpi systemd[1]: mariadb.service: main process exited, code=exited, status=1/FAILURE
Sep 11 15:33:09 ecnextcloudpi systemd[1]: Failed to start MariaDB database server.
Sep 11 15:33:09 ecnextcloudpi systemd[1]: Unit mariadb.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.

nachoparker, 2017-09-12:
hi @cottoa00,

This is a different issue from OP, so you'd better open a new thread.

Did you try rebooting?

Output of 

```
ls -l /media
```

and

```
grep datadir /etc/mysql/mariadb.conf.d/50-server.cnf
```

nachoparker, 2017-09-12:
I'll close now. @deljones , we can continue talking about your case still

closed 2017-09-12 by nachoparker