nextcloud/nextcloudpi #181: Mount point of /dev/sda1 changes on boot

ObfuscateTheWeb, 2017-09-05:
My datadir is in /media/USBdrive

When I reboot, /dev/sda1 will be mounted to /media/myCloudDrive1 

I have to do `sudo umount /media/myCloudDrive1` then `sudo mount /dev/sda1 /media/USBdrive` everytime I reboot

How do I make it so that it auto mounts to /media/USBdrive on boot?
It wasn't always like this, I think the updates somehow introduced something new, like a bug

nachoparker, 2017-09-05:
Hello,

Unfortunatedly, nc-automount was not working well with Raspbian 9, so I decided to rewrite it completely.

But you should have a symlink to /media/USBdrive, so there is something strange.

Can you do the following steps for me?

1. Reboot
2. What's the output of `systemctl status nc-automount-links` and `ls -l /media`
3. After that, try to run it `sudo systemctl start nc-automount-links`
4. And again `systemctl status nc-automount-links` and `ls -l /media`

Thanks

nachoparker, 2017-09-05:
Ok, nevermind those instructions.

I was investigating myself and I found a bug. Please, update to `v0.26.11` and reboot again.

If you do `ls -l /media`, you should have the link named `/media/USBdrive`

For instance, I have 2 USB drives plugged in and this is how it looks

```
$ ls -l /media            
total 6                      
drwxr-xr-x 1 root root 2048 Aug  1 13:15 ARCH_201708       
drwxr-xr-x 7 root root 4096 Sep  1 19:03 myclouddrive      
lrwxrwxrwx 1 root root   18 Sep  1 19:39 USBdrive -> /media/ARCH_201708                                                
lrwxrwxrwx 1 root root   19 Sep  1 19:39 USBdrive1 -> /media/myclouddrive
```

Please, reopen if the problem persists

closed 2017-09-05 by nachoparker

ObfuscateTheWeb, 2017-09-06:
Updated to `v0.26.11` and still problem persists

output of `systemctl status nc-automount-links`

```● nc-automount-links.service - Monitor /media for mountpoints and create USBdrive* symlinks
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; enabled)
   Active: active (running) since Wed 2017-09-06 01:16:13 UTC; 2min 14s ago
 Main PID: 446 (nc-automount-li)
   CGroup: /system.slice/nc-automount-links.service
           ├─446 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─469 inotifywait --monitor --event create --event delete --format...
           ├─470 grep --line-buffered ISDIR
           └─471 /bin/bash /usr/local/etc/nc-automount-links-mon
```

`ls -l /media`
```
total 20
drwxr-xr-x 4 root root 4096 Aug 10 12:10 myCloudDrive
drwxr-xr-x 2 root root 4096 Sep  2 21:17 USBdrive
drwxr-xr-x 2 root root 4096 Sep  2 21:17 USBdrive1
drwxr-xr-x 2 root root 4096 Sep  2 21:17 USBdrive2
drwxr-xr-x 2 root root 4096 Sep  2 21:17 USBdrive3
```

Ran `sudo systemctl start nc-automount-links` and then do `systemctl status nc-automount-links` 

```
● nc-automount-links.service - Monitor /media for mountpoints and create USBdrive* symlinks
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; enabled)
   Active: active (running) since Wed 2017-09-06 01:16:13 UTC; 5min ago
 Main PID: 446 (nc-automount-li)
   CGroup: /system.slice/nc-automount-links.service
           ├─446 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─469 inotifywait --monitor --event create --event delete --format...
           ├─470 grep --line-buffered ISDIR
           └─471 /bin/bash /usr/local/etc/nc-automount-links-mon
```
`ls -l /media`
```
total 20
drwxr-xr-x 4 root root 4096 Aug 10 12:10 myCloudDrive
drwxr-xr-x 2 root root 4096 Sep  2 21:17 USBdrive
drwxr-xr-x 2 root root 4096 Sep  2 21:17 USBdrive1
drwxr-xr-x 2 root root 4096 Sep  2 21:17 USBdrive2
drwxr-xr-x 2 root root 4096 Sep  2 21:17 USBdrive3
```

zombie5320, 2017-09-06:
I having the same issue. Updating did not help, i've tried multiple different drives but nothing

nachoparker, 2017-09-06:
I am puzzled. Why would it work for me and not for you? I think that the problem is that you have directories that are not symlinks ( like /media/USBdrive ), so the symlink with that same name cannot be generated.

Try this
```
sudo ncp-update
sudo umount /media/*
sudo rmdir /media/*
```

ObfuscateTheWeb, 2017-09-06:
Did all the other stuff, output of `systemctl status nc-automount-links`

```
● nc-automount-links.service - Monitor /media for mountpoints and create USBdrive* symlinks
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; enabled)
   Active: active (running) since Wed 2017-09-06 10:46:34 UTC; 1min 28s ago
 Main PID: 445 (nc-automount-li)
   CGroup: /system.slice/nc-automount-links.service
           ├─445 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─470 inotifywait --monitor --event create --event delete --format...
           ├─471 grep --line-buffered ISDIR
           └─472 /bin/bash /usr/local/etc/nc-automount-links-mon
```

nachoparker, 2017-09-06:
and, still no links? `ls -l /media`?

ObfuscateTheWeb, 2017-09-06:
No links, output of `ls -l /media`

```
total 4
drwxr-xr-x 4 root root 4096 Aug 10 12:10 myCloudDrive
```

nachoparker, 2017-09-06:
I just tried with a fresh image and it works

Try this command and let's see if the links appear

```
/usr/local/etc/nc-automount-links
ls -l /media
```

ObfuscateTheWeb, 2017-09-06:
This works and creates the link ! What went wrong before?
```
total 4
drwxr-xr-x 4 root root 4096 Aug 10 12:10 myCloudDrive
lrwxrwxrwx 1 root root   19 Sep  6 13:49 USBdrive -> /media/myCloudDrive
```

nachoparker, 2017-09-06:
yeah, that's the missing piece. I am starting to think that this happens because you are still in an old image. I cannot replicate, so I suggest you try the following

```
sudo /usr/local/etc/nc-automount-links
sudo systemctl disable nc-automount-links
```

ObfuscateTheWeb, 2017-09-06:
Result:
```
Removed symlink /etc/systemd/system/multi-user.target.wants/nc-automount-links.service.
```
And how am I still in an old image, is there like a cache or something? Isn't updating via `sudo ncp-update` enough?

nachoparker, 2017-09-06:
> And how am I still in an old image, is there like a cache or something? Isn't updating via sudo ncp-update enough?

Should be, but there can be surprises of course. This is an example. When that happens, I need your help to fix it for other people. Exactly like we are doing now  ;)

Does the link survive a reboot now?

ObfuscateTheWeb, 2017-09-06:
I'm not complaining, just curious that's all ;)

Oh yes, the two lines did the trick, it now survives a reboot, so I guess this means another new version of nextcloudpi that patches this? Lol

nachoparker, 2017-09-06:
> I'm not complaining

lol, I know :)

I already pushed a patch that I think could fix it but I am not 100% sure. Just in case, don't activate automount again because it will remove my dirty fix.

If it's still an issue somebody else will complain, you have been off your cloud enough ;)

ObfuscateTheWeb, 2017-09-06:
Well, it's fixed for now, thanks @nachoparker  and thanks for your work and prompt responses !
Probably one of the best support I've seen so far

BETEP-ok, 2017-12-25:
sorry to say, but I also have issues with the fresh install of NextcloudPi

after fresh install nd configuration- every time I reboot- the symlink is lost.
I still can see my usb drive as /media/myCloudDrive
but /media/USBDrive is just not there.

good enough the trick with /usr/local/etc/nc-automount-links worked and symlink popped up, but it doesn't seem like a bulletproof solution.

I don't want to type in ssh commands after every powerloss ))) 

nextcloudpi ver 0.43.1 (the latest as of now)

ernesst, 2018-08-05:
I was having the similar issue after a power failure.
It seems to use the package udiskie, which has is cache corrupted, try to run udiskie-info to see
After un `sudo find /usr -name '*.pyc' -delete ` and a reboot it works.