nextcloud/nextcloudpi #182: Wrong size for big folders

rottw75, 2017-09-05:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v0.26.6
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
NextCloudPi_08-20-17
### Expected behavior
Web UI to show correct user quota
Web UI to show correct folder size
### Actual behaviour
When adding folders with big content (I tried 4.5 GB), the content shown for the folder is wrong and the user quota shown on the page is wrong
### Steps to reproduce, from a freshly installed image
I have assigned to a user a quota of 200 GB on an external USB disk.
I copy from an NFS mount some big folders (4.5 GB) into the nextcloud data folder for that user
I fix ownersip/group to www-data
I run the scan file from the NCPi web admin page, no errors on screen
Check the user web page:
the used size is wrong (expected 5.5 GB, says 1.9)
the user quota is wrong (expected 200 GB, says 196)
the folder size is wrong (expected 5,5 GB, says 0.7)
[image]
[image]
[image]

### Include logs
<details>
<summary>Nextcloud logs</summary>

Level | App | Message | Time
-- | -- | -- | --
Error | index | OCP\Files\NotFoundException: | 2017-09-05T06:58:56+0200
Error | PHP | touch(): Unable to create file /nextcloud/data/appdata_ocl1ces56tzv/css/theming/06e729704af0add597048a63e3bc9cb9-theming.css.gzip because No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#199 | 2017-09-05T06:58:56+0200
Error | PHP | touch(): Unable to create file /nextcloud/data/appdata_ocl1ces56tzv/css/theming/06e729704af0add597048a63e3bc9cb9-theming.css.deps because No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#199 | 2017-09-05T06:58:56+0200
Error | PHP | touch(): Unable to create file /nextcloud/data/appdata_ocl1ces56tzv/css/theming/06e729704af0add597048a63e3bc9cb9-theming.css because No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#199 | 2017-09-05T06:58:56+0200
Error | index | OCP\Files\NotFoundException: | 2017-09-05T06:58:14+0200
Error | PHP | touch(): Unable to create file /nextcloud/data/appdata_ocl1ces56tzv/css/theming/06e729704af0add597048a63e3bc9cb9-theming.css.gzip because No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#199 | 2017-09-05T06:58:14+0200
Error | PHP | touch(): Unable to create file /nextcloud/data/appdata_ocl1ces56tzv/css/theming/06e729704af0add597048a63e3bc9cb9-theming.css.deps because No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#199 | 2017-09-05T06:58:14+0200
Error | PHP | touch(): Unable to create file /nextcloud/data/appdata_ocl1ces56tzv/css/theming/06e729704af0add597048a63e3bc9cb9-theming.css because No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#199 | 2017-09-05T06:58:14+0200
Error | index | OCP\Files\NotFoundException: | 2017-09-05T05:50:16+0200
Error | PHP | touch(): Unable to create file /nextcloud/data/appdata_ocl1ces56tzv/theming/9/icon-core-filetypes_text.svg because No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#199 | 2017-09-05T05:50:16+0200
Warning | core | Trusted domain error. "193.63.58.83" tried to access using "87.4.28.109" as host. | 2017-09-05T05:01:01+0200
Error | PHP | file_get_contents(/nextcloud/data/appdata_ocl1ces56tzv/theming/images/background): failed to open stream: No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#209 | 2017-09-04T23:09:55+0200
Warning | core | Login failed: 'mauro' (Remote IP: '82.54.105.254') | 2017-09-04T22:43:26+0200
Warning | core | Trusted domain error. "193.205.213.82" tried to access using "mfgehueller.hopto.org" as host. | 2017-09-04T17:17:06+0200
Error | PHP | OC\HintException: Memcache OCMemcacheAPCu not available for local cache at /var/www/nextcloud/lib/private/Memcache/Factory.php#96 | 2017-09-04T15:53:40+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:51:12+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:51:12+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:51:07+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:51:01+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:51:01+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:51:01+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:51:01+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:51:01+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:51:00+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:57+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:57+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:57+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:57+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:57+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:56+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:51+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:51+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:50+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:50+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:50+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:49+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:32+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:31+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:31+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:31+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:31+0200
Warning | no app in context | Temporary directory /nextcloud/tmp is not present or writable | 2017-09-04T14:50:28+0200
[... 97 lines cut ...]

</details>
<details>
<summary>mariaDB logs</summary>
Paste the output of `systemctl status mysqld`
 mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2017-09-04 20:01:31 CEST; 11h ago
  Process: 4379 ExecStartPost=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 4376 ExecStartPost=/etc/mysql/debian-start (code=exited, status=0/SUCCESS)
  Process: 4346 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
  Process: 4248 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environmen
  Process: 4244 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 4241 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 4349 (mysqld)
   Status: "Taking your SQL requests now..."
   CGroup: /system.slice/mariadb.service
           └─4349 /usr/sbin/mysqld

set 04 20:01:18 mauro-pi systemd[1]: Starting MariaDB database server...
set 04 20:01:30 mauro-pi mysqld[4349]: 2017-09-04 20:01:30 1989152768 [Note] /usr/sbin/mysqld (mysqld 10.1.23-MariaDB-9+deb9u1) starting as process 4349 ...
set 04 20:01:31 mauro-pi systemd[1]: Started MariaDB database server.

```
</details>

JimmyKater, 2017-09-05:
it's clearly a problem of nextcloud not of nextcloudpi.
pls refer to https://help.nextcloud.com/t/update-12-size-of-cloud-data-wrongly-displayed/19370 
and/or here https://help.nextcloud.com/t/files-folders-wrong-or-pending-size/14133

nachoparker, 2017-09-05:
Hello,

Yes, like says it is a known issue of the NC server. I will mark as external bug. You can try applying the correction manually if you want.

Thanks @JimmyKater

closed 2017-09-05 by nachoparker