nextcloud/nextcloudpi #183: nc-autoupdate-ncp doesn't work

JimmyKater, 2017-09-05:
hey nacho

now that signalling updates for ncp-web work beautifully for other users than admin it came apparent that nc-autoupdate-ncp doesn't seem to work the way i expected it.
i got signalled that there was an update waiting (26.9) and where/how i could update it. but it didn't got autoupdated as expected. 
do you have a waiting time implemented that needs to pass between signalling and autoupdating or what else went wrong?

cheers
jimmy

nachoparker, 2017-09-05:
> now that signalling updates for ncp-web work beautifully for other users than admin

Good to know. So I guess you tried again after I made it an option to choose the user.

The way it is implemented now, it can take maximum another day from the moment NCP realizes that there is an update until the autoupdate kicks in. 

For me it has been working, but you never know... for how long did you wait?

This all means that it can take maximum two days between when a new version is available and when autoupdate installs it.

nachoparker, 2017-09-05:
ahhhhhhhh

But looking at the code again I realize that we are in the same case as before... only user admin gets notified.

I'll improve that the same we did with the other notifications

Thanks for your testing. Is very helpful

nachoparker, 2017-09-05:
I'll close. Please reopen if the update + notification don't come up in more than 2 days.

closed 2017-09-05 by nachoparker

JimmyKater, 2017-09-05:
will wait for at least 2 days from the next notification and tell you then.

JimmyKater, 2017-09-07:
so it updated automatically after 24 hrs (max).
it didn't update 26.9 to 26.11 (like the announcement told me there was such an update) but to 26.16 (without any notification about having any further update than 26.11)
anyways... it updated automatically ;)

nachoparker, 2017-09-07:
great! Yeah, it just updates to the latest thing

There's room for improvement, for instance we could check more often for updates but at least it works ;)

Thanks for your feedback