nextcloud/nextcloudpi #185: 26.11 disk mounting

deljones, 2017-09-05:
Had my two disks all cleaned up
1. myCloudDrive for ncdata, swap and database
2. ncbackup for backups

Applied 26.11 now my myCloudDrive has only the swap and a new USBdrive has appeared with my ncdata, swap and database.:(

All the data is there on the disks but NC wont open now as the links have all changed.

What am I doing wrong here? How can this be fixed?

DJ

deljones, 2017-09-05:
Fixed...

Just went into the db config file and reset database location followed by NC config.php to change location there..

D

nachoparker, 2017-09-05:
Hi Del,

There was a bug with the new implementation, where the links for /media/USBdriveX were not being generated. ( #181 )

They are just links to the real mount points which are `/media/myCloudDrive` and `/media/ncbackup`, so you can use `/media/USBdrive` if you are only going to plug one drive in. 

Still you have the right setup now, which uses label names, but now the only change is that the links appeared.

You can see that by doing 

```
ls -l /media
```

Sorry about that confusion again. This is the final arrangement and won't change again because it uses a more stable interface.

closed 2017-09-05 by nachoparker

zombie5320, 2017-09-05:
I've updated but links still fail to appear. I have used multiple different drives with no success. What could I be doing wrong. 

ls -l /media outputs:

total 8
drwxr-xr-x 3 root root 4096 Sep  5 21:15 CloudDrive
drwxr-xr-x 3 root root 4096 Sep  5 21:41 USBdrive

nachoparker, 2017-09-06:
hi zombie,

try this

```
sudo ncp-update
sudo umount /media/*
sudo rmdir /media/*
```

then, go to

```
sudo nextcloudpi-config
```
or the web interface, and activate `nc-automount`. Do it again even if it says it's already active.

After that, reboot and if it still doesn't show the links, show me the output of

```
systemctl status nc-automount-links
```

nachoparker, 2017-09-06:
let's continue the discussion in #181