nextcloud/nextcloudpi #186: Port Forwarding

albrechtar, 2017-09-06:
I have a question with regards to port forwarding.  I followed your instructions and when I attempt to check my ports are forwarding I get a timeout error.  I am using a NetGear X8 with DD-WRT flashed on it.

  I forwarded both 80 and 443 to my static IP for the pi.  I contacted my ISP to ensure they are not blocking any ports (they informed me they are not).  I am not sure what I am doing wrong. I am running a modem/router in full bridged mode and my router behind it.

  I am certain I have set up the port forwarding rules properly with the name (can be anything as I understand), the port to forward from i.e. 80 and the port to forward to i.e. 80, and then the static ip of my pi.

  I used sudo reboot for my pi, and rebooted my router after clicking enable on the port forwarding rule and apply changes.

  ny guidance or information you may have would be greatly appreciated.  Also may I use a RAID drive as my drive and just plug it into the USB port on my pi (it has its own power source of course).  Thank you, and this was by far the best tutorial I have seen for setting up NextCloud with my Pi.

AL

### What version of NextCloudPi are you using? ( eg: v0.17.2 )

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

### Expected behavior

### Actual behaviour

### Steps to reproduce, from a freshly installed image

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -n 100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -n 100 /var/log/mysql/*.log`
```
</details>

Pant, 2017-09-06:
On my Router, under the Port Forwarding tab, there is an option to select the WAN Connection. I have to use PTM_DSL (not ATM_DSL). Also there are two fields to specify the WAN Host IP Range. I have to set this to 0.0.0.0 - 0.0.0.0 , so that it forwards everything to my Raspberry Pi.

Share if you can what are your other options on your Router web interface.

nachoparker, 2017-09-06:
you can try this to see if your ports are really open or not

http://www.yougetsignal.com/tools/open-ports/

albrechtar, 2017-09-06:
Thank you Pant and Nacho for your reply.  Please see the screen shots.  I am using DD-WRT Kong build on a NetGear R8500 router.  Please see these screen shots I also have both my pi MAC address set to static (the Wifi MAC and the LAN MAC).
[image]
[image]
[image]
[image]
[image]

nachoparker, 2017-09-06:
I don't know if you will need a 'source address'.

Did you check this? http://www.yougetsignal.com/tools/open-ports/

In any case, you can also activate uPnP and try have NextCloudPi open the ports for you, using `nc-forward-ports` in the web interface or nextcloudpi-config

albrechtar, 2017-09-06:
Yes I used the link it does not show anything and I attempted to let NextCloudPi open the ports it places them into the uPnP ports list and then I get the same results when I use the site to check.

nachoparker, 2017-09-06:
uPnP active in the router config?

albrechtar, 2017-09-06:
yes

albrechtar, 2017-09-06:
You mention to close it after your finished (in your instructions) I assume you mean after it attempts to open the ports.  Or should that be left open indefinitely?

nachoparker, 2017-09-06:
it is just a helper. some routers will close the ports again after disabling upnp. Others will leave them configured.

It is not considered secure, so it is better to have it disabled

The best approach is to do it manually, like you are trying to do.

I am not very familiar with that particular router, but your config seems correct to me. I think that the problem must be somewhere else.

albrechtar, 2017-09-06:
Yeah, I have been reading all day and I believe it was correct also.  But I am new to this, so was not sure if perhaps I made a mistake.  But I think your detailed instructions are rather easy to follow along with. Do you think I should reflash my SD card and start over?

Pant, 2017-09-06:
Try to configure port forwarding in the Port Range Forwarding tab following these [steps](https://github.com/nextcloud/nextcloudpi/issues/186#issuecomment-327459502) from the dd-wrt wiki.

albrechtar, 2017-09-06:
Ill try that also Pant, I attempted that earlier as I read that some people had success.  I did not but I will try it once more to make sure.  Thank you both for trying to help me out, I really appreciate it.

nachoparker, 2017-09-06:
> Do you think I should reflash my SD card and start over?

No, this has nothing to do with NextCloudPi, but with your network configuration

albrechtar, 2017-09-06:
Okay, this may seem like a stupid question but if the pi is running with NextCloudPi installed then it is running provided I followed your steps.  I was reading that if there is nothing running on teh port it could show as not open.  I have the NextCloudPi just booted up attached via Ethernet cable to my router and running.  I did not go past the step in your instructions once it instructed me to open the ports.

albrechtar, 2017-09-06:
Pant I attempted using the range forwarding with no better results.

nachoparker, 2017-09-06:
> was reading that if there is nothing running on teh port it could show as not open.

show where?

At this point you can access NextCloudPi and try it, opening the ports is for accessing from internet (from outside)

albrechtar, 2017-09-06:
When you check if the port was open or closed, with the link you provided.

nachoparker, 2017-09-06:
ah, ok. Not our case

nachoparker, 2017-09-06:
I'll close because it's not directly related to NCP. But we can continue talking

closed 2017-09-06 by nachoparker

albrechtar, 2017-09-06:
OK, Nacho any ideas on what to do next to resolve the forwarding issue?  Other than that this seems to work fine a bit slow, but I think that may also be in part to my drive I am currently using.  Would it be possible to plug in a RAID drive directly into the pi or would that cause issues?  Thank you again for your help..

nachoparker, 2017-09-06:
The first time it is slow becauses it initializes and catches things.

RAID performance is very bad on the Pi, because it shares network and USB in one PCIe lane. This means now you have three things trying to write simultaneously in one bus (2 USB plus network)

Read this https://github.com/nextcloud/nextcloudpi/issues/107

nachoparker, 2017-09-06:
> OK, Nacho any ideas on what to do next to resolve the forwarding issue?

Contact your provider. Maybe try to open another port first, like 8085 and use the web I gave you to see if that is blocked as well or not

Pant, 2017-09-06:
>  ...any ideas on what to do next to resolve the forwarding issue?

Maybe you have to set up something at the modem which is closer to the public. There may be a firewall which blocks, for example, ports 1 - 1024.

albrechtar, 2017-09-06:
Ok I will try that, my provider said they block no ports.  I did just notice that when I look at status on my router for ip it shows 100.120.132.241  and when I go to google and look at my IP it shows 110.169.139.67 should those not match?

albrechtar, 2017-09-06:
And my ISP provides equipment so in order to use my own router I had to request they bring a CISCO router/Modem that supports full bridged mode.  It is placed in bridged mode and I cant access it (or at least not with an ip, I was reading that is normal as it no longer has an ip).  Sorry I do not think I mentioned that earlier.  But if it is in bridged mode that should not have any effect correct?  Thanks again for all your help.

Pant, 2017-09-06:
Ok so, please check the firewall settings at the Router (not the Modem/Router).

albrechtar, 2017-09-06:
Just spoke to my ISP and they said no ports are blocks, and they confirmed  that if the CISCO modem/router is placed into bridged mode its only working as a modem.

albrechtar, 2017-09-06:
Ok Pant Ill send you some screen shots, am i looking for anything specific?

Pant, 2017-09-06:
As I see in DD-WRT wiki the firewall doesn't have a gui. Is there a way to disable it temporarily to see if that what blocks incoming connections?

albrechtar, 2017-09-06:
I think so but IM not 100% sure what SPI firewall is?

[image]
[image]

Pant, 2017-09-06:
I think you may have to disable "Filter WAN NAT Redirection". Try disabling that.

albrechtar, 2017-09-06:
I tried to click disable and applied changes and then tested the ports again with the site Nacho provided but still am unable to get through.

albrechtar, 2017-09-06:
Dont know if this helps at all as far as providing you with any useful information.  Here is the build information for my router firmware.

[image]

Pant, 2017-09-06:
Try disabling the "SPI Firewall"

albrechtar, 2017-09-06:
Tried that with no luck :(

Pant, 2017-09-06:
Try turn off UPnP

albrechtar, 2017-09-06:
Got it off.  Pant does this difference in ip's mean anything?  My router status page shows one ip but when I use the page to check the port forwarding it shows another?  100.120.132.241 (router) and 110.169.139.67 (from the port forwarding check site).

albrechtar, 2017-09-06:
[image]
[image]

Pant, 2017-09-06:
Recheck with this: whatismyipaddress.com.

Pant, 2017-09-06:
Also check the official [DD-WRT Troubleshooting with Port Forwarding](https://www.dd-wrt.com/wiki/index.php/Port_Forwarding_Troubleshooting#Basic_Troubleshooting)

albrechtar, 2017-09-06:
I think the next thing to try is a hard reset and reconfigure the router reading the official wiki?

Pant, 2017-09-06:
Yes I would suggest that.

albrechtar, 2017-09-06:
Pant thank you again for all your help, tomorrow I will do a hard reset and see if that corrects it since according to the wiki that may correct it.  I also spoke again to my ISP and they reset my connection and verified no ports are being blocked.  I just made copies of all the pages so I can enter everything in.  The wiki said do it from scratch im guessing if I made a backup and reinstalled it if there was an underlying issue it would just be there again.  Again thank you both I appreciate all you and Nacho have done to help me out.

  I will let you both know how I make out tomorrow.  Is there a specific drive speed that you recommend for best performance.  Since my RAID drive I currently use would not be good performance I will go get another one.  IS it ok to plug in say 2 drives as long as they are powered?

  Take care guys and have  nice afternoon.  Thanks again.

nachoparker, 2017-09-06:
good luck guys

thanks @Pant

albrechtar, 2017-09-06:
Thanks Nacho Ill keep you updated, maybe my troubles can help someone else out :)

albrechtar, 2017-09-06:
I just noticed something, when I was taking screen shots to have as reference resetting my router.  I have a VPN (I live and work overseas) and I had static IPs set up so that only my streaming devics go through the VPN.  I noticed on the router they no longer show in the list, yet those devices are still maintaining static ips.  Just thought that was odd and maybe there is something a bit buggy.  Anyway just thought I would share that with you guys.

Pant, 2017-09-06:
You will have to dissable VPN in order for this to work. Imagine VPN as a third router that needs to have Port Forwarding on your Router and the Router Port Forward to your Raspberry Pi.

albrechtar, 2017-09-06:
I did not think of that but makes sense, thank you again Pant.

albrechtar, 2017-09-06:
@Pant  is it possible to change the ports to a higher number so the VPN provider is able to open them? Would that then allow the VPN to still run?  And reading more that makes sense now why I was getting the two ips.

Pant, 2017-09-06:
Technically? It is possible. The VPN provider will have to port forward to your ip his public ports 80 and 443. That said I don't think he is going to do that.

albrechtar, 2017-09-07:
@Pant Yes I spoke to the VPN provider they can port anything as long as it is above 2048.  But Ill worry about that next first Ill get it running normally without the VPN and then we can work the VPN.  Once I am home today I will try and reset the router.

albrechtar, 2017-09-08:
@nachoparker and @Pant.  Ok I finally got this working, very strange (or at least to me).  I will send you some screen shots.  So I did the following:

1.  Verified with my ISP they are not blocking any ports.
2.  Hard reset my DD-WRT router.
3.  Reconfigured router (without firewall and without any VPN (as we already know that would create an issue as Pant had mentioned).
4.  Still had issues.  I again noticed my WAN IP in my router was diffrent from what shows up when I attempt to do a port test. I used port test on my laptop (mac network utility) and used the WAN address my router showed with success.  
5.  I then updated my noip with that address and attempted to access NCloud although slow it worked. 

I am unsure why the IPs are different?  Should my public IP not be the same as my WAN IP? I also requested a static IP from my ISP and they should call back in an hour (or so they say).  This should then alleviate the need for DDNS correct?

I just wanted to get back with you and get your thoughts on that, as well and more importantly to update you in the evening someone else has that issue.  I really do appreciate all your help guys.  You are providing a great service and invaluable for anyone with cloud storage needs.

[image]
[image]
[image]
[image]
[image]

nachoparker, 2017-09-08:
great, good job!

nachoparker, 2017-09-08:
> This should then alleviate the need for DDNS correct?

Correct, but it is still nice to be able to access by domain name, even if your IP doesn't change.

albrechtar, 2017-09-08:
@nachoparker Thank you.  I still am not sure why the different IPs...  Again thank you for all your help, as I know you didint need to help on this as it was not directly related to NCloudPi

albrechtar, 2017-09-08:
@nachoparker @Pant 

Guys another update, so although I thought I was good to go I was not.  I did a traceroute and noticed I had two 10.X.X.X hops after leaving my router.  So I was thinking perhaps double NAT as 10 is a private address.  Called back to my ISP, and got a tech that was pretty sharp.  He knew exactly what was going on.  He said they are still providing NAT from thier side it needs to be disabled they should have done it when you placed our router into bridged mode.  Reset my router and their modem.  Checked and both my router WAN and my public IP match now.  Checked my port on the site you provided and low and behold it works.  

Before although I was using my domain name to access it was still only local when I attempted access via my phone it failed.  Again guys thanks.  I just wanted to update again so you and Pant do not have to go through all this with someone else in the future. It is working great now.

Pant, 2017-09-08:
You are welcome! Glad we've helped!

albrechtar, 2017-09-09:
@Pant @nachoparker Hello guys, sorry to trouble you again.  I am trying to put SSL certs onto NCloudPi and I have both txt files domain-key and domain-crt.  I am unsure exactly what I should do with them.  I think I need to update the /etc/apache2/sites-available/nextcloud.conf file and change the file names (that currently point to the snakeoil pem and key?  And then do I need to go into the /etc/ssl/private dir but then I am unsure if I would just create a new file there and if so which file should be associated to the crt, or the key?  The crt file looks just like it has two keys in the same file.

I was able to get the files with DNS verification, that was rather easy.  I would also need to change my ports but I think I found the answer for that on the netcloud forums.  So I will need to update my certs manually which im ok with, once I understand how to properly do that :)

Changing port settings:
Edit /etc/apache2/ports.conf
a. Change the port numbers as required or add new
Edit /etc/apache2/sites-available/yournextcloud.conf
a. Change 80/443 there to the ports in <VirtualHost *:80> (or 443) to what you specified in 1a.
Restart Apache: sudo service apache2 restart

Hope you both are enjoying your weekend.

nachoparker, 2017-09-10:
> I was able to get the files with DNS verification

for let's encrypt? 

Nice, can you share an outline of the steps?

albrechtar, 2017-09-10:
Sure let me put something together for you that would be easy for folks to
follow.  If I needed to change ports I would be able to do so, in the end I
was able to use the NCloudPi option.

[8 more replies]