nextcloud/nextcloudpi #187: I did a wrong update | nc-update before dist-upgrade :-(

steve-ohh, 2017-09-06:
Sorry, 
i didn't read the blog or anything, just started `ncp-update`, it failed. So i made an normal `sudo apt-get update` and `sudo apt-get upgrade`. after that again `ncp-update` failed. then i read about upgrade to stretch. i did that but `nc-update` still doesn't work:
`Downloading updates
Performing updates
Activating unattended-upgrades
Unattended upgrades active: yes (autoreboot yes)
./update.sh: Zeile 76: /home/www/ncp-launcher.sh: Datei oder Verzeichnis nicht gefunden
chmod: Zugriff auf '/home/www/ncp-launcher.sh' nicht möglich: Datei oder Verzeichnis nicht gefunden
NextCloudPi updated to version v0.26.11
`
it's german
`./update.sh: Zeile 76: /home/www/ncp-launcher.sh: Datei oder Verzeichnis nicht gefunden`

> ./update.sh: row 76: /home/www/ncp-launcher.sh: File or directory not found

`chmod: Zugriff auf '/home/www/ncp-launcher.sh' nicht möglich: Datei oder Verzeichnis nicht gefunden`

> chmod: access '/home/www/ncp-launcher.sh' not possible: file or directory not found`

 I know it's my bad because I missed to read before update :-( Can you please help anyway?

nachoparker, 2017-09-06:
You have an image that is quite old. I would consider starting from scratch ( backup and restore nextcloud ).

In any case, to make updates and ncp-web work for you try pasting the following code in the terminal

```
mkdir /home/www -p
  chown www-data:www-data /home/www
  chmod 700 /home/www

  cat > /home/www/ncp-launcher.sh <<'EOF'
#!/bin/bash
DIR=/usr/local/etc/nextcloudpi-config.d
test -f $DIR/$1 || { echo "File not found"; exit 1; }
source /usr/local/etc/library.sh
cd $DIR
touch /run/ncp.log
chmod 640 /run/ncp.log
chown root:www-data /run/ncp.log
launch_script $1 &> /run/ncp.log
EOF
  chmod 700 /home/www/ncp-launcher.sh
  echo "www-data ALL = NOPASSWD: /home/www/ncp-launcher.sh , /sbin/halt" >> /etc/sudoers
```

steve-ohh, 2017-09-06:
would have done that already, but unfortunately nc-backup isn't working anymore :cry: 

```
Launching nc-backup
Error reading data directory. Is NextCloud running and configured?
Done. Press any key...
```

INCLUDEDATA=no

steve-ohh, 2017-09-06:
i moved my data directory to /media/USBdrive

nachoparker, 2017-09-06:
great!

closed 2017-09-06 by nachoparker