nextcloud/nextcloudpi #188: test very big backups

nachoparker, 2017-09-06:
It would be great to see if the Raspberry Pi struggles with huge backups.

Maybe 5GB? maybe 10GB? 100GB? 400GB?

It would be interesting to see what current limits are.

Has anybody done this before?

nachoparker, 2017-09-08:
Cool, thanks!

Let us know what you find.

> I also tried to backup including data once and switch over to backups not including data afterwards, but unfortunately this results in the backup containing all data being deleted.

Yes, in reality data is not deleted, but in that situation, NC points your datadirectory back to the defaults, which is empty. If you have your original datadir around you can still point NC to it and then run `nc-update` so that NC learns of the new files.

> on a sidenote: Whenever I start the backup process containing data, after a few seconds an error message appears:

I have seen that before too. That happens when you have the nextcloud web open while backing up. Fixed in `v0.26.22`

nachoparker, 2017-09-20:
any progress with this?

HrJosch, 2017-09-21:
My findings until now:
1) Manually created backups >100G seem to work flawlessly, except for the fact that the limit of backups is not being respected. I have it set to 4, shouldn't the oldest backup being deleted when it comes to creating backup number 5? 

`Launching nc-backup
Maintenance mode enabled
backup database...
backup base files...
backup data files...
tar: /media/USBdrive2/Backup/nextcloud-bkp_20170921.tar: Nur 4096 von 10240 Bytes geschrieben
tar: Error is not recoverable: exiting now
error generating backup
Nextcloud is in maintenance mode - no app have been loaded

Maintenance mode disabled
Done. Press any key...`

I would assume this is happening because USBdrive2 running out of free space due too the oldest backup not being deleted.

2)Still experimenting around with periodic backups. 
Having BACKUPDAYS set to 7 should result in a backup being written every day, or once a week? 
In fact none of my backup files has been scheduled. They were all initiated manually.

deljones, 2017-09-21:
The periodic backup... The time the backup takes place... Is it the time you make the changes, create the job or....?

Dj

nachoparker, 2017-09-21:
>  2)Still experimenting around with periodic backups.
Having BACKUPDAYS set to 7 should result in a backup being written every day, or once a week?
In fact none of my backup files has been scheduled. They were all initiated 

It means one backup every week.

> I have it set to 4, shouldn't the oldest backup being deleted when it comes to creating backup number 5?

Looking at the [code](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-backup.sh#L68) 

It does the fifth backup and _after that_, it checks if we are over the limit and removes the older one(s). I guess I'll fix it to check/cleanup first, and backup later. It makes more sense.

Great feedback, thanks!

nachoparker, 2017-09-21:
> The periodic backup... The time the backup takes place... Is it the time you make the changes, create the job or....?

From last time it ran. You can check when the last one was and when the next one is going to be with the command

```
systemctl list-timers
```

deljones, 2017-09-21:
Thanks

Dj

nachoparker, 2017-09-21:
anytime ;D

deljones, 2017-09-21:
Is there a way to influence this time? Cron edit? Would like to run the back up overnight

Dj

nachoparker, 2017-09-21:
That would need to be implemented. 

It doesn't use cron, but systemd timers, which are easier to manage and more powerful for recurring tasks.

If you want to do it manually, you can modify the file [nc-backup.timers](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-backup-auto.sh#L108) to control when it happens. You can set it sunday  at night at a certain time for example.

It's called 'calendar events' and [here is the syntax](https://www.freedesktop.org/software/systemd/man/systemd.time.html#Calendar%20Events)

lsousa, 2017-09-21:
Sorry for the kind of offtopic, but this could be useful for future though. Instead of tar, the backup could be made with a deduplication backup software, to save some space. 
I'm using BorgBackup (https://borgbackup.readthedocs.io/en/stable/)  to backup and upload to a 2nd rPi in another location and so far it's working great. This way I have daily backups without increasing the storage space as much, this was my last report:

`                       Original size      Compressed size    Deduplicated size
This archive:               79.70 GB             79.71 GB              1.88 MB
All archives:              795.95 GB            796.01 GB             75.06 GB`

nachoparker, 2017-09-21:
@deljones 

I just remembered, there is a feature request already for this https://github.com/nextcloud/nextcloudpi/issues/79

Probably just setting it to run by default at 3:00 am would be enough

nachoparker, 2017-09-21:
@lsousa 

Thanks, I have also been looking at BorgBackup, and even at incremental snapshots with BTRFS. But do you run Borg in the Pi? guide?

deljones, 2017-09-21:
Certainly would....

Dj

lsousa, 2017-09-21:
@nachoparker 

Machine 1 - rPi 3 + NextcloudPi + Borgbackup
Machine 2 - rPi 1 + Raspbian + Borgbackup

No guide yet, but I might do one if you think it'll be useful.

nachoparker, 2017-09-21:
do you need borg on both ends?

Well, the benefit over BTRFS (which is what I am playing with right now) is that it doesn't matter the underlying filesystem if I am not mistaken. The problem might be more complex setup.

lsousa, 2017-09-21:
To take advantage of deduplication yes, borg is needed on both ends. 
You can create a borg repository on the NextcloudPi machine, and then scp/rsync to another location, but AFAIK you'll be copying all the repository instead of only the modifications.  

Using my latest log as example, it would transfer 75.06GB instead of 1.88MB.

nachoparker, 2017-09-21:
I see, thanks

That's one benefit of BTRFS, but I would like to try both.

stratacast, 2017-09-22:
I'll throw in my input on Borg + BTRFS:

Been using BTRFS for 3 years now with no instability issues on all my computers. I did btrfs on my Pi at one point (on external media) but I didn't really take advantage of it when I tried. As far as Borg + BTRFS, I just take a snapshot (in this case of my /home dir) then tell borg to use that snapshot. For the last 3 backups so far I have had no issues (only tried restoring parts of a borg backup, not a full recover...too much data and nowhere to put it for testing. 

I already have a btrfs script written placed in cron.daily where it'll take a snapshot, then check for other snapshots that are 30 days old and delete those snapshots. This is for my desktop though. Also with borg + btrfs, your backup media doesn't have to be using btrfs. 

One last thought: I see some talk of putting Nextcloud into maintenance mode for backups. If we really wanted to be atomic about not backing up hot data, you could do maintenance mode, take the snapshot, take down maintenance mode and it could be done efficiently in under a second (I'd assume)

nachoparker, 2017-09-22:
> Also with borg + btrfs, your backup media doesn't have to be using btrfs.

That's one plus of using BTRFS

> One last thought: I see some talk of putting Nextcloud into maintenance mode for backups. If we really wanted to be atomic about not backing up hot data, you could do maintenance mode, take the snapshot, take down maintenance mode and it could be done efficiently in under a second (I'd assume)

Yeah that's cool, the only problem is that IDK if we could/should enforce BTRFS as a filesystem

Thanks

stratacast, 2017-09-22:
I don't see a problem with that. Btrfs is stable in single-disk configurations, especially in Debian 9. It could provide the option to additionally do a scrub and, since it'll be a single-disk setup with the Pi there could at the very least be detection of data errors on disk. The major problem would be with the folks who already have a Pi setup without btrfs. There would have to be a migration process which would (I think) defeat the purpose of NCP and its ease-of-use. But maybe it could be considered for a future release where you say "hey! NC1x uses btrfs by default on external disks"

nachoparker, 2017-09-23:
Well, I like BTRFS personally... we can just offer the option of BTRFS snapshot backups in case of BTRFS subvolumes, and change `nc-format-USB` from ext4 to BTRFS without enforcing anything.

For ext4 we can keep what we have and we don't enforce anything. But if you want to use ext4 just create the filesystem yourself, which is not a big deal anyways

Ricardosgeral, 2017-09-25:
I'm doing a nc-backup that results in 420Gb tar file. This takes about 12 hours. Ok I can live with that. 

My question now is

If I do automatic backup is it possible to set a specific hour to start the backup?
I would like it to start like 11pm so that it stays in maintenance mode during night.
Maybe this is a simple enhancement to the code, that is, set an hour to start automatic backup. By default it could be 00:00.
This is my suggestion

nachoparker, 2017-09-26:
wow that's a big one! I am glad that it works

> Maybe this is a simple enhancement to the code, that is, set an hour to start automatic backup. By default it could be 00:00.

Ok, this has already been requested in #79 

I'll close this, as I consider big backups tested. Thanks very much. 

I won't lock the thread, people can still comment here

closed 2017-09-26 by nachoparker

stratacast, 2017-09-26:
I THINK I'll be switching an organization's Pi to store nextcloud/data on BTRFS instead of EXT4 (Thank goodness the data amount is small so it can be quick) but I will use that as an opportunity to piece together something that could at least act as a template (or just be fully functional) if NCP decides to switch to BTRFS for nextcloud/data

nachoparker, 2017-09-26:
cool, I want to add BTRFS support one day, so backups can be more efficient