nextcloud/nextcloudpi #190: Apache fails to start after modsecurity is activated in v0.26.16

GHoeher, 2017-09-07:
### PROBLEMS
After upgrading to v0.26.16 and activating modsecurity, apache fails to start.

### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v0.26.16

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
NextCloudPI_07-29-17 upgraded to Strech and now v0.26.16

### Expected behavior
Apache start normally after activation modsecurity

### Actual behaviour
Apache fails to start

### Steps to reproduce, from a freshly installed image

### Include logs
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2017-09-07 09:32:30 CEST; 8s ago
  Process: 2494 ExecStop=/usr/sbin/apachectl stop (code=exited, status=1/FAILURE)
  Process: 2754 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)
  Process: 2751 ExecStartPre=/bin/mkdir -p /var/log/apache2 (code=exited, status=0/SUCCESS)
 Main PID: 2008 (code=exited, status=0/SUCCESS)

Sep 07 09:32:30 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Sep 07 09:32:30 raspberrypi apachectl[2754]: apache2: Syntax error on line 147 of /etc/apache2 apache2.conf: Syntax error on line 9 of /etc/apache2/mods-enabled/security2.conf: Could not open configuration file /etc/modsecurity/modsecurity_crs_10_ignore_static.conf: No such file or directory
Sep 07 09:32:30 raspberrypi apachectl[2754]: Action 'start' failed.
Sep 07 09:32:30 raspberrypi apachectl[2754]: The Apache error log may have more information.
Sep 07 09:32:30 raspberrypi systemd[1]: apache2.service: Control process exited, code=exited status=1
Sep 07 09:32:30 raspberrypi systemd[1]: Failed to start The Apache HTTP Server.
Sep 07 09:32:30 raspberrypi systemd[1]: apache2.service: Unit entered failed state.
Sep 07 09:32:30 raspberrypi systemd[1]: apache2.service: Failed with result 'exit-code'.

nachoparker, 2017-09-07:
Mmmm, interesting, so you still have your old apache configuration for stretch regarding modsecurity.

Can you try this?

```
sudo apt-get -y purge libapache2-mod-security2 modsecurity-crs
sudo echo "" >  /usr/local/etc/nextcloudpi-config.d/modsecurity.sh
sudo ncp-update
```

GHoeher, 2017-09-07:
Hi,

I tried the steps above but already failed at :

> pi@raspberrypi:~$ sudo echo "" >  /usr/local/etc/nextcloudpi-config.d/modsecurity.sh

> -bash: /usr/local/etc/nextcloudpi-config.d/modsecurity.sh: Keine Berechtigung

Access is denied even with sudo. After running "sudo su" I was able to remove the file. Then I run "sudo ncp-update" (which now updated me to 0.26.17 ;-)

Anyway, the issue still stays the same:

● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2017-09-07 18:49:46 CEST; 1min 2s ago
  Process: 15273 ExecStop=/usr/sbin/apachectl stop (code=exited, status=1/FAILURE)
  Process: 15282 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)
  Process: 15280 ExecStartPre=/bin/mkdir -p /var/log/apache2 (code=exited, status=0/SUCCESS)
 Main PID: 14153 (code=exited, status=0/SUCCESS)

Sep 07 18:49:46 raspberrypi systemd[1]: apache2.service: Unit entered failed state.
Sep 07 18:49:46 raspberrypi systemd[1]: apache2.service: Failed with result 'exit-code'.
Sep 07 18:49:46 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Sep 07 18:49:46 raspberrypi apachectl[15282]: apache2: Syntax error on line 147 of /etc/apache2/apache2.conf: Syntax error on line 9 of /etc/apache2/mods-enabled/security2.conf:
 Could not open configuration file /etc/modsecurity/modsecurity_crs_10_ignore_static.conf: No such file or directory
Sep 07 18:49:46 raspberrypi apachectl[15282]: Action 'start' failed.
Sep 07 18:49:46 raspberrypi apachectl[15282]: The Apache error log may have more information.
Sep 07 18:49:46 raspberrypi systemd[1]: apache2.service: Control process exited, code=exited status=1
Sep 07 18:49:46 raspberrypi systemd[1]: Failed to start The Apache HTTP Server.
Sep 07 18:49:46 raspberrypi systemd[1]: apache2.service: Unit entered failed state.
Sep 07 18:49:46 raspberrypi systemd[1]: apache2.service: Failed with result 'exit-code'.

nachoparker, 2017-09-07:
please, try to purge again. After purging... do you still have this files?

```
/etc/apache2/mods-enabled/security2.conf
/etc/apache2/mods-available/security2.conf
```

GHoeher, 2017-09-07:
Sure ...

The two files don'T seem to exist any longer:

root@raspberrypi:/home/pi# ls -l /etc/apache2/mods-available/se*
-rw-r--r-- 1 root root   58 Jul 16 10:15 /etc/apache2/mods-available/sed.load
-rw-r--r-- 1 root root   99 Jul 16 10:15 /etc/apache2/mods-available/session_cookie.load
-rw-r--r-- 1 root root   99 Jul 16 10:15 /etc/apache2/mods-available/session_crypto.load
-rw-r--r-- 1 root root   93 Jul 16 10:15 /etc/apache2/mods-available/session_dbd.load
-rw-r--r-- 1 root root   66 Jul 16 10:15 /etc/apache2/mods-available/session.load
-rw-r--r-- 1 root root 1280 Jul 16 10:15 /etc/apache2/mods-available/setenvif.conf
-rw-r--r-- 1 root root   68 Jul 16 10:15 /etc/apache2/mods-available/setenvif.load
root@raspberrypi:/home/pi# ls /etc/apache2/mods-enabled/se* -l
lrwxrwxrwx 1 root root 31 Aug  8 11:39 /etc/apache2/mods-enabled/setenvif.conf -> ../mods-available/setenvif.conf
lrwxrwxrwx 1 root root 31 Aug  8 11:39 /etc/apache2/mods-enabled/setenvif.load -> ../mods-available/setenvif.load

nachoparker, 2017-09-07:
and how about after you install modsecurity?

```
sudo apt-get install libapache2-mod-security2 modsecurity-crs
```

GHoeher, 2017-09-08:
After installing, I have:

pi@raspberrypi:~ $ ls -l /etc/apache2/mods-enabled/security2*
ls: Zugriff auf '/etc/apache2/mods-enabled/security2*' nicht möglich: Datei oder Verzeichnis nicht gefunden
pi@raspberrypi:~ $ ls -l /etc/apache2/mods-available/security2*
-rw-r--r-- 1 root root 475 Dez 20  2016 /etc/apache2/mods-available/security2.conf
-rw-r--r-- 1 root root  91 Dez 20  2016 /etc/apache2/mods-available/security2.load

after enabling modsecurity it looks like this:

pi@raspberrypi:~ $ ls -l /etc/apache2/mods-available/security2*
-rw-r--r-- 1 root root 475 Dez 20  2016 /etc/apache2/mods-available/security2.conf
-rw-r--r-- 1 root root  91 Dez 20  2016 /etc/apache2/mods-available/security2.load
pi@raspberrypi:~ $ ls -l /etc/apache2/mods-enabled/security2*
lrwxrwxrwx 1 root root 32 Sep  8 11:29 /etc/apache2/mods-enabled/security2.conf -> ../mods-available/security2.conf
lrwxrwxrwx 1 root root 32 Sep  8 11:29 /etc/apache2/mods-enabled/security2.load -> ../mods-available/security2.load

I also did some more checks:
The link to the configuration file which is also mentioned in the error message "/etc/modsecurity/modsecurity_crs_10_ignore_static.conf -> /usr/share/modsecurity-crs/optional_rules/modsecurity_crs_10_ignore_static.conf" exists but the target folder /usr/share/modsecurity-crs/optional_rules does not exist.
In /usr/share/modsecurity-crs I only see the following folders:

pi@raspberrypi:/usr/share/modsecurity-crs $ ls -la
insgesamt 24
drwxr-xr-x   5 root root 4096 Sep  8 11:27 .
drwxr-xr-x 129 root root 4096 Sep  8 11:27 ..
drwxr-xr-x   2 root root 4096 Sep  8 11:27 id_renumbering
-rw-r--r--   1 root root  373 Dez 21  2016 owasp-crs.load
drwxr-xr-x   2 root root 4096 Sep  8 11:27 rules
drwxr-xr-x  12 root root 4096 Sep  8 11:27 util

nachoparker, 2017-09-08:
yes, like I said, somehow your configuration file `/etc/apache2/mods-available/security2.conf` is from jessie, and your modsecurity-crs is from stretch.

How did you upgrade? did you remove the apt sources from jessie in `/etc/apt/sources.list`?

Did you follow the steps in [ownyourbits](https://ownyourbits.com/2017/08/20/nextcloudpi-updated-to-raspbian-9-stretch/)?

GHoeher, 2017-09-08:
I'm pretty sure that I followed the instructions on your homepage.

In my sources.list, Jessie still is available as dev-scr, but commented out, so it should be ok. The deb points to stretch:

># Uncomment line below then 'apt-get update' to enable 'apt-get source'
>#deb-src http://archive.raspbian.org/raspbian/ jessie main contrib non-free rpi
>deb http://mirrordirector.raspbian.org/raspbian/ stretch main contrib non-free rpi

nachoparker, 2017-09-08:
yeah, that's correct

output of 

```
apt-cache policy libapache2-mod-security2 modsecurity-crs
```

GHoeher, 2017-09-08:
Looks like I fixed it. One should never do configuration without having time to check all steps in detail.
I again purged the packages and this time I saw that two folders did not get deleted as they were not empty:
/var/cache/modsecurity
/etc/modsecurity

so I removed them manually this time, started ncp-update again and configured it to start.
Now Apache starts without any issue and everything seems to work.

Thanks for your help anyway, you pointed me to the correct direction :-)

nachoparker, 2017-09-08:
nice! good job ;)

closed 2017-09-08 by nachoparker