nextcloud/nextcloudpi #192: media/USBdrive* showing up as many different directories.

sunjam, 2017-09-08:
### What version of NextCloudPi are you using? latest version 0.26*

### What is the base image that you installed on the SD card?
First NextcloudPi version 12 image release, continually updated to the present.

### Expected behavior
NextcloudPi will run as expected after dist-upgrade. datadir and database have been previously moved to a single ext4 usb thumb drive with no problems via NextcloudPi-config

### Actual behaviour
NextcloudPi fails to locate the system database since my recent NextcloudPi dist-upgrade and I get a white screen when trying to login.  `ls /media` now shows `usb USBdrive USDdrive1 USDdrive2 USDdrive3 USDdrive4 USDdrive5 USDdrive6 USDdrive7 USDdrive8`

All directories are empty except the original USBdrive2, which the pi displays as now only containing ncdatabase folder and I'm unable to `cd` or `ls` into it on the pi `command not found`.

Plugging the usbdrive into my laptop revealed both the ncdatabase and ncdata directories are on the drive as expected, but the Pi is unable to display the directories over ssh.  Is it possible to edit my entries from the nextcloudpi-config menu or command line to the correct directories? Or scan for the correct directories on the usb drive without re-installing them?

nachoparker, 2017-09-11:
that probably happened because of the new automount mechanism conflicting with the old files.

Try this to clean up

```
sudo umount /media/* &>/dev/null
sudo rmdir /media/* &>/dev/null
sudo rm /media/* &>/dev/null
sudo reboot
```

After reboot, everything should be on `/media/USBdrive`. We can verify with

```
ls -l /media
ls -l /media/*
```

Then, it is a matter of making sure that the datadir and database are pointing to the correct path

nachoparker, 2017-09-12:
reopen if that does'nt work for you

closed 2017-09-12 by nachoparker