nextcloud/nextcloudpi #194: App Password Question/Possible bug

albrechtar, 2017-09-10:
I recently installed the NCloudPi and it is great, overall an amazing and easy to use setup thanks to @nachoparker.  

  I have recently come accross an issue if I create a new user and then they want to log into say their iPhone.  From their account (or impersonating from the admin panel) click on personal/Security and then for App Passwords it only generated passwords for the admin.  With no option to change to another username.

  So if the newly created user generated a token, uses it to log into their device they are actually logged into the admin account. (or at least thats what it looks like to me, since it shows my profile image I have for the amin).

I just wanted to post for two things, one of course to get help with resolving this as is is probably something I am doing incorrectly.  And two and potentially more important if this is a bug or something that Nacho and the good folks that build NCloudPi need to fix they are aware of it.

Thank you.

JimmyKater, 2017-09-10:
well from what i understood it's no wonder that app-passwords are created for user admin since you ARE still admin. even if you are logged in in someone other's account.
this is no nextcloudpi-issue - if it's an issue at all it come with nextcloud. i think there even was some similar comment in the forum about this issue.

nextcloudpi provides you with some helpful tools to get your nextcloudsoftware installed to your raspberry pi (so ncp is more or less made for beginners, helping them to selfhost their data using nextcloud)

albrechtar, 2017-09-10:
If you are logged into another account as that user without using the impersonate function, then why would you still be admin? It does not appear to be a NextCloud issue or at least I could not find anything in regards to this in the forum.  I do appreciate your feedback @JimmyKater If I find anything out I will post it here for you and anyone else.  Thank's again for your response I hope you enjoy your day.

JimmyKater, 2017-09-10:
> (or impersonating from the admin panel)

that led me to my answer. it didnt get clear that you logged in with the original user credentials (making sure that said user isn't a member of admin-group) and applied for an app-password which then led you to admin-directory instead of user-directory. if this would always be happening it would be a clear bug. and needed to be reported to github/nextcloud repro. so please make sure that you aren't logged in - by coincidence - as the admin using impersonate.
by all means... it's still no problem with nextcloudpi. ;)

nachoparker, 2017-09-10:
> it's still no problem with nextcloudpi. ;)

True, I'll close this.

You can continue the conversation here, even though I recommend that you use the [Nextcloud forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) for this kind of conversation

Thanks @JimmyKater !

closed 2017-09-10 by nachoparker

albrechtar, 2017-09-11:
@JimmyKater @nachoparker Thanks guys.

albrechtar, 2017-09-11:
@JimmyKater @nachoparker 
UPDATE:  As Jimmy said, there must have been a hangup and it was not going out of Admin.  I updated my instance of NCloudPi, updated and upgraded my Pi, rebooted.  Cleared the cache on my browser (doubt that did anything). Deleted all previously created user accounts from the admin panel, created a new test user, logged out, logged in with the test account, checked and it was working as expected.  

Thanks again @JimmyKater and @nachoparker.  I hope you both enjoy your evening.

nachoparker, 2017-09-11:
great!