nextcloud/nextcloudpi #195: Web-GUI not accessable after fresh install, Apache start fails

SprickW, 2017-09-10:
Version 09_03_17

activated ssh für headless server via file "ssh" in boot 

ssh ist working fine

updated raspi-config with first start

nmap shows open port 22 (ssh), ports 80, 443 and 4443 are closed

Apache log shows:

   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: 
   Active: failed (Result: exit-code) since Tue 2017-09-05 11:30:03 UTC; 4 days 
  Process: 507 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILUR

Sep 05 11:30:02 raspberrypi systemd[1]: Starting The Apache HTTP Server...
**Sep 05 11:30:03 raspberrypi apachectl[507]: AH00526: Syntax error on line 1 of /
Sep 05 11:30:03 raspberrypi apachectl[507]: Invalid command '&\xe0\x11\xe5\xc2\x**
Sep 05 11:30:03 raspberrypi apachectl[507]: Action 'start' failed.
Sep 05 11:30:03 raspberrypi apachectl[507]: The Apache error log may have more i
Sep 05 11:30:03 raspberrypi systemd[1]: apache2.service: Control process exited,
Sep 05 11:30:03 raspberrypi systemd[1]: Failed to start The Apache HTTP Server.
Sep 05 11:30:03 raspberrypi systemd[1]: apache2.service: Unit entered failed sta
Sep 05 11:30:03 raspberrypi systemd[1]: apache2.service: Failed with result 'exi

pi@raspberrypi:~ $ sudo systemctl restart apache2
Job for apache2.service failed because the control process exited with error code.
See "systemctl status apache2.service" and "journalctl -xe" for details.

hardware is Pi3 with 32 or 64 GB SD card

deljones, 2017-09-10:
Is the problem I have in #180 related or connected to this?

Dj

nachoparker, 2017-09-11:
> Is the problem I have in #180 related or connected to this?

I don't think so. That invalid command is clearly garbage. Smells like data corruption to me. 

Try reading [this](https://github.com/nextcloud/nextcloudpi/wiki/FAQ#nextcloud-is-too-slow--i-get-random-failures--i-get-server-errors-even-with-a-fresh-image)

nachoparker, 2017-09-14:
reopen if not fixed

closed 2017-09-14 by nachoparker