nextcloud/nextcloudpi #200: NexcloudPi Web No Longer Accessible

cottoa00, 2017-09-12:


nachoparker, 2017-09-12:
your database is down. did you move it with `nc-database`?

cottoa00, 2017-09-12:
only during initial setup some months ago.
here is other diagnostic you requested;

pi@ecnextcloudpi:~ $ ls -l /media
total 8
drwxr-xr-x 2 root root 4096 Jul 14 17:21 USBdrive
drwxr-xr-x 2 root root 4096 Jul 14 20:18 USBdrive1
lrwxrwxrwx 1 root root   15 Jul 14 17:21 ecCloudDrive -> /media/USBdrive

pi@ecnextcloudpi:~ $ grep datadir /etc/mysql/mariadb.conf.d/50-server.cnf
datadir = /media/USBdrive/ncdatabase

nachoparker, 2017-09-12:
Did you update to stretch? automount changed a little bit, so we might need some cleaning up.

I suggest you

```
sudo umount /media/* &>/dev/null
sudo rmdir /media/* &>/dev/null
sudo rm /media/* &>/dev/null
sudo reboot
```
And then let's see how automonout did

```
ls -l /media
ls -l /media/USBdrive
```

cottoa00, 2017-09-12:
performed cleanup and reboot... automount did not add anything to /media

I have not upgraded to stretch... just DDG'd it... familiar with apt-get update and somewhat with apt-get upgrade... not sure how to move to new OS level and maintain what I have?

nachoparker, 2017-09-12:
see manual instructions [here](https://help.nextcloud.com/t/nextcloudpi-updated-to-raspbian-9-stretch/19871)

In any case, before that you can try activating automount again (even if it says it's active)

cottoa00, 2017-09-12:
Went to nextcloud-config and set automout to yes... was not on? Looked to work and ls -l /media had mappings. Did shutdown restart... now can get nextcloud UI up if use IP but not if use hostname? Also nothing available on IP:4443.. yet. Have not yet upgraded to stretch.... will do that next before dig into hostname access failure.

nachoparker, 2017-09-12:
the hostname stuff is weird, but probably unrelated. Please open a new one or go to the forums for that

closed 2017-09-12 by nachoparker