nextcloud/nextcloudpi #201: Best Way to Recover from full SD card during upgrade to Stretch?

cottoa00, 2017-09-12:
### QUESTIONS

What is the best way to recover from a full SD card during upgrade to Stretch? 
Guess I was not paying enpough attention to space prompts... my DB is on a USB drive so did not expect a full SD card so early in it's life ;-(

nachoparker, 2017-09-12:
I don't understand, can you explain in detail?

cottoa00, 2017-09-12:
Unpacking libbind9-140:armhf (1:9.10.3.dfsg.P4-12.3+deb9u2) ...
Selecting previously unselected package liblwres141:armhf.
Preparing to unpack .../liblwres141_1%3a9.10.3.dfsg.P4-12.3+deb9u2_armhf.deb ...
Unpacking liblwres141:armhf (1:9.10.3.dfsg.P4-12.3+deb9u2) ...
Preparing to unpack .../bind9-host_1%3a9.10.3.dfsg.P4-12.3+deb9u2_armhf.deb ...
Unpacking bind9-host (1:9.10.3.dfsg.P4-12.3+deb9u2) over (1:9.9.5.dfsg-9+deb8u13) ...
dpkg: error processing archive /var/cache/apt/archives/bind9-host_1%3a9.10.3.dfsg.P4-12.3+deb9u2_armhf.deb (--unpack):
 cannot copy extracted data for './usr/share/doc/bind9-host/changelog.gz' to '/usr/share/doc/bind9-host/changelog.gz.dpkg-new': failed to write (No space left on device)
Preparing to unpack .../dnsutils_1%3a9.10.3.dfsg.P4-12.3+deb9u2_armhf.deb ...
dpkg: unrecoverable fatal error, aborting:
 failed to write status database record about 'systemd-sysv' to '/var/lib/dpkg/status': No space left on device
E: Sub-process /usr/bin/dpkg returned an error code (2)

cottoa00, 2017-09-12:
Is there a tmp space I can clear from the aborted upgrade and then do some of my own cleanup?

nachoparker, 2017-09-12:
oh your SD card is full

```
sudo apt-get autoremove -y
sudo apt-get clean
sudo rm /var/lib/apt/lists/* -r
```
Then have a look at space, you might want to expand your partition if you haven't

cottoa00, 2017-09-12:
Having chicken and egg... not enough space to clear itself?!

Tried sudo apt-get autoremove -y but got error (at bottom of listing below) can't do that while dpkg is in interrupted state... must do sudo dpkg --configure -a to correct the problem... but it in turn fails due to space ;-)

Processing triggers for python-support (1.0.15) ...
Setting up python-tdb (1.3.11-2) ...
Setting up gcc-4.9 (4.9.3-14) ...
Setting up libdbus-glib-1-2:armhf (0.108-2) ...
Setting up python-wicd (1.7.4+tb2-4) ...
Setting up python-dev (2.7.13-2) ...
Setting up libgnutls30:armhf (3.5.8-5+deb9u2) ...
Setting up wpasupplicant (2:2.4-1) ...
Installing new version of config file /etc/wpa_supplicant/functions.sh ...
Setting up librtmp1:armhf (2.4+20151223.gitfa8646d.1-1) ...
Setting up python-pkg-resources (33.1.1-1) ...
Setting up gcc-6 (6.3.0-18+rpi1) ...
Setting up g++-6 (6.3.0-18+rpi1) ...
Setting up python-gobject-2 (2.28.6-13) ...
Setting up python-gi (3.22.0-2) ...
Setting up libldap-2.4-2:armhf (2.4.44+dfsg-5) ...
Setting up python3.4 (3.4.4-3) ...
Setting up wget (1.18-5) ...
Installing new version of config file /etc/wgetrc ...
Setting up python-virtualenv (15.1.0+ds-1) ...
Setting up g++-4.9 (4.9.3-14) ...
Setting up libavahi-client3:armhf (0.6.32-2) ...
Setting up libkrb5-3:armhf (1.15-1) ...
Setting up python-dbus (1.2.4-1) ...
Setting up dirmngr (2.1.18-6) ...
Setting up libnfsidmap2:armhf (0.25-5.1) ...
Setting up python-gobject (3.22.0-2) ...
Setting up libldb1:armhf (2:1.1.27-1) ...
Setting up gcc (4:6.3.0-4) ...
Setting up libgssapi-krb5-2:armhf (1.15-1) ...
Setting up libtirpc1:armhf (0.2.5-1.2) ...
Setting up g++ (4:6.3.0-4) ...
Setting up python-apt (1.1.0~beta5) ...
dpkg: unrecoverable fatal error, aborting:
 failed to write status database record about 'libsmbclient:armhf' to '/var/lib/dpkg/status': No space left on device
pi@ecnextcloudpi:~ $ sudo apt-get autoremove -y
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem. 
pi@ecnextcloudpi:~ $

nachoparker, 2017-09-13:
you can clean those debs with the commands I put before

```
sudo apt-get autoremove -y
sudo apt-get clean
sudo rm /var/lib/apt/lists/* -r
```

at this point I consider it a non NCP issue. I'll close but we can continue talking.

I recommend that you use the forums for general questions

https://help.nextcloud.com/c/support/appliances-docker-snappy-vm

closed 2017-09-13 by nachoparker

cottoa00, 2017-09-13:
On Tue, 12 Sep 2017 23:51:22 -0700
nachoparker <notifications@github.com> wrote:

> you can clean those debs with the commands I put before
>
> ```
> sudo apt-get autoremove -y
> sudo apt-get clean
> sudo rm /var/lib/apt/lists/* -r
> ```
>
> at this point I consider it a non NCP issue. I'll [...]

The above will not work... the first command fails due dpkg failure.
The "fix" -> sudo dpkg --configure -a ...chokes on no space.

My question to you is given what is installed on a relatively fresh
NCPi box where might I go to manually clear some things that are not
critical to make some room. I would think you are the best resource to
make that call. I put nothing on the box other than NCPi and it's
related components.

The only other option is to some make a space on the USB drive and copy
the /var/cache/apt/archives files there with some sort of symlink until
I get away from the chicken and egg problem. Once I make some space I'm
sure I can recover from the dpkg failure and then run your autoremove.

===============
output when run suggested commands:

pi@ecnextcloudpi:~ $ sudo apt-get autoremove -y
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure
-a' to correct the problem. 

pi@ecnextcloudpi:~ $ sudo dpkg --configure -a 
Setting up libdns162:armhf (1:9.10.3.dfsg.P4-12.3+deb9u2) ... 
Setting up
wicd-daemon (1.7.4+tb2-4) ... Installing new version of config
file /etc/default/wicd ... Installing new version of config
file /etc/init.d/wicd ... Installing new version of config
file /etc/wicd/dhclient.conf.template.default ... Installing new
version of config file /etc/wicd/encryption/templates/active ...
Installing new version of config
file /etc/wicd/encryption/templates/peap-tkip ... Installing new
version of config file /etc/wicd/encryption/templates/psu ...
Installing new version of config
file /etc/wicd/encryption/templates/wep-hex ... Installing new version
of config file /etc/wicd/encryption/templates/wep-passphrase ...
Installing new version of config
file /etc/wicd/encryption/templates/wep-shared ... Installing new
version of config file /etc/wicd/encryption/templates/wired_8021x ...
Installing new version of config
file /etc/wicd/encryption/templates/wpa-peap ... Installing new version
of config file /etc/wicd/encryption/templates/wpa2-peap ... insserv:
warning: current start runlevel(s) (empty) of script `wicd' overrides
LSB defaults (2 3 4 5). insserv: warning: current stop runlevel(s) (0 1
2 3 4 5 6) of script `wicd' overrides LSB defaults (0 1 6). debconf:
DbDriver "templatedb": could not
write /var/cache/debconf/templates.dat-new: No space left on device
Warning: unable to close filehandle $fh properly: No space left on
device at /usr/share/perl5/Debconf/DbDriver.pm line 55, <GEN6> line 2.
dpkg: error processing package wicd-daemon (--configure): subprocess
installed post-installation script returned error exit status 1 Setting
up build-essential (12.3) ... dpkg: unrecoverable fatal error,
aborting: unable to flush /var/lib/dpkg/updates/tmp.i after padding:
 
No space left on device 

pi@ecnextcloudpi:~ $

nachoparker, 2017-09-13:
at least you should be able to do this

```
sudo rm /var/lib/apt/lists/* 
```
Anyways... If you don't have anything valuable in nextcloud, you can probably just delete it

```
rm -r /var/www/nextcloud
```

you can reinstall it from `nextcloudpi-config`.

you can also try removing logs

```
rm /var/log/*
rm /var/log/apache2/*
rm /var/www/nextcloud/data/nextcloud.log
```

testbird, 2017-09-19:
Using `sudo apt-get clean` first should remove the cached .deb files with indexes (/var/cache/apt et al.), and make room for the more complex tasks above.