nextcloud/nextcloudpi #203: nc-static-ip - what happes if there is already a steady ip set before? (question)

JimmyKater, 2017-09-15:
hey nacho & the gang :)

just updated v28.1 (running on stretch now)... and discovered the new option "nc-static-ip" - you can choose between on and off (meaning dhcp come in).
so as i have set an internal static-ip since long ago (almost from the beginning on and it was kind of a hazzle for me to get dhcp out from everywhere) i am now wondering: nc-static-ip isn't turned on in my ncp-web. does that mean it's gonna switch to dhcp again? or do i need to put my local static ip in and turn in on? how to proceed? 
isn't it possible that ncp would detect whether there is already a static one set and fill in these info to the field in nc-static-ip? or do we need an "enable"-button with nc-static-ip?

cheerio
jimmy

nachoparker, 2017-09-15:
Hi!

Would you mind moving this question to the forums please? we'll talk over there ;)

JimmyKater, 2017-09-15:
argh. of course.... sorry. sometimes i just get carried away and don't think of things i should know myself. 
crossposted there

nachoparker, 2017-09-15:
haha thanks!

closed 2017-09-15 by nachoparker