nextcloud/nextcloudpi #204: unprotected web panel (no password?)

testbird, 2017-09-17:
Installed NextCloudPi on 09-16-17

Accessed https://192.168.0.8:4443

And the web panel that allows to change very important server settings was accessible for anyone through the local network (lan,wlan,vpn,...), without any password protection!

The secure https protocol connection is nice, but without at least a "basic auth" password configured for the web panel, I think the configuration is way too much open to getting messed up.

Is the user "pi" password not used by intention?
https://serverfault.com/questions/45278/authenticate-in-apache-via-system-account

nachoparker, 2017-09-17:
Hi, 

Thanks for your feedback. I aggree.

ncp-web is not complete yet, and at this point the most important missing piece is that. There is already a feature request #132 that I opened for this ( with the tag 'help wanted' also ).

I have been actively [looking for help](https://ownyourbits.com/2017/09/10/nextcloudpi-at-the-nextcloud-conference/) for this. Somebody offered to help with password authentication some days ago, I don't know at this point what the status of that is.

At this moment for anyone concerned with this, you can disable ncp-web with the option `nc-webui`.

Of course, help is welcome and much needed. This is slow because it is 99% a one person effort.

I'll close as duplicate, please we can continue the conversation over there

nachoparker, 2017-09-17:
Duplicate of #132

closed 2017-09-17 by nachoparker