nextcloud/nextcloudpi #205: letsencrypt failing. Suddenly

deljones, 2017-09-17:
I was recently wondering why I had not received any notification that my domain would be auto renewed for my NCPi. I had been getting regular notices telling me of the auto renewal for a while.

I decide to to run the letsencrypt script in nextcloudpi-config. The script ran and failed with the following:

Launching letsencrypt
sed: can't read /etc/apache2/sites-available/ncp.conf: No such file or directory
sed: can't read /etc/apache2/sites-available/ncp.conf: No such file or directory
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Cert is due for renewal, auto-renewing...
Renewing an existing certificate
Performing the following challenges:
http-01 challenge for digitalcarnage.webhop.me
Using the webroot path /var/www/nextcloud for all unmatched domains.
Waiting for verification...
Cleaning up challenges
Failed authorization procedure. digitalcarnage.webhop.me (http-01): urn:acme:error:unauthorized :: The client lacks sufficient authorization :: Invalid response from http://digitalcarnage.webhop.me/.well-known/acme-challenge/pPVUH8m3UwhibX65ey6vpZIQk38WUSpd16jSGj7Ir2U: "<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>404 Not Found</title>
</head><body>
<h1>Not Found</h1>
<p"

IMPORTANT NOTES:
 - The following errors were reported by the server:

   Domain: digitalcarnage.webhop.me
   Type:   unauthorized
   Detail: Invalid response from
   http://digitalcarnage.webhop.me/.well-known/acme-challenge/pPVUH8m3UwhibX65ey6vpZIQk38WUSpd16jSGj7Ir2U:
   "<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
   <html><head>
   <title>404 Not Found</title>
   </head><body>
   <h1>Not Found</h1>
   <p"

   To fix these errors, please make sure that your domain name was
   entered correctly and the DNS A record(s) for that domain
   contain(s) the right IP address.
Done. Press any key...

My domain does work ok, if you go to the address you get the front page of NC and it works fine.
I've also noticed that Forced https no longer works.

if I just type in the domain I get the 'IT WORKS!' from Apache. I have to type in the https even though Forced https is selected in nextcloudpi-config...

Surely related now then is the inability if using ncp-webUI?

Is there a quick fix..?

nachoparker, 2017-09-18:
It seems that you come from an old image. Do you have ncp-web installed? Coming from an old image it has to be done manually. If I recall correctly, you did, didn't you?

It doesn't find the file `/etc/apache2/sites-available/ncp.conf`

nachoparker, 2017-09-18:
Regarding the redirection, what is the output of

```
cat /etc/apache2/sites-available/000-default.conf
```

deljones, 2017-09-18:
Indeed I have no ncp.conf

and the output of 000-default.conf is:

<VirtualHost _default_:80>
  DocumentRoot /var/www/nextcloud
  <IfModule mod_rewrite.c>
    RewriteEngine On
    RewriteCond %{HTTPS} !=on
    RewriteRule ^/?(.*) https://%{SERVER_NAME}/$1 [R,L]
  </IfModule>
</VirtualHost>

I'm quite happy to rebuild if thats the best way forward... I have full backup of all my staggering 5 users!!

Is the latest image: NextCloudPi_09-12-17.tar.bz2

Del

nachoparker, 2017-09-18:
Well, redirection is enabled so it should work :S 

You can try enabling it maybe

```
sudo a2ensite 000-default
```

Of course, starting with a fresh image would be nice and safe, but we can try to see what's happening. Whatever you decide!

deljones, 2017-09-18:
It would be good maybe to see whats going on first…

sudo a2ensite 000-default

gave:

Site 00o-default already enabled

Del

> 
> Well, redirection is enabled so it should work :S
> 
> You can try enabling it maybe
> 
> sudo a2ensite 000-default
> Of course, starting with a fresh image would be nice and safe, but we can try to see what's happening. Whatever you decide!

Derek Jones Stage and Television Lighting Design <http://www.derekjones.net/>

w: www.derekjones.net <http://www.derekjones.net/>
t: +44 (0)7957 140 512
e: derek@derekjones.net
derekjonesphotography.co.uk <http://derekjonesphotography.co.uk/>

Get my PGP public key. <https://keyserver.pgp.com/vkd/GetWelcomeScreen.event>

The information in this E-Mail is confidential and may be legally privileged. 
It may not represent the views of Derek Jones. 
It is intended solely for the addressees. 
Access to this E-Mail by anyone else is unauthorised. 
If you are not the intended recipient, any disclosure, copying, distribution or any action taken or omitted to be taken in reliance on it, is prohibited and may be unlawful. 
Any unauthorised recipient should advise the sender immediately of the error in transmission.

nachoparker, 2017-09-19:
I don't know why the redirection is being ignored

```
sudo apache2ctl -S
```

Maybe the apache module is disabled?

```
sudo a2enmod rewrite
```

deljones, 2017-09-19:
Morning Np...

yes that did the trick... Thank you.

I did flash a new SD card last night which I might move over to, as you say its nice and neat and tidy.

If I use the new SD card and change the NC config file to point to my existing drive for ncdata will nextcloudpi-config pick up the change? My understanding is that changing the database and ncdata location in nextcloudpi-config is a MOVE of data so use with care as I may over write?

Thanks again for your work. Watched your video at NC conf.. Very good :)

Dj

nachoparker, 2017-09-19:
> yes that did the trick... Thank you.

lol how did it get disabled? xD

> If I use the new SD card and change the NC config file to point to my existing drive for ncdata will nextcloudpi-config pick up the change? My understanding is that changing the database and ncdata location in nextcloudpi-config is a MOVE of data so use with care as I may over write?

It is a move of data, but NC will not pick the database info just by pointing `config.php` to the old datadir.

I suggest you use `nc-restore`, which will populate the database again (wherever it is, you can move it around later or you may have moved it already). If you include data in the `nc-backup`, then your data will be copied to the same place where it was. If you didn't, then your database info will be there but there will be no files.

In that case, you can point `config.php` to the old place and run `nc-scan` to refresh the file list in the database.

> Thanks again for your work. Watched your video at NC conf.. Very good :)

Thanks! it was cool to be there ;D

closed 2017-09-19 by nachoparker