nextcloud/nextcloudpi #208: changelog.md not updated

Shadowstreik, 2017-09-19:
Hello. With the past few update releases, changelog.md has not been kept up to date. Please keep it up to date in line with the releases.

Thank you.

nachoparker, 2017-09-19:
Yesterday I realized that I was pointing at the wrong branch for the changelog, and I fixed it in `v0.28.2`

https://github.com/nextcloud/nextcloudpi/commit/ac3fcab48b0fea1bdab714c760da65723614503b

In any case, in the wiki there is a link to the changelog

https://github.com/nextcloud/nextcloudpi/wiki/Change-Log

closed 2017-09-19 by nachoparker

Shadowstreik, 2017-09-19:
Clicking on the version in NCPWeb's upper-left corner points to https://github.com/nextcloud/nextcloudpi/blob/master/changelog.md where the version name and changes are expected to be listed, but aren't. It currently lists up to v0.28.1 and stops, there. The wiki entry doesn't notate the changes in order version in the fashion that changelog.md does (preferred).

Could that be resolved?

nachoparker, 2017-09-19:
we are in `v0.28.2`. You are right, that last one is not there. I guess I could write a push script so I don't forget again 

automating wiki entries... IDK if it could be done but I think that the link to changelog.md is good enough, right?

Shadowstreik, 2017-09-21:
Hi. Yeah, i think the link to changlelog.md is good. Though, I would recommend putting "Changelog" in parentheses next to the version number hyperlink. That would provide some identification it's the link for the changelog.

nachoparker, 2017-09-21:
Yeah I've thought that too, I decided against that because I think it would be too big. 

I'll try it next time I work on ncp-web