nextcloud/nextcloudpi #209: doc: steps to install and boot NCP from an external USB drive.

nachoparker, 2017-09-20:
Some people reported to have done it using [this guide](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md). It would be nice to have the exact steps written down by somebody that has done it in the wiki for reference.

Larsene, 2017-10-06:
That's right. I've followed this way to install NextCloudPi on an USB SSD drive. This way is a lot faster Nexcloud than on my SD card !

My english is not good enough, i think, to create a step by step guide anyway...

nachoparker, 2017-10-08:
of course! that would be awesome thanks!

Larsene, 2017-10-13:
okay, i'll do it this weekend.

ovpc, 2017-10-25:
I can check and correct English.

nachoparker, 2017-10-25:
@ovpc  thanks!!

> BTW I have installed Rasbian on external drives many times using the berrybootloader

mmm this is a different topic, but maybe it would be interesting to talk to berryboot and see what does it take to adapt NCP

ovpc, 2017-10-25:
@nachoparker did a quick look over topics on [github](https://github.com/maxnet/berryboot/issues/425) and found pointer to just that question

> see what does it take to adapt NCP

http://www.berryterminal.com/doku.php/berryboot/adding_custom_distributions

nachoparker, 2017-10-26:
wow that's great! 

it would be nice if you wrote down the steps, maybe it can be added to the wiki, or maybe it we can talk with berryboot to see if they would be willing to include it.

let us know how it goes ;)

ovpc, 2017-10-26:
@nachoparker 
I 'll gladly write down the steps, the berryboot parts are already well documented, so would be mostly copy paste or links to steps to follow.
No luck sofar tho, in that the image gets imported and boots initially, but **[FAILED]: Failed to start Remounting Root  and kernel file system.**
I have screenshots taken with my phone, so thinking: write down steps below I took and post as issue/question at maxnet/berryboot? 
Nothing happens when hit enter or try to type anything in console. 
[image]
[image]

Part 1 (Getting RPI to boot with berryboot)

Create single partition and format to fat32 a microsd 
Unzip [downloaded berryboot.zip ](http://www.berryterminal.com/doku.php/berryboot)into the microsd

Part 2 (paste from [berryboot doc](http://www.berryterminal.com/doku.php/berryboot/adding_custom_distributions))
Adding your own custom operating systems to the menu (input/output of[ my console at pastebin](https://pastebin.com/1gkFmp6K))

You can add your own extra operating systems to the Berryboot menu. However this requires that you convert your file system image to SquashFS format first.

Most Raspberry Pi operating system images are disk images containing two partitions. A FAT partition with the boot loader and kernel files, and a second ext4 partition with everything else. We are interested in the second partition.

With a regular Linux desktop computer that has kpartx and mksquashfs installed, you can convert the second partition to SquashFS like this: (My Linux Mint 18 had kpartx already installed, used [squashfs-tools provided by Mint Devs](https://community.linuxmint.com/software/view/squashfs-tools))

$ sudo kpartx -av image_you_want_to_convert.img 
add map loop0p1 (252:5): 0 117187 linear /dev/loop0 1
add map loop0p2 (252:6): 0 3493888 linear /dev/loop0 118784
$ sudo mount /dev/mapper/loop0p2 /mnt
$ sudo sed -i 's/^\/dev\/mmcblk/#\0/g' /mnt/etc/fstab
$ sudo mksquashfs /mnt converted_image_for_berryboot.img -comp lzo -e lib/modules
$ sudo umount /mnt
$ sudo kpartx -d image_you_want_to_convert.img 
Notes:

If kpartx reports it created a mapping different than loop0p2 (e.g. loop4p2) mount that instead. This can happen if loop0 is already in use by something else on the system.
We are excluding /lib/modules from the image, because the kernel modules shipped with Berryboot are used instead, and shared with all distributions.
Some older versions of mksquashfs do not support the ”-comp lzo” option *(I tried with and without this option, both failed)*. You can leave it out to let it use gzip compression instead. Advantage of LZO is that it is faster to uncompress, which is a big plus on slow ARM devices, and therefore preferred. This does come at a cost of reduced compression ratio (LZO images are larger than gzip ones).
Put your SquashFS formatted image on a USB stick, go to the “Operating system installer”, hold down your mouse button over “Add OS” and select “Install from USB stick” If your image prefers to have a certain memory split use the extension .img128 .img192, .img224 or .img240 instead of .img.

Tweaks

If the image you are converting is based on Debian/Raspbian delete the etc/console-setup/cached_UTF-8_del.kmap.gz file before converting the image to force regeneration of the cached keyboard mapping on first boot. This is to make sure it uses the keyboard layout set in Berryboot instead of default British.

nachoparker, 2017-10-27:
Thanks for your work!

> I have screenshots taken with my phone, so thinking: write down steps below I took and post as issue/question at maxnet/berryboot?

Yeah, I would  say that's the way to go, no doubt they will be able to help out.

nachoparker, 2017-10-27:
@Larsene , we are kind of going offtopic here, but still would be great to have the instructions for booting from USB drive.

ovpc, 2017-10-27:
@nachoparker  By all means, move it to a new topic called.

**Install and boot NCP from an external USB drive with Berryboot**

My offer  @Larsene to check/correct English on step-by-step the Rasbian way still stands.

The main difference looks to be:
1.The official rasbian way; Booting from sd card is permanently disabled in this method. Not sure if and how it can be undone, a how to would need to include this info.  
2.With Berryboot you're still booting from (micro) sd card, but OS and files live on the external device chosen during initial boot.

nachoparker, 2017-12-17:
I created a note in the wiki asking for help to write down the steps in more detail. It will be more visible there.

closed 2017-12-17 by nachoparker