nextcloud/nextcloudpi #211: Can't connect to my nextcloud

JVAQUEROM, 2017-09-21:
Hi there,

I had nextcloudpi before it was udpdated to Stretch. I did the update and I believe that something went wrong, because I started having a lot of problems and errors....I think the filesystem got some error or something, so I started from the last nextcloudpi image available, with Stretch. 

I've been trying to set everything as it was before, but I have realized that I cannot connect to port 80. Letsencrypt fails with something about permissions and ".well-known/..." (I can post the specific error). Thing is I cannot even connect with the local IP, so something is quite wrong. I'd love not to have to reinstall because I also install kodi and a few things more and it is a pain to start from scratch again...

I'd like to give more information on the error but I don't really know where to get it. The page just says "I can't connect"

Thank you

nachoparker, 2017-09-22:
Please, use the github template to provide the necessary information to help you. Provide this info

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -n 100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -n 100 /var/log/mysql/*.log`
```
</details>

JVAQUEROM, 2017-09-22:
Thank you for the tip!! I am sorry!!

### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v0.28.2
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
NextcloudPi_09_03_17 (the command says "untagged" but I keep the downloaded image in my pc).
### Expected behavior
Connect to Nextcloud via 192.168.1.xxx or no-ip address
### Actual behaviour
It says "I cannot connect" ("No se puede conectar"). However, other ports, like 9091 (transmission) work fine.
 
### Steps to reproduce, from a freshly installed image
I am not sure because I mostly did all the configuration before checking what made all wrong.
### Include logs
<details>
<summary>Nextcloud logs</summary>
Nextcloud log file is empty
</details>
<details>
<summary>Apache logs</summary>
`systemctl status apache2`
```
● apache2.service - The Apache HTTP Server                                                                                                                                                      
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)                                                                                                        
   Active: failed (Result: exit-code) since Fri 2017-09-22 07:00:39 CEST; 12h ago                                                                                                               
                                                                                                                                                                                                
sep 22 07:00:38 raspberrypi systemd[1]: Starting The Apache HTTP Server...                                                                                                                      
sep 22 07:00:39 raspberrypi apachectl[4194]: AH00526: Syntax error on line 5 of /etc/apache2/sites-enabled/ncp.conf:                                                                            
sep 22 07:00:39 raspberrypi apachectl[4194]: SSLCertificateFile: file '/etc/letsencrypt/live/javicloud.ddns.net/fullchain.pem' does not exist or is empty                                       
sep 22 07:00:39 raspberrypi apachectl[4194]: Action 'start' failed.                                                                                                                             
sep 22 07:00:39 raspberrypi apachectl[4194]: The Apache error log may have more information.                                                                                                    
sep 22 07:00:39 raspberrypi systemd[1]: apache2.service: Control process exited, code=exited status=1                                                                                           
sep 22 07:00:39 raspberrypi systemd[1]: Failed to start The Apache HTTP Server.                                                                                                                 
sep 22 07:00:39 raspberrypi systemd[1]: apache2.service: Unit entered failed state.                                                                                                             
sep 22 07:00:39 raspberrypi systemd[1]: apache2.service: Failed with result 'exit-code'.      

````
`tail -n 100 /var/log/apache2/*.log`
These are empty too.

</details>
<details>
<summary>mariaDB logs</summary>
`systemctl status mysqld`
```

● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-09-21 19:44:03 CEST; 23h ago
 Main PID: 1003 (mysqld)
   Status: "Taking your SQL requests now..."
   CGroup: /system.slice/mariadb.service
           └─1003 /usr/sbin/mysqld

sep 21 19:43:23 raspberrypi systemd[1]: Starting MariaDB database server...
sep 21 19:43:51 raspberrypi mysqld[1003]: 2017-09-21 19:43:51 1995571200 [Note] /usr/sbin/mysqld (mysqld 10.1.23-MariaDB-9+deb9u1) starting as process 1003 ...
sep 21 19:44:03 raspberrypi systemd[1]: Started MariaDB database server.

```
`tail -n 100 /var/log/mysql/*.log`
this is empty too. It only exists error.log but it is empty

</details>

Thank you for the AWESOME job you're doing with nextcloudpi!

testbird, 2017-09-23:
> because I also install kodi and a few things

Just a thought you may investigate: Updating those packages may have interfered with the apache config, and maybe try https://192.168.1.xxx:4443 for the web interface.

JVAQUEROM, 2017-09-23:
I modified the files that gave the error (`etc/apache2/sites-enabled/ncp.conf` and `etc/apache2/sites-enabled/nextcloud.conf`) and now I can access https://192.168.1.xxx:4443 the web interface but cannot see nextcloud (port 80). 

```

pi@raspberrypi:~ $ systemctl status apache2.service 
● apache2.service - The Apache HTTP Server                                                                                                                                                      
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)                                                                                                        
   Active: active (running) since Fri 2017-09-22 23:39:03 CEST; 17h ago                                                                                                                         
  Process: 11538 ExecReload=/usr/sbin/apachectl graceful (code=exited, status=0/SUCCESS)                                                                                                        
  Process: 9025 ExecStart=/usr/sbin/apachectl start (code=exited, status=0/SUCCESS)                                                                                                             
 Main PID: 9029 (apache2)
   CGroup: /system.slice/apache2.service
           ├─ 9029 /usr/sbin/apache2 -k start
           ├─11542 /usr/sbin/apache2 -k start
           └─11543 /usr/sbin/apache2 -k start

sep 22 23:39:03 raspberrypi systemd[1]: Starting The Apache HTTP Server...
sep 22 23:39:03 raspberrypi apachectl[9025]: AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive globally t
sep 22 23:39:03 raspberrypi systemd[1]: Started The Apache HTTP Server.
sep 23 06:25:02 raspberrypi systemd[1]: Reloading The Apache HTTP Server.
sep 23 06:25:02 raspberrypi apachectl[11538]: AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive globally 
sep 23 06:25:02 raspberrypi systemd[1]: Reloaded The Apache HTTP Server.
```

Anyway, I don't think that those package have interfered because I used them in the previous image (with Jessie) and there was no problem. Basically it uses port 9091 only. I think it might have to do with using the same IP for a "different" (in the sense that I started from scratch) machine and maybe some security issue with the router or letsencrypt, I do not know....

nachoparker, 2017-09-23:
Maybe it's not enabled

```
sudo apache2ctl -S
```

try

```
sudo a2ensite nextcloud
```

JVAQUEROM, 2017-09-23:
```
pi@raspberrypi:~ $ sudo apache2ctl -S
AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive globally to suppress this message
VirtualHost configuration:
*:80                   127.0.1.1 (/etc/apache2/sites-enabled/000-default.conf:1)
*:4443                 127.0.1.1 (/etc/apache2/sites-enabled/ncp.conf:2)
*:443                  javicloud.ddns.net (/etc/apache2/sites-enabled/nextcloud.conf:2)
ServerRoot: "/etc/apache2"
Main DocumentRoot: "/var/www/html"
Main ErrorLog: "/var/log/apache2/error.log"
Mutex default: dir="/var/run/apache2/" mechanism=default 
Mutex watchdog-callback: using_defaults
Mutex rewrite-map: using_defaults
Mutex ssl-stapling-refresh: using_defaults
Mutex ssl-stapling: using_defaults
Mutex proxy: using_defaults
Mutex ssl-cache: using_defaults
PidFile: "/var/run/apache2/apache2.pid"
Define: DUMP_VHOSTS
Define: DUMP_RUN_CFG
User: name="www-data" id=33
Group: name="www-data" id=33
pi@raspberrypi:~ $ sudo a2ensite nextcloud
Site nextcloud already enabled
pi@raspberrypi:~ $ 

```

Right now I'm getting a blank page instead of "cannot connect". If I try with https, I get:
`
An error occurred during a connection to 192.168.1.222. SSL received a record that exceeded the maximum permissible length. Error code: SSL_ERROR_RX_RECORD_TOO_LONG
`

nachoparker, 2017-09-24:
honestly it is a bit hard to tell you what to do without knowing what happened to your system when you installed other things.

What is the current situation of `etc/apache2/sites-enabled/nextcloud.conf`? what did you do with the `/etc/letsencrypt/live/javicloud.ddns.net/fullchain.pem` line?

JVAQUEROM, 2017-09-24:
The file `/etc/apache2/sites-enabled/nextcloud.conf`is the following:

```
<IfModule mod_ssl.c>
  <VirtualHost _default_:443>
    DocumentRoot /var/www/nextcloud
ServerName javicloud.ddns.net
    CustomLog /var/www/nextcloud/data/access.log combined
    ErrorLog /var/www/nextcloud/data/error.log
  </VirtualHost>
  <Directory /var/www/nextcloud/>
    Options +FollowSymlinks
    AllowOverride All
    <IfModule mod_dav.c>
      Dav off
    </IfModule>
    LimitRequestBody 0
    SSLRenegBufferSize 10486000
  </Directory>
</IfModule>

```

I took out the lines relative to letsencrypt from there and from `/etc/apache2/sites-enabled/nextcloud.conf`

```

Listen 4443
<VirtualHost _default_:4443>
  DocumentRoot /var/www/ncp-web
</VirtualHost>
<Directory /var/www/ncp-web/>
  Require host localhost
  Require local
  Require ip 192.168
  Require ip 10
</Directory>

```

nachoparker, 2017-09-25:
nc-init won't do because your problem is in apache.

Use this configuration for `nextcloud.conf`

```
<IfModule mod_ssl.c>   
  <VirtualHost _default_:443>                  
    ServerName javicloud.ddns.net
    DocumentRoot /var/www/nextcloud            
    CustomLog /var/www/nextcloud/data/access.log combined                                     
    ErrorLog /var/www/nextcloud/data/error.log 
    SSLEngine on       
    SSLCertificateFile      /etc/ssl/certs/ssl-cert-snakeoil.pem                              
    SSLCertificateKeyFile /etc/ssl/private/ssl-cert-snakeoil.key                              
  </VirtualHost>       
  <Directory /var/www/nextcloud/>              
    Options +FollowSymlinks                    
    AllowOverride All  
    <IfModule mod_dav.c>                       
      Dav off          
    </IfModule>        
    LimitRequestBody 0 
    SSLRenegBufferSize 10486000                
  </Directory>         
</IfModule>
```

user-jan, 2017-09-25:
Hi Nacho,
I seem to have the same problem as JVAQUEROM.
Best,
Jan

**What version of NextCloudPi are you using? ( eg: v0.17.2 )**

v0.28.2

**What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )**

NextcloudPi_09_03_17 (the command says "untagged" but I keep the downloaded image in my pc).

**Expected behavior**

Connect to Nextcloud via 192.168.1.xxx or no-ip address

**Actual behaviour**

It says "I cannot connect" ("No se puede conectar"). However, other ports, like 9091 (transmission) work fine.

**Steps to reproduce, from a freshly installed image**

I am not sure because I mostly did all the configuration before checking what made all wrong.

nachoparker, 2017-09-26:
@user-jan

So you can't open Nextcloud? :(

Let's see, please give me the following info

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -n 100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -n 100 /var/log/mysql/*.log`
```
</details>

user-jan, 2017-09-26:
Hi Nacho,

Thanks for getting back to me.
Here's a geek question at beginner level:

"How do I call up the logs?"

Best,
Jan :)

nachoparker, 2017-09-26:
type this, and paste it here

```
systemctl status apache2
tail -n 100 /var/log/apache2/*.log
```

user-jan, 2017-09-26:
Hi Nacho,

Here's the output as requested by you :)

Cheers,
Jan

pi@raspberrypi:~ $ systemctl status apache2
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: 
   Active: failed (Result: exit-code) since Mon 2017-09-25 15:14:01 UTC; 23h ago
  Process: 470 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILUR

Sep 25 15:13:59 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Sep 25 15:14:01 raspberrypi apachectl[470]: AH00558: apache2: Could not reliably
Sep 25 15:14:01 raspberrypi apachectl[470]: (2)No such file or directory: AH0229
Sep 25 15:14:01 raspberrypi apachectl[470]: AH00014: Configuration check failed
Sep 25 15:14:01 raspberrypi apachectl[470]: Action 'start' failed.
Sep 25 15:14:01 raspberrypi apachectl[470]: The Apache error log may have more i
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Control process exited,
Sep 25 15:14:01 raspberrypi systemd[1]: Failed to start The Apache HTTP Server.
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Unit entered failed sta
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Failed with result 'exi
lines 1-15/15 (END)...skipping...
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: 
   Active: failed (Result: exit-code) since Mon 2017-09-25 15:14:01 UTC; 23h ago
  Process: 470 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILUR

Sep 25 15:13:59 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Sep 25 15:14:01 raspberrypi apachectl[470]: AH00558: apache2: Could not reliably
Sep 25 15:14:01 raspberrypi apachectl[470]: (2)No such file or directory: AH0229
Sep 25 15:14:01 raspberrypi apachectl[470]: AH00014: Configuration check failed
Sep 25 15:14:01 raspberrypi apachectl[470]: Action 'start' failed.
Sep 25 15:14:01 raspberrypi apachectl[470]: The Apache error log may have more i
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Control process exited,
Sep 25 15:14:01 raspberrypi systemd[1]: Failed to start The Apache HTTP Server.
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Unit entered failed sta
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Failed with result 'exi
~
~
~
~
~
~
~
~

● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: 
   Active: failed (Result: exit-code) since Mon 2017-09-25 15:14:01 UTC; 23h ago
  Process: 470 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILUR

Sep 25 15:13:59 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Sep 25 15:14:01 raspberrypi apachectl[470]: AH00558: apache2: Could not reliably
Sep 25 15:14:01 raspberrypi apachectl[470]: (2)No such file or directory: AH0229
Sep 25 15:14:01 raspberrypi apachectl[470]: AH00014: Configuration check failed
Sep 25 15:14:01 raspberrypi apachectl[470]: Action 'start' failed.
Sep 25 15:14:01 raspberrypi apachectl[470]: The Apache error log may have more i
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Control process exited,
Sep 25 15:14:01 raspberrypi systemd[1]: Failed to start The Apache HTTP Server.
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Unit entered failed sta
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Failed with result 'exi
~
~
~
~
~
~
~
~
/var/log/apache2/*.log...skipping...
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: 
   Active: failed (Result: exit-code) since Mon 2017-09-25 15:14:01 UTC; 23h ago
  Process: 470 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILUR

Sep 25 15:13:59 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Sep 25 15:14:01 raspberrypi apachectl[470]: AH00558: apache2: Could not reliably
Sep 25 15:14:01 raspberrypi apachectl[470]: (2)No such file or directory: AH0229
Sep 25 15:14:01 raspberrypi apachectl[470]: AH00014: Configuration check failed
Sep 25 15:14:01 raspberrypi apachectl[470]: Action 'start' failed.
Sep 25 15:14:01 raspberrypi apachectl[470]: The Apache error log may have more i
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Control process exited,
Sep 25 15:14:01 raspberrypi systemd[1]: Failed to start The Apache HTTP Server.
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Unit entered failed sta
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Failed with result 'exi
~
~
~
~
~
~
~
~
~
~
~
byte 1420/1420 (END)...skipping...
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2017-09-25 15:14:01 UTC; 23h ago
  Process: 470 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

Sep 25 15:13:59 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Sep 25 15:14:01 raspberrypi apachectl[470]: AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive 
Sep 25 15:14:01 raspberrypi apachectl[470]: (2)No such file or directory: AH02291: Cannot access directory '/var/log/apache2/' for main error log
Sep 25 15:14:01 raspberrypi apachectl[470]: AH00014: Configuration check failed
Sep 25 15:14:01 raspberrypi apachectl[470]: Action 'start' failed.
Sep 25 15:14:01 raspberrypi apachectl[470]: The Apache error log may have more information.
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Control process exited, code=exited status=1
Sep 25 15:14:01 raspberrypi systemd[1]: Failed to start The Apache HTTP Server.
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Unit entered failed state.
Sep 25 15:14:01 raspberrypi systemd[1]: apache2.service: Failed with result 'exit-code'.
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
byte 1420/1420 (END)

nachoparker, 2017-09-26:
ok, that's new. Did you set up 'ramlogs' in NextCloudPi?

user-jan, 2017-09-26:
yes, I did.

nachoparker, 2017-09-26:
Ok, I think that's the problem. Can you try disabling it to see if it fixes things? When did this start failing for you?

user-jan, 2017-09-26:
Bingo, it now works after disabling 'ramlogs' :-)

I think the problem started after updating to v0.28.2

Thanks alot, I'll be out now for a couple of hours.
Best,
Jan

nachoparker, 2017-09-26:
I am glad it works.

Thanks for reporting. Will investigate and fix

closed 2017-09-26 by nachoparker

JVAQUEROM, 2017-09-26:
I tried the config you suggested but still no luck. What I have is a blank page now.

On the other hand, I realized that for some reason, USBdrive became USBdrive1, so I had to change ncdatadir in nexcloudpi-config. But I got this message:

```
Launching nc-datadir
Error reading data directory. Is NextCloud running and configured?
Done. Press any key...

```

Maybe is there something related to premissions?

`ls -al` gives this

```
...
drwxrwx---  9 www-data root        4096 ago  3 19:36 ncdata
drwxr-xr-x  5 mysql    mysql       4096 sep 23 17:23 ncdatabase
...
```

but for instance, if I try nc-fix-permissions i get:

```

Launching nc-fix-permissions
data dir not found
Done. Press any key...

```

nachoparker, 2017-09-26:
aaah then try to change also `nc-database`

JVAQUEROM, 2017-09-26:
thank you!

```
pi@raspberrypi:~ $ ls -l /media
total 16
drwx------  2 root root 4096 sep 23 17:24 86c213e2-a9e2-4071-ae34-d46e13cba51f
drwxrwxrwx 13 root root 4096 sep 21 17:48 86c213e2-a9e2-4071-ae34-d46e13cba51f1
dr-x------  1 root root 4096 sep  9 01:08 Seagate Expansion Drive
drwxr-xr-x  2 root root 4096 sep 26 20:19 USBdrive
lrwxrwxrwx  1 root root   44 sep 26 21:06 USBdrive1 -> /media/86c213e2-a9e2-4071-ae34-d46e13cba51f1

```

```
pi@raspberrypi:~ $ systemctl status nc-automount-links
● nc-automount-links.service - Monitor /media for mountpoints and create USBdrive* symlinks
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2017-09-26 21:06:16 CEST; 3min 58s ago
 Main PID: 358 (nc-automount-li)
   CGroup: /system.slice/nc-automount-links.service
           ├─358 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─397 inotifywait --monitor --event create --event delete --format %f %e /media/
           ├─398 grep --line-buffered ISDIR
           └─399 /bin/bash /usr/local/etc/nc-automount-links-mon

sep 26 21:06:16 raspberrypi systemd[1]: Started Monitor /media for mountpoints and create USBdrive* symlinks.
sep 26 21:06:16 raspberrypi nc-automount-links-mon[358]: Setting up watches.
sep 26 21:06:16 raspberrypi nc-automount-links-mon[358]: Watches established.
sep 26 21:06:23 raspberrypi nc-automount-links-mon[358]: Seagate Expansion Drive DELETE,ISDIR
sep 26 21:06:24 raspberrypi nc-automount-links-mon[358]: 86c213e2-a9e2-4071-ae34-d46e13cba51f1 DELETE,ISDIR
sep 26 21:06:25 raspberrypi nc-automount-links-mon[358]: 86c213e2-a9e2-4071-ae34-d46e13cba51f1 CREATE,ISDIR
sep 26 21:06:25 raspberrypi nc-automount-links-mon[358]: Seagate Expansion Drive CREATE,ISDIR

```

nachoparker, 2017-09-26:
ok, so you have several partitions it seems

```
lsblk -f
```

I would remove `/media/USBdrive` (if is empty) and then reboot, so the link that now is USBdrive1 should be USBdrive after reboot. It looks like that remains from the old automount setup.

JVAQUEROM, 2017-09-26:
```
pi@raspberrypi:~ $ sudo lsblk -f
NAME        FSTYPE LABEL                   UUID                                 MOUNTPOINT
sda                                                                             
├─sda1      ntfs   Seagate Expansion Drive 4CB8F12CB8F11564                     /media/Seagate Expansion Drive
└─sda2      ext4                           86c213e2-a9e2-4071-ae34-d46e13cba51f /media/86c213e2-a9e2-4071-ae34-d46e13cba51f1
mmcblk0                                                                         
├─mmcblk0p1 vfat   boot                    DF02-C322                            /boot
└─mmcblk0p2 ext4                           28590797-4810-4851-b4ec-bf9672c2918c /

```

Yes, I decided to keep a small NTFS partition in case I had to use the drive some time to pass something from the pi to any Windows...although I usually use GNU/Linux.

For some reason, the swapfile has mounted on USBdrive...
```

sudo ls /media/USBdrive
swapfile

```

nachoparker, 2017-09-27:
just remove the swapfile, if you can't then

```
sudo swapoff /media/USBdrive/swapfile
```

and try again. If you can delete it, reboot and try again

```
ls -l /media
```

JVAQUEROM, 2017-09-27:
I managed to do as you said, with `swapoff` and remove the swapfile and USBdrive later, but now I get the folowing:

```
pi@raspberrypi:~ $ ls -l /media
total 12
drwx------  2 root root 4096 sep 23 17:24 86c213e2-a9e2-4071-ae34-d46e13cba51f
drwxrwxrwx 13 root root 4096 sep 21 17:48 86c213e2-a9e2-4071-ae34-d46e13cba51f1
dr-x------  1 root root 4096 sep  9 01:08 Seagate Expansion Drive
lrwxrwxrwx  1 root root   44 sep 27 19:32 USBdrive1 -> /media/86c213e2-a9e2-4071-ae34-d46e13cba51f1

```

so it jumps USBdrive and goes to USBdrive1 I don't know why....

nachoparker, 2017-09-28:
cool, good job

It should be in `/media/USBdrive`... have you tried rebooting? Maybe it's because of your partitioning?

JVAQUEROM, 2017-09-28:
Maybe it is because of the partitioning...I think that, as the first partition is NTFS, it does not mount it, so no USBdrive, but the second it does mount it, so I have USBdrive1. 

I think it is properly working now, I'll try it these days

ttoopp, 2018-02-28:
Hey, since few days I have the same problem. I cannot access my nextcloud system. I have nothing change in the last days.
Now I post some infos :)

systemctl status apache2
`* apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-02-28 21:51:49 CET; 5min ago
  Process: 5528 ExecReload=/usr/sbin/apachectl graceful (code=exited, status=0/SUCCESS)
 Main PID: 678 (apache2)
   CGroup: /system.slice/apache2.service
           |- 678 /usr/sbin/apache2 -k start
           |-5534 /usr/sbin/apache2 -k start
           `-5535 /usr/sbin/apache2 -k start

Feb 28 21:51:46 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Feb 28 21:51:49 raspberrypi systemd[1]: Started The Apache HTTP Server.
Feb 28 21:55:42 raspberrypi systemd[1]: Reloading The Apache HTTP Server.
Feb 28 21:55:42 raspberrypi systemd[1]: Reloaded The Apache HTTP Server.
`

tail -n 100 /var/log/apache2/*.log
`==> /var/log/apache2/access.log <==

==> /var/log/apache2/error.log <==
[Wed Feb 28 06:25:03.477265 2018] [ssl:warn] [pid 694:tid 1992269824] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Feb 28 06:25:04.000896 2018] [mpm_event:notice] [pid 694:tid 1992269824] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Wed Feb 28 06:25:04.001004 2018] [core:notice] [pid 694:tid 1992269824] AH00094: Command line: '/usr/sbin/apache2'
[Wed Feb 28 20:17:06.415586 2018] [ssl:warn] [pid 520:tid 1991987200] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Feb 28 20:17:07.053335 2018] [ssl:warn] [pid 682:tid 1991987200] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Feb 28 20:17:08.004619 2018] [mpm_event:notice] [pid 682:tid 1991987200] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Wed Feb 28 20:17:08.004801 2018] [core:notice] [pid 682:tid 1991987200] AH00094: Command line: '/usr/sbin/apache2'
[Wed Feb 28 21:29:56.766887 2018] [mpm_event:notice] [pid 682:tid 1991987200] AH00493: SIGUSR1 received.  Doing graceful restart
[Wed Feb 28 21:29:56.816065 2018] [ssl:warn] [pid 682:tid 1991987200] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Feb 28 21:29:57.000373 2018] [mpm_event:notice] [pid 682:tid 1991987200] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Wed Feb 28 21:29:57.000445 2018] [core:notice] [pid 682:tid 1991987200] AH00094: Command line: '/usr/sbin/apache2'
[Wed Feb 28 21:17:06.236295 2018] [ssl:warn] [pid 490:tid 1992421376] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Feb 28 21:17:07.058582 2018] [ssl:warn] [pid 643:tid 1992421376] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Feb 28 21:17:08.003828 2018] [mpm_event:notice] [pid 643:tid 1992421376] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Wed Feb 28 21:17:08.003989 2018] [core:notice] [pid 643:tid 1992421376] AH00094: Command line: '/usr/sbin/apache2'
[Wed Feb 28 21:17:06.306501 2018] [ssl:warn] [pid 473:tid 1992892416] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Feb 28 21:17:07.050919 2018] [ssl:warn] [pid 678:tid 1992892416] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Feb 28 21:17:08.003395 2018] [mpm_event:notice] [pid 678:tid 1992892416] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Wed Feb 28 21:17:08.003575 2018] [core:notice] [pid 678:tid 1992892416] AH00094: Command line: '/usr/sbin/apache2'
[Wed Feb 28 21:38:06.331046 2018] [mpm_event:notice] [pid 678:tid 1992892416] AH00491: caught SIGTERM, shutting down
[Wed Feb 28 21:38:06.499451 2018] [ssl:warn] [pid 2800:tid 1992347648] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Feb 28 21:38:07.096932 2018] [ssl:warn] [pid 2801:tid 1992347648] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Feb 28 21:38:08.005509 2018] [mpm_event:notice] [pid 2801:tid 1992347648] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Wed Feb 28 21:38:08.005754 2018] [core:notice] [pid 2801:tid 1992347648] AH00094: Command line: '/usr/sbin/apache2'
[Wed Feb 28 21:40:51.438240 2018] [mpm_event:notice] [pid 2801:tid 1992347648] AH00491: caught SIGTERM, shutting down
[Wed Feb 28 21:41:03.748596 2018] [ssl:warn] [pid 505:tid 1992003584] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Feb 28 21:41:04.049571 2018] [ssl:warn] [pid 676:tid 1992003584] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Feb 28 21:41:05.005610 2018] [mpm_event:notice] [pid 676:tid 1992003584] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Wed Feb 28 21:41:05.005915 2018] [core:notice] [pid 676:tid 1992003584] AH00094: Command line: '/usr/sbin/apache2'
[... cut ...]
Feb 28 21:51:44 raspberrypi systemd[1]: Started Monitor /media for mountpoints and create USBdrive* symlinks.
Feb 28 21:51:45 raspberrypi nc-automount-links-mon[344]: Setting up watches.
Feb 28 21:51:45 raspberrypi nc-automount-links-mon[344]: Watches established.
`
lsblk -f
`NAME        FSTYPE LABEL UUID                                 MOUNTPOINT
sda         ext4         edf8cb11-e31c-4a5b-984b-5be24915ab6c /media/USBdrive
mmcblk0
|-mmcblk0p1 vfat   boot  70CE-EB76                            /boot
`-mmcblk0p2 ext4         f2100b2f-ed84-4647-b5ae-089280112716 /
`

ttoopp, 2018-02-28:
Okay, it looks like, there is a problem ,with the mysql password.
i stopped the mysql daemon and start them with this command
mysqld_safe --skip-grant-tables
And now it works

ttoopp, 2018-02-28:
Okay, I have tried to change the password with this how to
https://www.howtoforge.com/reset-forgotten-mysql-root-password

After changing it, I put it in /root/.my.cnf and restart my Raspberry, but I get still no connection to my cloud.

Only if i start mysql with mysqld_safe --skip-grant-tables, I get access to the website.