nextcloud/nextcloudpi #212: Docker image with NC 12.0.3 / Update of NC in Docker

Gorofad, 2017-09-22:
Hello :)
I'm using the Docker image for the Rasperry Pi with NC 12.0.0. Recently I got the notification of my NC installation that a new Version 12.0.3 is available. So to update in the way docker usual works I did a `docker pull ownyourbits/nextcloudpi` to get the latest version. But the version available on the DockerHub is still 12.0.0. published over 3 months ago.
So then I tried to update my current installation via the web based updater, but it just initialize the updater and then stops: 
[image]
In my opinion a updated docker image with NC version 12.0.3 should be the correct way to update, so it is fine that the web based updater is not working.
So if you could publish an updated docker image that would be great :)
Thanks in advance!

nachoparker, 2017-09-22:
Hi,

The docker image is pretty old and limited. There are various reasons: have been busy with migrating to Stretch and the web interface and others.

The good news is that I have been caching up this week, so I have an almost working NCP docker based on Stretch with the latest stuff, including the web interface ncp-web.

I want to pay more attention to the docker image from now on because it is more portable to other boards, but it is still in its infancy compared to the SD card version. Some extras will be disabled or need extra work, and definitely the update mechanism will have to be revisited to make it work on docker pull.

Suscribe to ownyourbits.com to get the announcement ;)

Also: I know that there is a bunch of people using the docker container, but I don't think I have got much feedback on it. How does it work for you? Anything interesting for feedback?

Gorofad, 2017-09-22:
whow, that was a fast reply :) 
it's good to know that you are working on an update!
I am using the Docker image now for two or three months (so nearly from the beginning?) and apart from the update problem it works really good. But I do not have many users - in fact I am the only one using it - and I have only a few apps installed (calendar, contacts, tasks). But I'm really interested in using it more and inviting familiy members and friends to use it.
What really confuses me is that the database is included in the image. The normal docker way is I think to break down everything to the smallest possible piece, like it is done in the offial NC docker images. In my opinion a NC Docker image for Rasperry should only contain a working PHP enviroment but nothing else.
If you have specific questions regarding my setup or if you want me to test something just ask :)

PS: I wanted to subscribe to ownyourbits.com but couldn't find any form for that :see_no_evil:, could you provide me a direct link?

nachoparker, 2017-09-22:
Yeah, it's the docker way and I'll probably end up splitting it. It doesn't really matter that much, it's a matter of purism but if you want to reuse the database or balance it with docker swarm and that kind of thing it is good practice.

I will end up doing it. At this moment it's only an encapsulated version of the SD card image, and the main goal of this project is to make things dead easy so no docker compose yaml writing ;)

Also the new docker image will have more that just the database: letsencrypt, DDNS and all the rest. It's a package. All accessible from ncp-web, in a single container, that's the idea. I will probably end up splitting most of it, but I think there are things that are not that easy to combine efficiently between containers. Anyway, it is work in progress ;D

> PS: I wanted to subscribe to ownyourbits.com but couldn't find any form for that 🙈, could you provide me a direct link?

Haha, I was only talking about suscribing to the RSS feed. The Nextcloud News app is really good for that ;)

testbird, 2017-09-23:
Could you mention your thoughts on how this will work?
If I understand it correctly, docker images put the burden of security updates on the image maintainer, as he has to supply a new image quickly for each patch in any of the packages installed in the image.
Will the ncp-web mechanisms be able to update the packages within the image?

I'm thinking of the docker docker image because I've seen the openmediavault webinterface already allows to create users can set up syncthing nodes for them very well with per user services! 

Syncthing is just way better for serious, reliable, and quick file synchronization, and may even be combinable with the nextcloud file access: https://forum.syncthing.net/t/integrating-syncthing-with-owncloud/6332

nachoparker, 2017-09-23:
> Could you mention your thoughts on how this will work?
If I understand it correctly, docker images put the burden of security updates on the image maintainer, as he has to supply a new image quickly for each patch in any of the packages installed in the image.
Will the ncp-web mechanisms be able to update the packages within the image?

Well, I have some general ideas about how to push the container updates for the future, but at least for now unattended upgrades is going to be running inside the container and will keep the software up to date, just like with the SD card image.

> I'm thinking of the docker image because I've seen the openmediavault webinterface already allows to create users, and can set up syncthing nodes for them very well with per user services!

> Syncthing is just way better for serious, reliable, and quick file synchronization, and may even be combinable with the nextcloud file access: https://forum.syncthing.net/t/integrating-syncthing-with-owncloud/6332

so what is  your idea exactly? At this moment you can mount your files with samba and nfs if you want to access without syncing files, which I personally prefer.

testbird, 2017-09-26:
OK then, package manager updates and no separate data container, good to know and sounds ok that updates get installed.

> so what is your idea exactly? At this moment you can mount your files with samba and nfs if you want to access without syncing files, which I personally prefer.

I'd like using the Nextcloud Webinterface for sharing and collaborating on things with others, and samba access for devices without mass storage (in LAN), but the nextcloud syncing over php/http is just not good enough. Syncthing on the other hand is able to sync verry efficiently even handling small changes in large files, renaming, moving, and most importantly also local peer-to-peer sync (say between laptop and phone) if no server connection is available.

nachoparker, 2017-09-27:
yeah, I've heard of it but have not tried it. I like NAS like SAMBA because you don't have several duplicated copies around that need to be in sync. I guess something like syncthing would be nice... one day ;D

testbird, 2017-10-02:
> something like syncthing would be nice... one day ;D

Yep, files available (and replicated) even if offline!

nachoparker, 2017-11-15:
Sorry for the wait, but finally we now have the creation of docker containers streamlined, so they will be updated at the same time as the rest of NCP.

https://ownyourbits.com/2017/11/15/nextcloudpi-dockers-for-x86-and-arm/

closed 2017-11-15 by nachoparker

testbird, 2017-11-25:
Maybe useful suggestions in this comment: https://joeyh.name/blog/entry/docker_run_debian/

nachoparker, 2017-11-28:
very interesting, thanks