nextcloud/nextcloudpi #214: Moving data directory

fzrkid, 2017-09-24:
Am I doing something in the wrong order?

When I try to move the data directory I get: "/media/USBDrive/ncdata is not empty"
 
I partitioned the drive and named it ncdata. I tried it both with gparted on my local machine and with the web interface. Both times when trying to move the data directory I get the same error msg.
"/media/USBDrive/ncdata is not empty"

Would be handy to have a 1st time setup panel. So the first time a user accessed the web interface it give a configuration dialogue so everything is done in the correct order. sort of like a dummy guide to nextcloudpi configuration.

Example, do I create a user first, or move the data directory first? Does it matter? Can I do it in the wrong order?

Also a way to browse or see the directories on the USB drive from within the web interface. Something that will show directory size so one can verify the files are being written to the correct location. Would be handy to be able to compare between the data drive and the back-up drive.

nachoparker, 2017-09-25:
Seems that that directory is not empty. Try a different name, or remove the current one if there is no important data in there.

> Would be handy to have a 1st time setup panel. So the first time a user accessed the web interface it give a configuration dialogue so everything is done in the correct order. sort of like a dummy guide to nextcloudpi configuration.

We are releasing the new setup wizard this week! (#129)

> Also a way to browse or see the directories on the USB drive from within the web interface. Something that will show directory size so one can verify the files are being written to the correct location. Would be handy to be able to compare between the data drive and the back-up drive.

That would go in #159

Thank you, that is good feedback to learn what the user thinks.

closed 2017-09-26 by nachoparker

Redsalamander, 2017-11-16:
I know what you're looking for here just figured it out. 
Is doesn't show you a list of directories and doesn't actually point to the external drive as you would expect. 

To make this work you need to change the directory form /media/USBDrive/ncdata to /media/myCloudDrive/ncdata "myCloudDrive" is the default its formatted as in the wizard so you should change that if its the case. 
The /ncdata adds directory on the external drive for the data. If you leave that out it tells you the drive already has data on. " ie not empty"