nextcloud/nextcloudpi #215: ncp-scan locks files, can't rerun

thibmaek, 2017-09-24:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
latest

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
latest

### Expected behavior
ncp-scan correctly scans and sets permissions for external data

### Actual behaviour
ncp-scan locks files and can't run again

```
Exception during scan: "files/1064ff7e7d41b3e49dea4fb5365abe8d" is locked
#0 /var/www/nextcloud/lib/private/Files/Storage/Common.php(704): OC\Lock\DBLockingProvider->acquireLock('files/1064ff7e7...', 2)
#1 /var/www/nextcloud/lib/private/Files/Storage/Wrapper/Wrapper.php(588): OC\Files\Storage\Common->acquireLock('scanner::', 2, Object(OC\Lock\DBLockingProvider))
#2 /var/www/nextcloud/lib/private/Files/Cache/Scanner.php(318): OC\Files\Storage\Wrapper\Wrapper->acquireLock('scanner::', 2, Object(OC\Lock\DBLockingProvider))
#3 /var/www/nextcloud/lib/private/Files/Utils/Scanner.php(242): OC\Files\Cache\Scanner->scan('', true, 3)
#4 /var/www/nextcloud/apps/files/lib/Command/Scan.php(161): OC\Files\Utils\Scanner->scan('/thibmaek')
#5 /var/www/nextcloud/apps/files/lib/Command/Scan.php(234): OCA\Files\Command\Scan->scanFiles('thibmaek', '/thibmaek', false, Object(Symfony\Component\Console\Output\ConsoleOutput), false)
#6 /var/www/nextcloud/3rdparty/symfony/console/Command/Command.php(256): OCA\Files\Command\Scan->execute(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#7 /var/www/nextcloud/core/Command/Base.php(161): Symfony\Component\Console\Command\Command->run(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#8 /var/www/nextcloud/3rdparty/symfony/console/Application.php(818): OC\Core\Command\Base->run(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#9 /var/www/nextcloud/3rdparty/symfony/console/Application.php(186): Symfony\Component\Console\Application->doRunCommand(Object(OCA\Files\Command\Scan), Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#10 /var/www/nextcloud/3rdparty/symfony/console/Application.php(117): Symfony\Component\Console\Application->doRun(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#11 /var/www/nextcloud/lib/private/Console/Application.php(170): Symfony\Component\Console\Application->run(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#12 /var/www/nextcloud/console.php(100): OC\Console\Application->run()
#13 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#14 {main}

+---------+-------+--------------+
| Folders | Files | Elapsed time |
+---------+-------+--------------+
| 0       | 0     | 00:00:00     |
+---------+-------+--------------+
```

Already checked https://github.com/nextcloud/nextcloudpi/issues/154 and tried rebooting. No way to fix so far. File does not exist.

nachoparker, 2017-09-24:
any file with weird names, like '$TRASH'?

thibmaek, 2017-09-24:
No files found with `find . -type f -name '$*'` or `find . -type d -name '$*'`

thibmaek, 2017-09-24:
oc_file_locks table also shows 0 entries.

[image]

nachoparker, 2017-09-25:
Mmmm, I don't know, never happened to me. I suggest you ask in the Nextcloud forums

https://help.nextcloud.com/c/support/

thibmaek, 2017-09-25:
I'm going to backup all this important stuff to Google Drive first…
Don't really feel confident moving stuff around on fs and database.

If I find something, I'll report back ;)

thibmaek, 2017-09-25:
Might just reinstall the whole thing since USB boot is really messy to work with if you use external drives.

nachoparker, 2017-09-30:
will close now. Please anyone reopen if you see this again or find a solution

closed 2017-09-30 by nachoparker