nextcloud/nextcloudpi #220: Domain in overwrite.cli.url in config.php commented out?

ObfuscateTheWeb, 2017-09-26:
In config.php the domain.ddns.net part was a blue color in my putty terminal, indicating that it is being commented out

Recently, my domain isn't automatically redirecting to https. It used to redirect to https if I typed in `domain.ddns.net` now it doesn't redirect and I need to type out `https://domain.ddns.net` for it to work, does the // comment out the domain.ddns.net and somehow causes issues? 
I don't know what it looked like before, this is the first time I noticed it was blue

Looks like this:

[image]

nachoparker, 2017-09-26:
Ok, this has nothing to do with `config.php` but with apache redirection. 

So when you try http do you get a blank screen? or a simple website saying 'it works!'?

ObfuscateTheWeb, 2017-09-26:
blank screen, with browser saying problem loading page

nachoparker, 2017-09-26:
can you try (re)activating `nc-httpsonly`?

ObfuscateTheWeb, 2017-09-26:
done

nachoparker, 2017-09-26:
does it work now?

ObfuscateTheWeb, 2017-09-26:
nope, still does not load. I tried `domain.ddns.net`, `http://domain.ddns.net `and the only one that works is `https://domain.ddns.net`

nachoparker, 2017-09-26:
Ok, then let's see what happens with virtual hosts

```
sudo apache2ctl -S
cat /etc/apache2/sites-available/000-default.conf
```

This is how mine looks 

```
<VirtualHost _default_:80>                       
  DocumentRoot /var/www/nextcloud                
  <IfModule mod_rewrite.c>                       
    RewriteEngine On    
    RewriteCond %{HTTPS} !=on                    
    RewriteRule ^/?(.*) https://%{SERVER_NAME}/$1 [R,L]                                           
  </IfModule>           
</VirtualHost>
```

ObfuscateTheWeb, 2017-09-26:
Mine:

```
<VirtualHost _default_:80>
  DocumentRoot /var/www/nextcloud
  <IfModule mod_rewrite.c>
    RewriteEngine On
    RewriteCond %{HTTPS} !=on
    RewriteRule ^/?(.*) https://%{SERVER_NAME}/$1 [R,L]
  </IfModule>
</VirtualHost>
```

nachoparker, 2017-09-26:
looks good, so is your 000-default.conf enabled?

```
sudo apache2ctl -S
```

ObfuscateTheWeb, 2017-09-26:
```
AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive globally to suppress this message
VirtualHost configuration:
*:80                   127.0.1.1 (/etc/apache2/sites-enabled/000-default.conf:1)
*:4443                 127.0.1.1 (/etc/apache2/sites-enabled/ncp.conf:2)
*:443                  skyspace.ddns.net (/etc/apache2/sites-enabled/nextcloud.conf:2)
ServerRoot: "/etc/apache2"
Main DocumentRoot: "/var/www/html"
Main ErrorLog: "/var/log/apache2/error.log"
Mutex watchdog-callback: using_defaults
Mutex rewrite-map: using_defaults
Mutex ssl-stapling-refresh: using_defaults
Mutex ssl-stapling: using_defaults
Mutex proxy: using_defaults
Mutex ssl-cache: using_defaults
Mutex default: dir="/var/run/apache2/" mechanism=default 
PidFile: "/var/run/apache2/apache2.pid"
Define: DUMP_VHOSTS
Define: DUMP_RUN_CFG
User: name="www-data" id=33
Group: name="www-data" id=33
```

nachoparker, 2017-09-26:
looks alright... is the module disabled by any chance??

```
sudo a2ensite 000-default
sudo a2enmod rewrite
```

And try again

ObfuscateTheWeb, 2017-09-26:
```
sudo a2ensite 000-default
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_PAPER = "ms_MY.UTF-8",
	LC_ADDRESS = "ms_MY.UTF-8",
	LC_MONETARY = "ms_MY.UTF-8",
	LC_NUMERIC = "ms_MY.UTF-8",
	LC_TELEPHONE = "ms_MY.UTF-8",
	LC_IDENTIFICATION = "ms_MY.UTF-8",
	LC_MEASUREMENT = "ms_MY.UTF-8",
	LC_NAME = "ms_MY.UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
Site 000-default already enabled
pi@raspberrypi:~ $ sudo a2enmod rewrite
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_PAPER = "ms_MY.UTF-8",
	LC_ADDRESS = "ms_MY.UTF-8",
	LC_MONETARY = "ms_MY.UTF-8",
	LC_NUMERIC = "ms_MY.UTF-8",
	LC_TELEPHONE = "ms_MY.UTF-8",
	LC_IDENTIFICATION = "ms_MY.UTF-8",
	LC_MEASUREMENT = "ms_MY.UTF-8",
	LC_NAME = "ms_MY.UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
Module rewrite already enabled
```

nachoparker, 2017-09-26:
everything is enabled, everything is configured, it should work...

```
sudo systemctl restart apache2
```

and try to access again

ObfuscateTheWeb, 2017-09-26:
not working

nachoparker, 2017-09-26:
I don't know if those locale warnings will have anything to do

ObfuscateTheWeb, 2017-09-26:
hmm..okay, could I pm you my domain so you could try it and see if it works for you on your end?

nachoparker, 2017-09-26:
ok, nacho at ownyourbits dot com

ObfuscateTheWeb, 2017-09-26:
email you?

nachoparker, 2017-09-26:
yeah, or pm in the nextcloud forums if you want

ObfuscateTheWeb, 2017-09-26:
emailed

ObfuscateTheWeb, 2017-09-26:
I try it on my mobile with firefox focus, this way I can have a fresh browser to test it on, avoids cache issues and all that stuff

Using firefox focus, I type in `domain.ddns.net ` and doesn't load, must type in https:// to load

nachoparker, 2017-09-26:
I don't even have ping to your domain from outside, maybe a router port issue?

nachoparker, 2017-09-26:
ok, no ping, no http but I am able to access through https.

Through http it just stays there, trying to load but doesn't return anything.

Maybe you have ISP blocking ports ? or in your router ?

ObfuscateTheWeb, 2017-09-26:
nope, I port forwarded it correctly and my ISP doesn't block stuff here

nachoparker, 2017-09-26:
wait, I tried http again and it just worked this time

ObfuscateTheWeb, 2017-09-26:
lol..what's going on here haha

nachoparker, 2017-09-26:
yes, it works for me right now, maybe I should have waited a bit longer the first time

ObfuscateTheWeb, 2017-09-26:
just type domain.ddns.net ?

nachoparker, 2017-09-26:
yes, both that and http://domain.ddns.net

nachoparker, 2017-09-26:
Maybe try on your phone without wifi.

I'll close now because it's not NCP related, can still talk here thouggh

closed 2017-09-26 by nachoparker

ObfuscateTheWeb, 2017-09-26:
super weird, doesn't work on my firefox focus with a clean browser instance..lol?

ObfuscateTheWeb, 2017-09-26:
no wifi? how will it connect ?

nachoparker, 2017-09-26:
I mean try from your phone over 4G, no wifi. That way it will be from outside your house effectively

ObfuscateTheWeb, 2017-09-26:
no data on my phone..

nachoparker, 2017-09-26:
ok, it seems to me like is something in your internal network, because it does work from outside

ObfuscateTheWeb, 2017-09-26:
only thing I could think of is my new dnsmasq dns server running on my second raspberry pi since yesterday at 4 am

Instead of typing 192.168.1.127 for my cloud, I wanted a domain to point to it instead, so I could now type cloud.home and it will point to 192.168.1.127

I introduced a new dns server basically, shouldn't be an issue hopefully..

nachoparker, 2017-09-26:
well, Nextcloudpi comes with dnsmasq integrated to do exactly that. There is an option for dnsmasq that will set it correctly for that use case.

See this

https://ownyourbits.com/2017/03/09/dnsmasq-as-dns-cache-server-for-nextcloudpi-and-raspbian/

ObfuscateTheWeb, 2017-09-26:
didn't notice, I configured it manually, certainly works though

ObfuscateTheWeb, 2017-09-26:
dig cloud.home isn't using my dnsmasq server..okay this is something new

ObfuscateTheWeb, 2017-09-27:
Okay, cleaned out my dnsmasq and used the dnsmasq on the nextcloudpi, `domain.ddns.net` now works
But what if I want 2 domains, one for the inside and one for the outside?
I want `domain.ddns.net` for the outside and I want `cloud.home` if I'm on the LAN

I know two places to edit, `/etc/hosts and config.php `

Here's my /etc/hosts:

```
127.0.0.1	localhost
::1		localhost ip6-localhost ip6-loopback
ff02::1		ip6-allnodes
ff02::2		ip6-allrouters

127.0.1.1	raspberrypi
192.168.1.127   domain.ddns.net
192.168.1.23    password.home
192.168.1.127   cloud.home
```

And config.php:
```
'trusted_domains' => 
  array (
    0 => 'localhost',
    1 => '192.168.1.127',
    4 => 'domain.ddns.net',
    2 => 'domain.ddns.net',
    3 => 'cloud.home',
  ),
```

Am I doing the configurations right?

nachoparker, 2017-09-27:
> set static IP and DNS in the network settings

That's what I do. If you want android to use your local DNS through DHCP I guess you will have to configure your router to use your NCP as the primary DNS. Some routers can do this, some others can't.

ObfuscateTheWeb, 2017-09-27:
Okay, everything works now..
One more thing, since updating to nextcloudpi v30 I'm now asked to enter username/password to login into ncp-web, what am I supposed to enter?

I tried both my nextcloud admin login and my raspberry pi login, both do not work

nachoparker, 2017-09-27:
cool

the username is `pi` and the password is your pi's password for ncp-web

ObfuscateTheWeb, 2017-09-27:
Great, that worked 
Thanks nacho