nextcloud/nextcloudpi #224: Update to 0.30.0 results in unauthorized access to ncp-web

Shadowstreik, 2017-09-27:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )

v0.30.0

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

You can discover this by executing

```
cat /usr/local/etc/ncp-baseimage
```
untagged
### Expected behavior

To access ncp-web

### Actual behaviour
error
"UNAUTHORIZED - This server could not verify that you are authorized to access the document requested. Either you supplied the wrong credentials (e.g., bad password), or your browser doesn't understand how to supply the credentials required."

Authentication pop-up appears (Linux Mint) asking for username and password. "Site says, 'ncp-web login"

### Steps to reproduce, from a freshly installed image
Run ncp-update from ncp-web, previous NCP version. Attempt to access.
### Include logs

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
```
Paste the output of `systemctl status apache2`

● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled)
   Active: active (running) since Wed 2017-09-27 10:49:57 EDT; 8min ago
  Process: 668 ExecStart=/usr/sbin/apachectl start (code=exited, status=0/SUCCESS)
 Main PID: 826 (apache2)
   CGroup: /system.slice/apache2.service
           ├─826 /usr/sbin/apache2 -k start
           ├─829 /usr/sbin/apache2 -k start
           └─830 /usr/sbin/apache2 -k start

Sep 27 10:49:56 ramrkry1 apachectl[668]: AH00558: apache2: Could not reliabl...e
Sep 27 10:49:57 ramrkry1 systemd[1]: Started The Apache HTTP Server.
Hint: Some lines were ellipsized, use -l to show in full.

Paste the output of `tail -n 100 /var/log/apache2/*.log`

==> /var/log/apache2/access.log <==

==> /var/log/apache2/error.log <==
[Wed Sep 27 14:47:14.461222 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:14.661501 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:14.861820 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:15.062109 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:15.262489 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:15.462743 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:15.663032 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:15.863337 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:16.063633 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:16.263934 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:16.464210 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:16.664538 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:16.864833 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:17.065124 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:17.265422 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Wed Sep 27 14:47:17.465690 2017] [proxy_fcgi:error] [pid 7694:tid 1935033392] [client 192.168.0.100:43610] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[... 142 lines cut ...]
2017-09-27 14:49:44 1638835184 [Note] InnoDB: Starting shutdown...
2017-09-27 14:49:44 1638835184 [Note] InnoDB: Waiting for page_cleaner to finish flushing of buffer pool
2017-09-27 14:49:46 1638835184 [Note] InnoDB: Shutdown completed; log sequence number 494814864
2017-09-27 14:49:46 1638835184 [Note] /usr/sbin/mysqld: Shutdown complete

2017-09-27 10:50:16 1989660672 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2017-09-27 10:50:16 1989660672 [Note] InnoDB: The InnoDB memory heap is disabled
2017-09-27 10:50:16 1989660672 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2017-09-27 10:50:16 1989660672 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2017-09-27 10:50:16 1989660672 [Note] InnoDB: Compressed tables use zlib 1.2.8
2017-09-27 10:50:16 1989660672 [Note] InnoDB: Using Linux native AIO
2017-09-27 10:50:17 1989660672 [Note] InnoDB: Using generic crc32 instructions
2017-09-27 10:50:17 1989660672 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2017-09-27 10:50:17 1989660672 [Note] InnoDB: Completed initialization of buffer pool
2017-09-27 10:50:17 1989660672 [Note] InnoDB: Highest supported file format is Barracuda.
2017-09-27 10:50:17 1989660672 [Note] InnoDB: 128 rollback segment(s) are active.
2017-09-27 10:50:17 1989660672 [Note] InnoDB: Waiting for purge to start
2017-09-27 10:50:17 1989660672 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 494814864
2017-09-27 10:50:18 1447031792 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-09-27 10:50:18 1989660672 [Note] Plugin 'FEEDBACK' is disabled.
2017-09-27 10:50:18 1989660672 [Note] Server socket created on IP: '127.0.0.1'.
2017-09-27 10:50:18 1989660672 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```

nachoparker, 2017-09-27:
This is an important security mechanism coming from #132.

The username is `pi` and the password is your pi's password ( `raspberry` by default ). It is recommended that everyone changes their default `pi` password through SSH.

The wiki has been updated to reflect this.

Shadowstreik, 2017-09-27:
Thank you. I put in the login info. A page appears with only text, no CSS, and it asks for the login, again, to which it doesn't accept the pi login.

karpaz, 2017-09-27:
When you try it 10x times etc. it will accept that. :)

Shadowstreik, 2017-09-27:
Performed approx. 25x times. That failed. I would believe once would be the needed amount to log in.

nachoparker, 2017-09-27:
Somebody reported having a password with weird characters that wasn't been accepted.

Are you using your system user `pi` password? 

You can try changing it

```
sudo passwd pi
```

hurrikan77, 2017-09-27:
I have to login approx. 4-6 times until all parts of the webUI are loaded. Could you chnage that, that only one login is nessesary?

nachoparker, 2017-09-27:
> I have to login approx. 4-6 times until all parts of the webUI are loaded. Could you chnage that, that only one login is nessesary?

I did not have that issue. Could you please open an issue with more information ( browser, cookies accepted, noscript or other addons... )

Shadowstreik, 2017-09-27:
Performed password change. No effect. The same thing happens. I've also done a warm reboot, cleared history in FF.

nachoparker, 2017-09-27:
output of 

```
type pwauth
```
?

Shadowstreik, 2017-09-27:
pwauth is /usr/sbin/pwauth

hurrikan77, 2017-09-28:
I tried with Firefox 56.0 and Vivaldi 1.12.

nachoparker, 2017-09-28:
It seems like it works, but it asks for password 4-5 times. That will have to be fixed.

nachoparker, 2017-09-30:
Solved in `v0.31.2`. I'll leave this open for now, for people to know that they have to update the latest image.

closed 2017-10-01 by nachoparker