nextcloud/nextcloudpi #226: Apache server fails

StefvanEgmond, 2017-09-27:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v0.30.0
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
NextCloudPi_09-12-17
You can discover this by executing

```
cat /usr/local/etc/ncp-baseimage
```

### Expected behavior
Starting up Apache server
### Actual behaviour
Not starting up Apache server

### Include logs
<details>
<summary>Apache logs</summary>

```
apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2017-09-27 20:45:47 UTC; 10min ago
  Process: 885 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

Sep 27 20:45:47 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Sep 27 20:45:47 raspberrypi apachectl[885]: AH00526: Syntax error on line 17 of /etc/apache2/sites-enabled/ncp.conf:
Sep 27 20:45:47 raspberrypi apachectl[885]: Unknown Authn provider: external
Sep 27 20:45:47 raspberrypi apachectl[885]: Action 'start' failed.
Sep 27 20:45:47 raspberrypi apachectl[885]: The Apache error log may have more information.
Sep 27 20:45:47 raspberrypi systemd[1]: apache2.service: Control process exited, code=exited status=1
Sep 27 20:45:47 raspberrypi systemd[1]: Failed to start The Apache HTTP Server.
Sep 27 20:45:47 raspberrypi systemd[1]: apache2.service: Unit entered failed state.
Sep 27 20:45:47 raspberrypi systemd[1]: apache2.service: Failed with result 'exit-code'.

==> /var/log/apache2/access.log <==

==> /var/log/apache2/error.log <==
[Wed Sep 27 19:53:53.623637 2017] [authz_host:error] [pid 664:tid 1885455408] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /poweroff.svg failed, reason: unable to get the remote host name, referer: https://192.168.178.100:4443/ncp.css
[Wed Sep 27 19:53:53.623698 2017] [authz_host:error] [pid 664:tid 1843512368] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /info.svg failed, reason: unable to get the remote host name, referer: https://192.168.178.100:4443/ncp.css
[Wed Sep 27 19:53:53.624252 2017] [authz_host:error] [pid 664:tid 1793180720] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-output.php failed, reason: unable to get the remote host name
[Wed Sep 27 19:53:53.651459 2017] [authz_host:error] [pid 664:tid 1801569328] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /favicon.png failed, reason: unable to get the remote host name
[Wed Sep 27 19:57:33.237952 2017] [authz_host:error] [pid 664:tid 1851900976] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Sep 27 19:57:33.243512 2017] [proxy_fcgi:error] [pid 664:tid 1851900976] [client 192.168.178.24:50605] AH01071: Got error 'PHP message: PHP Notice:  Undefined variable: checked in /var/www/ncp-web/ncp-launcher.php on line 46\n'
[Wed Sep 27 20:16:06.582610 2017] [authz_host:error] [pid 664:tid 1902232624] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Wed Sep 27 20:16:06.583461 2017] [authz_host:error] [pid 664:tid 1902232624] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /index.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:16:11.552418 2017] [authz_host:error] [pid 664:tid 1960952880] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:16:17.982227 2017] [authz_host:error] [pid 664:tid 1986118704] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp.css failed, reason: unable to get the remote host name, referer: https://192.168.178.100:4443/
[Wed Sep 27 20:16:18.105815 2017] [authz_host:error] [pid 664:tid 1784792112] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-output.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:16:18.128942 2017] [authz_host:error] [pid 664:tid 1935787056] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /favicon.png failed, reason: unable to get the remote host name
[Wed Sep 27 20:16:20.686048 2017] [authz_host:error] [pid 664:tid 1893844016] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:16:20.691263 2017] [proxy_fcgi:error] [pid 664:tid 1893844016] [client 192.168.178.24:50605] AH01071: Got error 'PHP message: PHP Notice:  Undefined variable: checked in /var/www/ncp-web/ncp-launcher.php on line 46\n'
[Wed Sep 27 20:16:21.501307 2017] [authz_host:error] [pid 664:tid 1868678192] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:16:22.990654 2017] [authz_host:error] [pid 664:tid 1969341488] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:16:39.211893 2017] [authz_host:error] [pid 664:tid 1877066800] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:17:01.787055 2017] [authz_host:error] [pid 664:tid 1952564272] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:17:03.314625 2017] [authz_host:error] [pid 664:tid 1944175664] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:17:03.319233 2017] [proxy_fcgi:error] [pid 664:tid 1944175664] [client 192.168.178.24:50605] AH01071: Got error 'PHP message: PHP Notice:  Undefined variable: checked in /var/www/ncp-web/ncp-launcher.php on line 46\n'
[Wed Sep 27 20:17:09.262125 2017] [authz_host:error] [pid 664:tid 1826735152] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:17:19.617112 2017] [authz_host:error] [pid 664:tid 1835123760] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:17:46.752257 2017] [authz_host:error] [pid 664:tid 1910621232] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:17:47.503719 2017] [authz_host:error] [pid 664:tid 1851900976] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:17:47.992238 2017] [authz_host:error] [pid 664:tid 1860289584] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Wed Sep 27 20:17:47.993149 2017] [authz_host:error] [pid 664:tid 1860289584] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /index.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:17:48.068557 2017] [authz_host:error] [pid 664:tid 1902232624] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp.css failed, reason: unable to get the remote host name, referer: https://192.168.178.100:4443/
[Wed Sep 27 20:17:48.130505 2017] [authz_host:error] [pid 664:tid 1960952880] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /ncp-output.php failed, reason: unable to get the remote host name
[Wed Sep 27 20:17:48.232274 2017] [authz_host:error] [pid 664:tid 1986118704] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /favicon.png failed, reason: unable to get the remote host name
[Wed Sep 27 20:17:49.212061 2017] [authz_host:error] [pid 664:tid 1843512368] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to / failed, reason: unable to get the remote host name
[Wed Sep 27 20:17:49.212942 2017] [authz_host:error] [pid 664:tid 1843512368] [client 192.168.178.24:50605] AH01753: access check of 'localhost' to /index.php failed, reason: unable to get the remote host name
[... cut ...]
localhost:4443 192.168.178.24 - - [27/Sep/2017:20:33:51 +0000] "POST /ncp-launcher.php HTTP/2.0" 200 102 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/61.0.3163.100 Safari/537.36"
localhost:4443 192.168.178.24 - - [27/Sep/2017:20:33:52 +0000] "POST /ncp-launcher.php HTTP/2.0" 200 101 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/61.0.3163.100 Safari/537.36"
localhost:4443 192.168.178.24 - - [27/Sep/2017:20:28:36 +0000] "GET /ncp-output.php HTTP/2.0" 200 4199008 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/61.0.3163.100 Safari/537.36"

```
</details>

nachoparker, 2017-09-27:
try this

```
apt-get update
apt-get install -y --no-install-recommends libapache2-mod-authnz-external pwauth
a2enmod authnz_external authn_core auth_basic
```

StefvanEgmond, 2017-09-28:
I reinstalled NextCloudPi and the server is booting up. Now the control panel on :4443 is asking for a username and password. I tried different stuff but it doesn't help. Any notes I missed?

nachoparker, 2017-09-28:
There is password protection now. User `pi` password the system pi password

See https://github.com/nextcloud/nextcloudpi/issues/224

closed 2017-09-28 by nachoparker

StefvanEgmond, 2017-09-28:
I tried that username & password. I changed the password in SSH to something else but I still can't login with that credentials.

nachoparker, 2017-09-28:
Try changing the pi password to something simple. Somebody reported problems with special characters

```
sudo passwd pi
```

StefvanEgmond, 2017-09-28:
I changed it to "1234" but it's still asking for auth.

StefvanEgmond, 2017-09-28:
I updated some things and now I get thing message on SSH:

> Failed to set wall message, ignoring: The name org.freedesktop.PolicyKit1 was not provided by any .service files
Failed to call ScheduleShutdown in logind, proceeding with immediate shutdown: The name org.freedesktop.PolicyKit1 was not provided by any .service files

StefvanEgmond, 2017-09-28:
@nachoparker Any solution yet?

nachoparker, 2017-09-28:
Ok, if your password for `pi` is `1234`, what is the output of

```
echo -e "pi\1234\n" | pwauth && echo success || echo fail
```

StefvanEgmond, 2017-09-28:
[image]
Fail

nachoparker, 2017-09-28:
it seems to me that that is not your `pi` password. Can you try this again

```
sudo passwd pi
```

nachoparker, 2017-09-28:
are you sure you have not disabled password login or anything like that for user `pi`?

StefvanEgmond, 2017-09-28:
I can login to SSH with pi and 1234 as password

StefvanEgmond, 2017-09-28:
Oh now it gives success as echo

StefvanEgmond, 2017-09-28:
I still can't login into the web admin of NextCloudPi

nachoparker, 2017-09-28:
ok, pwauth works, try then from a clean browser instance

StefvanEgmond, 2017-09-28:
I tried to login with "pi" and "1234" as password on a fresh incognito browser but it still doesn't allow me to enter. After a couple tries I see the website loading into the browser but the auth keeps popping up.

nachoparker, 2017-09-28:
Somebody reported having to input the password 4 or 5 times for the whole website to load. I still have to investigate that

StefvanEgmond, 2017-09-28:
Yep, I logged in a couple more times and now I can use the interface! But when I refresh it's asking for auth. again.

nachoparker, 2017-09-28:
what browser are you using? plugins? cookies allowed?

StefvanEgmond, 2017-09-28:
Chrome 61.0.3163.100. Plugins: AdBlock, cookie editor, streamkeys and sessionbuddy. Cookies are allowed

nachoparker, 2017-09-28:
I just reproduced it on a clean FF instance as well

nachoparker, 2017-09-28:
In any case, that would be another issue. Closing this now